perexg/satip-axe #69: Feature request: ACL

satip, 2016-06-27:
it would be nice, if i can prevent some client to connect to the satip-box
maybe simply with iptables?

iam using satip with tvheadend and user acl´s to prevent my child from being watching some of channels. but this cannot prevent a direct connection to the satip-box 

my workaround at this time: using vlan´s

perexg, 2016-06-28:
It would be probably better (performance) to blacklist/whitelist IPs in minisatip. The netfilter is just another layer in kernel which may decrease performance.

Another solutions:

1) direct cable to TVH server
2) static ARP !!!

lars18th, 2016-10-24:
Hi,

+1 for a blacklist/whitelists.
I hope it will be included in the original minisatip (and perhaps also in TVH).