perexg/satip-axe #71: Digibit r1 don't boot thru USB

vuiperez, 2016-07-01:
hello Perexg how are you ?
i have one issue i made this,
Installation:

set DHCP in the original firmware (not 100% required, but several users reported that without this change the network might not work)
setup a DHCP server for the box ethernet MAC address
copy satip-axe-$(VERSION).usb as idl4k.scr (rename the file) to the USB stick
copy satip-axe-$(VERSION).fw (no rename) to the USB stick
both files should be in the root directory on the USB stick
plug the USB stick to the upper USB port in box
turn on the box
the tuner LEDs must blink (Knight Rider effect) for few seconds
box is accessible through telnet or ssh, IP address is obtained using DHCP
the default login/password is root/satip
the firmware is not written to the internal flash - when the USB stick is not in the box, the original firmware will be booted.

has you intructed i renamed the ***.usb to idl4k.scr and put the FW file without renaming but DIGIBIT don't boot with you r FW. what i'm doing wrong ?

best regards

perexg, 2016-07-01:
Try another USB stick. Some sticks do not work - if you don't see the LED blinking after power on, then uboot does not load (detect) the firmware on stick.

vuiperez, 2016-07-01:
i user a 4 GB usb with fat sys.
what do you advise ?

vuiperez, 2016-07-04:
working thanks :)

closed 2016-07-04 by perexg

trujulu, 2016-07-04:
Try fat32

vuiperez, 2016-07-04:
Thanks i Have allready solved.

Best regards.