perexg/satip-axe #72: Tuning failed after several hours (12-24)

spadge4711, 2016-07-02:
Since several months my box needs a reboot after about 12-24 hours (I don't know the time exactly).
I've tried to track down the problem but I havent found the real issue.
After a "axe-dabug reset" everything works fine again. 
I'm using tvheadend 4.0.9 as my only client and I've also tried the current axe beta 14 firmware with no better result. I've also tried to use the older firmwares (09, 10, 11, 12) but without success.
Tvheadend is configured to scan on tuner 4 when idle and is allowed to fetch ota-epg data of tuner 2,3 and 4.
The only message I will get of tvheadend if the issue occures is:
`satip: SAT>IP DVB-S Tuner #4 - RTSP error -22 (Invalid argument) [6-400]`

perexg, 2016-07-02:
Could you provide log for '--trace httpc' in tvh ? https://tvheadend.org/projects/tvheadend/wiki/Traces (post a link here where the log is stored)

spadge4711, 2016-07-03:
I've activated the log, maybe tomorrow I will have more...

spadge4711, 2016-07-04:
[tvheadend_short_log.txt](https://github.com/perexg/satip-axe/files/346686/tvheadend_short_log.txt)

perexg, 2016-07-04:
OK, it's something with the minisatip. Error 400 means that the transport cannot be decoded or UDP sockets cannot be allocated. Try to run minisatip on the command line:

<pre>
   touch /tmp/nosatip
  killall -9 minisatip
  minisatip -llll <your_args>
</pre>

And upload the stdout when things does not work.

Another way is to try to use 'minisatip5'.

perexg, 2016-07-04:
OK. It seems that minisatip does not close /dev/axe/frontend devices correctly. I'll try to look into it. Minisatip5 seems fine.

perexg, 2016-07-04:
Could you try firmware from: http://s000.tinyupload.com/?file_id=76010383569268262995 ?

spadge4711, 2016-07-05:
Thank you very much, that was quick :)
I've installed the new firmware. I'll tell you in about 24 hours if everything is working now.

spadge4711, 2016-07-06:
'til now everthing works fine, no issues...
Thanks a lot!

spadge4711, 2016-07-06:
Everything's fine, no issues until now, seems to be fixed, so I'll close it for now :)

closed 2016-07-06 by spadge4711