perexg/satip-axe #74: RTSP error -22 with TVHeadend, unstable

netforceatg, 2016-10-17:
Hi Guys, thanks for a great project!
I have a Tvheadend server, running version 4.1-2280~g47af3cd. This works great with a Digital Devices Cine DVB-S2 6.5 card. And a local instance of oscam running for decrypting. 
Last week I added a Digibit R1 connected to a GT-Sat LNB, type GT-S3SCR4. I ran the stock firmware for a a few hours connected to Tvheadend and this worked. The Digibit R1 is connected with 1gbit to a Mikrotik CCR1009 router, that is connected to VMWare ESXI host(10Gbit) running Debian with TVHeadend on it.
I then upgraded the firmware of the Digibit to satip-axe project. However, I’m having big troubles with getting this to work stable. It seems I get the following error message that prevents it from working stable:
“ 
satip: SAT>IP DVB-S Tuner #3 (192.168.2.16) - RTSP error -22 (Invalid argument) [8-200]

“
My problem is that it works about 1/5 of the times, I may have to try to start a stream form Tvheadend 5-10 times before it works. With the Digital Devices DVB-S2 card it just works, but I am then limited to 2 tuners.
Over the last 3 days, I have tried changing almost all settings in the DVB input under TVheadend, and the 3 lastest version of of satip-axe(version 11, 12 and the current Alpha/Beta from july, version14), all with the same result. Tvheadend and their clients work, but extremely unstable.

A few notes: I have tried to enable RTP/AVP/TCP (embedded data), in tvheadend, same result
I have tried both minisatip and minisatip5 of the satip-axe firmware, same result.
Currently running the following satip-axe config option:
MINISATIP5_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"

Any Advice? I am lost in the dark so any help is much appreciated! 

Here are some links to a few config screenshots:
https://dl.dropboxusercontent.com/u/14615299/satip-axe-webint.png
https://dl.dropboxusercontent.com/u/14615299/tvheadend-2.png
https://dl.dropboxusercontent.com/u/14615299/tvheadend.png

netforceatg, 2016-10-17:
I just tried to enable the RTP/AVP/TCP on the satip device in tvheadend, and i then get the followin RTSP error:

2016-10-17 21:02:01.711 satip: SAT>IP DVB-S Tuner #3 (192.168.2.16) - RTSP error -90 (Message too long) [5-0]

perexg, 2016-10-17:
Could you keep UDP and provide '--trace httpc,satip' from tvheadend ? https://tvheadend.org/projects/tvheadend/wiki/Traces

netforceatg, 2016-10-17:
https://dl.dropboxusercontent.com/u/14615299/tvheadend-debuglog.txt

perexg, 2016-10-17:
It appears like a bug in minisatip (it seems that some multicast packets breaks the RTSP communication). Could you isolate the box from the other network (disable multicasts for the LAN port and use --satip_xml option? Your XML url is: http://192.168.2.16:8080/desc.xml

netforceatg, 2016-10-17:
Thanks! I have now isolated the Digibit in a seperat vlan, and forwarded this vlan to the ESXI Host that run the Debian TVHeadend. 

It seems to work, I have only tested a couple of minutes but it looks very good! :)

perexg, 2016-10-18:
Please, create a new issue for minisatip here: https://github.com/catalinii/minisatip

It's definitely a bug.

perexg, 2017-04-25:
Closing here.

closed 2017-04-25 by perexg