perexg/satip-axe #75: Unicable 2

estimadarocha, 2016-10-21:
I am having huge problem with unicable 2 setup:

this is the setup i have:

Inverto IDLU-32UL40-UNBOO-OPP with this frequencies:

CH1 1210MHz (EN50494+EN50607)
CH2 1420MHz (EN50494+EN50607)
CH3 1680MHz (EN50494+EN50607)
CH4 2040MHz (EN50494+EN50607)
CH5 984MHz (EN50494+EN50607)
CH6 1020MHz (EN50494+EN50607)
CH7 1056MHz (EN50494+EN50607)
CH8 1092MHz (EN50494+EN50607)
CH9 1128MHz (EN50607)
CH10 1164MHz (EN50607)
CH11 1256MHz (EN50607)
CH12 1292MHz (EN50607)
CH13 1328MHz (EN50607)
CH14 1364MHz (EN50607)
CH15 1458MHz (EN50607)
CH16 1494MHz (EN50607)
CH17 1530MHz (EN50607)
CH18 1566MHz (EN50607)
CH19 1602MHz (EN50607)
CH20 1638MHz (EN50607)
CH21 1716MHz (EN50607)
CH22 1752MHz (EN50607)
CH23 1788MHz (EN50607)
CH24 1824MHz (EN50607)
CH25 1860MHz (EN50607)
CH26 1896MHz (EN50607)
CH27 1932MHz (EN50607)
CH28 1968MHz (EN50607)
CH29 2004MHz (EN50607)
CH30 2076MHz (EN50607)
CH31 2112MHz (EN50607)
CH32 2148MHz (EN50607)

from this lnb comes a single cable connected to a unicable splitter and from that splitter comes 2 cables one to each digibit r1 with satipaxe.

i want both satipaxe box's to be working on my tvheadend but i am not able to do it.

i have tried all this arguments on satipaxe:

1st box ----> MINISATIP5_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 P 1" 
2nd box ---> MINISATIP5_OPTS="-u 0:5-1020,1:4-984,2:6-1056,3:7-1092 -P 1"

....

1st box ----> MINISATIP5_OPTS="-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 P 1" 
2nd box ---> MINISATIP5_OPTS="-u 0:4-984,1:5-1020,2:6-1056,3:7-1092 -P 1"

....

1st box ----> MINISATIP5_OPTS="-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 P 1" 
2nd box ---> MINISATIP5_OPTS="-u 4:4-984,5:5-1020,6:6-1056,7:7-1092 -P 1"

and all of them also with -j beside -u

All these result in failed scans and other strange stuff... any ideas?

estimadarocha, 2016-10-21:
@perexg your answers on tvheadend:

Issue #4031 has been updated by Jaroslav Kysela.
Also, Unicable II is 'jess' for minisatip.

This was already resolved: https://github.com/perexg/satip-axe/issues/61

Check if you have the coax cable from LNB in the first physical coax input in the box.

estimadarocha, 2016-10-21:
I have checked so far the coax cable and everything its fine.

I also have tested all the minisat arguments with -j like i said in the first post.

estimadarocha, 2016-10-25:
i was able to put it working... so the only remaining problem with unicable 2 is first tune fail. But this is probably a problem in minisatip.

Same problem here: https://github.com/catalinii/minisatip/issues/313

tunip, 2016-11-17:
The last commit in minisatip (4669f80) resolves this issue.

niceborn, 2016-11-17:
Will there be a new build with this changes?

estimadarocha, 2016-11-17:
Probably @perexg will make a new beta/alpha 14 with this commitments... he always with an eye on this improves. Let's give him some time. 

P.S.: @catalinii also make a great great job

niceborn, 2016-12-07:
Hi @perexg,
could you please be so kind to make a new beta version with the latest Unicable changes? My wife doesn't like the first tuning pause:-(

perexg, 2017-03-24:
Sorry for the very log delay, but unfortunately, I have really no time to update to the latest minisatip at the moment. I expect to release something new in two or three weeks.

estimadarocha, 2017-03-24:
@perexg nice to see you back... take your time... at least for me i will be waiting... please use the 0.7 branch if it's possible! and one more time thanks

perexg, 2017-04-25:
New build with minisatip7 (0.7.4) is available in https://github.com/perexg/satip-axe/issues/65 .

closed 2017-04-25 by perexg