perexg/satip-axe #76: amiko shd 8900

catalinii, 2016-10-27:
Hi,

I have an amiko alien shd 8900 which uses sh4 arch, I have managed to compile minisatip (it seems to use stapi version 4), but the DVB devices seems different from axe or normal dvb.

Do you guys have some type of documentation or some pointers where I can find any to try to implement the dvb part ?

Thanks

perexg, 2016-10-27:
A wild guess: https://github.com/OpenSH4/enigma2-openpli-fulan/tree/master/lib/dvb ? Look for the duckbox sh4 project.

catalinii, 2016-10-28:
Thanks for the info...

it seems that the enigma kernel used for the box exposes /dev/dvb/adapter0/frontend0 but the spark does not.

Probably I will migrate to enigma kernel as I do not have much info about stapi and maybe look at using the ioctls for csa ....

closed 2016-10-28 by catalinii