perexg/satip-axe #77: Help to compile

tnds82, 2016-11-19:
Can you help me to compile in Ubuntu or other system please?

Regards

CvH, 2016-11-19:
@tnds82 is https://github.com/perexg/satip-axe/blob/master/README.md not working ?

tnds82, 2016-11-19:
Ok. Can you help me ? I need to update minisatip binary to make some test

rucati, 2016-11-20:
Hi, in launchpad, frodo build minisatip for ubuntu, https://launchpad.net/~frodo-vdr/+archive/ubuntu/main

tnds82, 2016-11-20:
I need build minisatip for satip-axe with ARCH SH4

perexg, 2017-04-30:
@tnds82 : Follow requirements in https://github.com/perexg/satip-axe/blob/master/README.md . That's the build environment I'm using to build the whole firmware. You may expect little issues, but if you develop something in linux, it's easy to go through.

closed 2017-04-30 by perexg