perexg/satip-axe #79: First tuning fails (unicable)

ecdlguy, 2017-01-20:
Hi, 

first I'd like to thank everybody involved in making this firmware! It's much more stable than the official releases, even the "new" version 1.24.0.15(6/7). 

There's one issue I have: If all tuners are idle and powered down, the first tuning attempt will fail.
In the VDR-Plugin satip this results in "Has lock: no", and the webinterface of minisatip (Port 8080) shows are very low SNR (about 13%) and a quality of 0%.

I guess the reason for this is that the [unicable router](https://www.technisat.com/de_DE/TechniRouter-5-1x4/352-2850-10504/) I use obtains its power from the receiver. It seems to me that the router needs some time to be ready, but minisatip5 immediately sends the commands to the router.

Is there a way to fix this using the diseqc timing-parameters (-q Option)? According to "minisatip5 -h", 
the default values are 15-54-15-15-15-0, which of these parameters would I have to change?

For watching TV, this is not really an issue, since I can just switch to another channel. But for a programmed recording - the recording will always fail if no other recording is active. My current workaround is to add a second recording just before and with a little overlap to the "main recording" just to power up the unicable router.

cheers,
Thorsten

tunip, 2017-02-27:
The issue is fixed in minisatip (https://github.com/catalinii/minisatip/issues/313).
perexg needs to compile a new version of satip-axe with that fix.

ecdlguy, 2017-02-27:
Ok thanks for clarification. 
Are there probably any minisatip-binaries for the Inverto clones out there?

perexg, 2017-04-25:
It should be fixed in minisatip7 in the last beta build 14 - https://github.com/perexg/satip-axe/issues/65 .

reey, 2018-01-03:
@ecdlguy is this problem fixed for you?
I'm having the same problem you described and the same hardware.
Did you need to change the configuration somehow?

I'm using the latest build (201705251043-14) with minisatip7.
My config line looks like this at the moment: MINISATIP7_OPTS="-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 --nopm *"

reey, 2018-01-08:
@perexg   @catalinii
could you reopen this please? Or is this more of a problem with minisatip and reopen the issue there?

ecdlguy, 2018-01-08:
Hi,

This is my config line:

`MINISATIP7_OPTS="-q *:100-15-0-5-5-0 -u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -W 0 -O -H 5:25 -b 4042752 -B 10000"`

without the 100ms the first tuning still fails (build 14).

cheers,
Thorsten

reey, 2018-01-08:
@ecdlguy Thanks a lot, it mow works just fine :+1: 
May ask you, why you needed the other options?

Regards,
reey

ecdlguy, 2018-01-09:
The other options were suggested by @perexg here: https://github.com/perexg/satip-axe/issues/65#issuecomment-298005952 and here: https://github.com/perexg/satip-axe/issues/65#issuecomment-299106677

I have to admit that I just put them in my config without much thinking :-)

cheers,
Thorsten

reey, 2018-01-09:
@ecdlguy 
Maybe I will try those later.
@perexg 
Problem is solved for me, you can close this again.
Thanks

closed 2018-09-17 by perexg

ankorth, 2019-02-13:
Even this doesn't seems to be a satip-axe related issue in my opinion, I would like to confirm and clarify, that diseqc-timing (-q) parameter solved a similar issue running
- satip-axe 201810211549-15 on Telestar Digibit R1
- Tvheadend 4.2.6-7
- DUR-line UK 124 Unicable I/II LNB

The corresponding minisatip configuration in /etc/sysconfig/config is as follows:
```
#
# minisatip 0.8
#
MINISATIP8="yes"
MINISATIP8_OPTS="-q *:1500-54-15-15-15-0 -j 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -E"
```
where the -E switch was necessary to solve an issue where Tvheadend got fails on a lot of transponders during initial scan. Note the large delay of 1500. With a delay of 1000 it still doesn't work with this LNB. Thus, I can only repeat @perexg's recommendation to play around with it.

9000h, 2020-01-29:
did not compile
```
gcc -Wall -Wno-switch -ggdb -fPIC  -I. -I../../vdr-mcli-plugin/mcast/client -I../../vdr-mcli-plugin/mcast/common -L../../vdr-mcli-plugin/mcast/client -I/usr/include/libxml2 -DVERSION=\"1.0.2-4682fb5\"  -c adapter.c -o ../build/adapter.o
adapter.c: In function ‘signal_thread’:
adapter.c:2103:16: error: ‘SMutex {aka struct struct_mutex}’ has no member named ‘mutex_state’
   if (ad->mutex.mutex_state != 0)
                ^
Makefile:156: recipe for target '../build/adapter.o' failed
make[1]: *** [../build/adapter.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/satip/minisatip/minisatip/src'
Makefile:11: recipe for target 'minisatip' failed
make: *** [minisatip] Error 2
````