perexg/satip-axe #80: Not really an issue :)

jaarsa, 2017-01-27:
Hi!

I simply love this feature, and was wondering if there are any solutions like this on for CATV?
I have noticed that Elgato Netstream 4C is very like the Digibit in features, but use a different CPU.
Does anyone know if it is possible to make a CFW like satip-axe for this unit?

Here are some info regarding this item.
https://www.geniatech.eu/eyetv/fr/product/eyetv-netstream-4c/

Blog (console/ssh-access):
https://grunenberger.net/2013/03/12/netstream-elgato/
http://urup.net/?p=42
http://urup.net/?p=53

Sources:
https://www.elgato.com/en/elgato-open-source-projects

exxon82, 2017-01-31:
I'm actually looking for this too...

perexg, 2017-04-30:
I'm sorry, the platform is really different, ARM with less memory and the hw vendor has not provided the useable source code to build CFW.

closed 2017-04-30 by perexg

x-source, 2018-12-08:
I'm actually looking for this too...

geniatech - the new company behind the eyetv products offers the source code to build original fw
https://www.geniatech.eu/de/open-source-compliance-information/

maybe there is an easy way to customize the code
Sources:
https://opensource.geniatech.eu/Netstream4X/1.1.0.402r1/