perexg/satip-axe #81: TVHEADEND don't detect digibit r1 tunners

vuiperez, 2017-03-22:
Hello Perezg how are you?

i don't know waht happend but TVH stop detenting my tunners from digibit r1. 
i hadd the argument in the satip xml but gives this message .

2017-03-22 22:28:21.000 satip: Cannot get http://192.168.2.27:49000/satipdesc.xml: Connection refused
2017-03-22 22:37:21.253 satip: Triggered new server discovery

what can i do to get the TVH detect the digibit tunners?

best regards

perexg, 2017-03-23:
It does not look like a standard configuration. Check your router or other network equipment on the way. Could you access r1 from the local network?

vuiperez, 2017-03-24:
yes from the net work i can enter in the R1 page, and with you software i can enter in the cccam but the tuners stop shoing in the TVH.

i think that might be the R1 it self that is not working ok

perexg, 2017-03-24:
Port 49000 is not the default. Did you modify the web server port number for minisatip ?

vuiperez, 2017-03-24:
manage to detect with original software, but with satip network led don't light up and don't detect.
i remove the argiument line.

perexg, 2017-04-30:
I cannot help without a proper description.

closed 2017-04-30 by perexg