perexg/satip-axe #82: LO1/LO2 frequency

arrobar, 2017-04-11:
To see some channels in Spain I need to change the LO1 / LO2 frequency to 9750/9750 on my DIGIBIT r1.
Which modifier should I use?
I currently use MINISAT5_OPTS = "-L 0: 1,0:2,0:3"  

Regards

perexg, 2017-04-11:
Unfortunately, this is not supported with minisatip versions used in satip-axe. Pls, wait for minisatip 0.7.

arrobar, 2017-04-11:
Thanks for your quick response. I'll wait for the new version.

flotux2, 2017-04-11:
> Pls, wait for minisatip 0.7.
Does it mean we can expect further development of satip-axe in 2017? That would be really great!

perexg, 2017-04-11:
Yes, in few days. Actually, I'm finishing my other delayed jobs from my backlog and I expect to look into satip-axe / latest minisatip in next days.

perexg, 2017-04-25:
The build with minisatip7 (0.7.4) is available in https://github.com/perexg/satip-axe/issues/65 .

closed 2017-04-25 by perexg