perexg/satip-axe #85: minisatip 0.7 slower channel switching

MarhyCZ, 2017-06-11:
Hi,
its just an observation, rather than a huge issue, but after using build 14 after a couple of days, I must say that minisatip v0.3 is still faster when switching channels.

Setup is
tvheadend on Arch Server (no VM)
Client Kodi, LibreElec 8 on Apollo Lake board
All connected through ethernet of course.

With 0.7 its about 5sec
With 0.3 its 2sec

Otherwise, the minisatip 0.7 is very stable and thanks for a new build :)

perexg, 2017-06-15:
It's really a big difference, I don't see it in my setup. Could you gather '--trace satip,mpegts' from tvheadend (and upload it somewhere to net) ?

MarhyCZ, 2017-09-04:
Hi, I am extremely sorry sir, I've noticed the answer just now by accident. 
I have tried to switch again to 0.7 and actually didn't notice any slow down. So before taking logs I measured with slow-mo recording times again just to be sure and no difference. I am using 0.7 for over a day now and the performance and reliability seems to be same, if not better. So it must have been something else that slowed down the 0.7 when I tried it last time.
Again I am sorry and thanks for your time and work! :-)

perexg, 2017-09-04:
OK, thanks for your feedback.

closed 2017-09-04 by perexg