perexg/satip-axe #86: minisatip7 can´t tune some frequencies...

nickiman, 2017-06-14:
Hi,
I´ve written this post in the minisatip forum some time ago...
https://www.minisatip.org/forum/viewtopic.php?f=5&t=413
... but since it seems like it´s stopped. I´ve decided to write it here.

As I wrote in the other forum... check the link above for further details, it seems like I can´t tune some frequencies with my Digibit R1. However with my STB I can do it.

Any comment about that?

Thank you and regards.

perexg, 2017-06-15:
Your diseqc setup is really wrong (this is the line from the log you provided on the forum):

<pre>
[16/05 06:56:10.895]: Frequency 2223000 is not within range
</pre>

Please, describe your reception path and why you're using '-L *:9750-9750-12000'.

nickiman, 2017-07-05:
> Please, describe your reception path and why you're using '-L *:9750-9750-12000'.

Hi @perexg 
First of all my apologize for the late reply. **_Sorry, I don´t know what you mean with the reception path..._**
About the use of that configuration... was because was a recommendation in another forum. 

In fact, with that configuration I was able to tune more frequencies than with the default one. Here are the different logs that I get with the different configurations:
**Default configuration**: minisatip7 -f -l -l -l -7 0:1,0:2,0:3 -L *:9750-10600-11700
[https://pastebin.com/raw/H9FwezEH](https://pastebin.com/raw/H9FwezEH)

**Previous configuration**: minisatip7 -f -l -l -l -7 0:1,0:2,0:3 -L *:9750-9750-12000
[https://pastebin.com/raw/NjzUiigX](https://pastebin.com/raw/NjzUiigX)

This is the same output (or very similar) than this minisatip7 -f -l -l -l -7 0:1,0:2,0:3 -L *:9750-10600-12000
[https://pastebin.com/raw/ipmaMup0](https://pastebin.com/raw/ipmaMup0)

If you have visited the thread in my first post from the minisatip forum, there are some updates about this issue and cata have compiled a new minisatip version.

I´m wondering what should I do to try this version in my Digibit R1. Could you help me?.

Thanks a lot and keep in touch

perexg, 2017-07-05:
I mean what type of LNB you are using. Also, check your transponder parameters:

<pre>
freq=11739&sr=27500&msys=dvbs&mtype=qpsk&pol=v&fec=34&ro=0.35
</pre>

are they correct?

perexg, 2017-07-05:
Ok, reading the first comment in the forum:

> I live in a Community building so the Satellite arrives to my home only in polarization vertical low (V/L).

You should determine, how the frequencies are really mapped to your coax cable. If you cannot use 22khz switch (lo-band,hi-band), then you cannot probably use the full frequency range from the satellite.

You should probably study, how things do work. The LNB frequencies specified in -L are substracted from the transponder frequency, thus for standard config: 11739Mhz - 10600Mhz = 1139Mhz (tuner frequency) and 11739 > 11700, thus hi-band is activated (which probably does not work for your coax). Note that standard tuner range is 950Mhz-2150Mhz.

Let see the -L *:9750-9750-12000 config:

11739Mhz - 9750Mhz = 1989Mhz which should be correct (in tuner range)

buf for (log from forum):

<pre>
freq=11973&sr=27500&msys=dvbs&mtype=qpsk&pol=v&fec=34&ro=0.35
</pre>

11973Mhz - 9750Mhz = 2223Mhz which is OUT OF TUNER RANGE (see comment 2 in this issue for the confirmation)

So, again, you should gather info what's really on your cable.

catalinii, 2017-07-05:
Hi,

I have generated a new binary where the max frequency 2250MHZ, not sure if it will work but worths giving it a try.

Thanks

perexg, 2017-07-06:
The axe hardware uses STV6120 tuner which operates in frequency range 250 MHz to 2150 MHz . So we can only eventually use the lower (250Mhz..950Mhz) range to extend frequencies. But if the custom driver (from Inverto) restrict frequencies to standard 950Mhz..2150Mhz range, then we cannot do anything.

http://www.st.com/en/digital-set-top-box-ics/stv6120.html

nickiman, 2017-08-11:
Hi @perexg and @catalinii !

First of all my apologize for the delay to answer you. I´ve been very busy in my work and after that I´ve taked my vacations without the possibility to test anything and being outside my home.

I´m back at home again and I´ve started to try your satip version that you provided:

Below you can find the log trying to tune the frequency 11973V from tvheadend:
https://pastebin.com/raw/iitWvH5k

As I said before with my STB (sat receiver) connected in the same wall conector is able to tune that frequencies. I´ve taked some pictures of how is configured:
[http://i.imgur.com/xIa5UeA.jpg](url)

I can tune channels from 11973V and watch them without problems in my reciever:
[http://i.imgur.com/E7Q08Y6.jpg](url)

Even, I can reach to tune frequencies till 11856V with a good signal:
[http://i.imgur.com/ZEs2p5b.jpg](url)

Is there any kind of configuration in the satip to be able to recieve all the channels in the same way like a reciever is connected straight to a sat dish? 
Because in that way, all frequencies Low and High should be recieved, not?

Thank you very much!

perexg, 2017-08-11:
Look to description of universal LNB:

https://en.wikipedia.org/w/index.php?title=Low-noise_block_downconverter&section=9#Universal_LNB_.28.22Astra.22_LNB.29

As you see, everyhing *over* 11700Mhz is non-standard (11700Mhz - 9750Mhz = 1950Mhz) without the hi-band switch. AXE hardware has 2150Mhz limit, thus the maximal working frequency is 11900Mhz with the config '-L *:9750-9750-12000'.

I guess that your STB has higher tuner operational range than 250Mhz...2150Mhz (you may check the specifications). Unfortunately, it's not the case for the axe hardware.

nickiman, 2017-08-11:
> As you see, everyhing over 11700Mhz is non-standard (11700Mhz - 9750Mhz = 1950Mhz) without the hi-band switch. AXE hardware has 2150Mhz limit, thus the maximal working frequency is 11900Mhz with the config '-L *:9750-9750-12000'.
I guess that your STB has higher tuner operational range than 250Mhz...2150Mhz (you may check the specifications). Unfortunately, it's not the case for the axe hardware

Ok I´m understand now.
I was wondering how could workaround this... maybe if I take another cable in another input and change the settings in the config?

perexg, 2017-08-11:
If your frequency is above 2150Mhz in the coax cable, then you need a frequency downconverter (check google) to modify the frequency to be in the standard satellite tuner range. Anyway, this is really technically complicated and everything which is added to the coax adds some noise. I would really check, why the wiring in your home does not follow the Astra specifications. Perhaps, the firm who installed / maintain the wiring can make a solution for you. Ask for unicable for example.

nickiman, 2017-09-05:
Hi @perexg I´ve been reading about Unicable... My intention is try to get the maximum channels (I mean all polarizations VL/VH/HH/HL) so, probably I will install my self antenna in the rooftop with a Universal LNB and my own coax cable connected straight to the first input of the Telestar... 
Which should be the configuration for satip-axe 0.7 in the config file for that scenario?
Thank you very much!

perexg, 2017-09-05:
https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md#unicableunicable-ii

closed 2017-09-05 by perexg