perexg/satip-axe #88: Feature request: Add rsyslog

jlazkano, 2017-08-21:
Is it possible to add rsyslog to send all logs to a central server?

Regards.

madmartin, 2017-10-21:
Hi @jlazkano,
I am just visiting this project (not using it yet) - so I give only an advice: rsyslog would probably far too big for an embedded device. 
But the feature you request is already implemented in busybox. Maybe the busybox compile configuration must be changed to produce binaries like "syslogd" and "logger".  The syslogd binary can then send logs to a remote computer. Read more on 
https://busybox.net/downloads/BusyBox.html
Bye
Martin

jlazkano, 2017-10-30:
Thanks!

Is it possible to add it in the firmware compilation?

Kind regards.

perexg, 2017-10-30:
There's already syslogd binary (and it's being used). The only missing part is to allow the remote logging throught the config files.

jlazkano, 2017-10-30:
Thanks perexg,

Is it possible to add it?

Regards.

perexg, 2018-01-25:
Fixed in new build 15.

closed 2018-01-25 by perexg