perexg/satip-axe #90: Could USALS support be added to the R1?

elsmandino, 2017-10-08:
I was just wondering whether USALS could be added to the R1 via a software update or whether was impossible, due to hardware limitations.

perexg, 2017-10-08:
The rotors are not officially supported by the SAT>IP specification. The problem may be the "long" delay among position switches, so the SAT>IP clients should be aware that there's a rotor (or another slow tuning hardware).

But rotors are DiseqC controlled so there's no problem to use them with the axe hardware. We need only to create own SAT>IP protocol extensions for minisatip (and the clients should be modified, too).

elsmandino, 2017-10-08:
Thanks ever so much for the swift response.

I also read that the SAT>IP spec did not support rotors.

That is why I found it odd that although the Megasat website says that their SAT>IP server 3 only supports Diseqc 1.0:

https://www.megasat.tv/en/produkte/megasat-satip-server-3-2/

the instructions seem to suggest that USALS is fully supported

https://www.manualslib.com/manual/1208795/Megasat-Sat-Ip-Server-3.html?page=27#manual

which do you think is correct?

Do you think rotor support could be a real possibility for the R1 in the not too distant future?

I have only just found out about USALS and really want to add a second motorised dish to my setup - not immediately but certainly in the next 12 months.

My research has suggested that R1 really is the best all round SAT>IP device but my need for USALS might push me to get the Megasat 3 (if indeed it is supported).

If, however, as you say SAT>IP and USALS are currently fully supported, do you think I am better off sticking with internal TV tuner cards?

Thanks again for your help.

perexg, 2017-10-08:
I doubt that Megasat supports USALS (a quick glance to other images in the manual referring to the web interface does not show anything about USALS).

elsmandino, 2017-10-10:
Thank you - I have ended up ordering a Digibit R1, because of this marvellous bit of firmware you are working on.

One quick thing, further to the above - as the R1 supports Diseqc 1.0, could it be used in conjunction with one of these devices:

http://www.pulsat.com/products/Jaeger-SuperJack-V%252dBox-II.html