perexg/satip-axe #91: Fail more four channels for each transponedor

Configuracionpordefecto, 2017-10-27:
1 - I have tried to put more than 4 channels at a time for each transponder, but it gives me failure.

2- Then I can only put 16 channels simultaneously at the same time. Does this firmware support more than 16 channels at the same time?

PsychoRS, 2017-10-29:
No, este hardware no permite más de 16 canales simultaneos (4 canales por transponder X 4 transponders). No es una limitación del firmware, es limitación hardware.
----

No, this hardware doesn't allow more than 16 simultaneous streamings (4 streams per transponder X 4 transponders). It's the hardware limit, not firmware.

perexg, 2017-10-29:
There's hardware limit for 32 MPEG-TS PIDs. I think that minisatip does not switch to the unfiltered stream automatically, so you should handle this yourself (through tvheadend for example) or create the software filter code in minisatip.

Configuracionpordefecto, 2017-10-29:
if the limit is 32, then as I can only put open 4 channels per transponder and therefore 16 channels as much?

perexg, 2017-10-30:
Show the status page when you subscribe more than four channels (TCP port 8080 like http://satipboxip:8080).

closed 2018-01-08 by perexg