perexg/satip-axe #93: using 2 or 3 Inverto 400 in the network

oesip, 2017-11-05:
Hello, 
I am using 2 boxes with satip axe firmware since a while. So far ok, but after upgrading to release 14 I am facing some issuese with tuner release after stipping playback. Only reboot helps in that case. Thougt this is caused by using tvheadend development version, but now I have found your comment about minisatip version 0.7 option "-d". All 8 tuners have a single connection to the multiswitch, no diseqc is used, as far as I understand it. TVheadend (about 6 clients) detects both boxes and 8 tuners normally. Since the comment says "Warning" I am guessing, its better to set up correctly before searcing for other cause. But I dont have any Idea how... do you have any hint, how can I set unique adapter numbers with that parameter? Any help welcome or is that simply my misunderstanding?

thanks for a hint
oesi

perexg, 2017-11-05:
I don't understand completely your problem. What do you mean with "tuner release" ? The option should be probably -D (--device-id).

oesip, 2017-11-05:
Hmmm, sorry for not being clear enough ;-) what I wanted to say with tuner release is, that I can see that all 4 LED at the box are stay on after 2-3 days and then tvheadend says no tuner available. option "-D" I have used before, as you mentioned, but now the help of minisatip (https://github.com/catalinii/minisatip)  says this is: -D --disable-dvb disable DVB adapter detection ... so I have deleted it. I am struggling whats right or wrong, tvheadend detects the 2 boxes correctly anyway. I have now added "-W" and also "-q "*:5-15-0-5-5-0" -d "*:*1-0" even I am not sure if thats needed or not, just to see if there is some improvement.

oesip, 2017-11-11:
so far, behavior seems to be gone after pussing in the parameters as shown above. No glue why, but fine at least no more Tuner LEDs where on a long thie even no tuner is in use. you can close it this.

perexg, 2017-11-24:
Always use the help provided with the minisatip in satip-axe firmware:

<pre>
~ # minisatip7 --help
* -D --device-id DVC_ID: specify the device id (in case there are multiple SAT>IP servers
                         in the network)
 	* eg: -D 4 
</pre>

It seems that cacalinii changed silenty this argument :( I've seen some really not-so-nice lockups (network loop) because the device id selection in minisatip is not so robust with more devices (actually there are 4 SAT>IP servers in my network).

oesip, 2017-11-24:
you are right! I only checked cacalinii pages... but anyway running well now, no further issues with my two boxes during christmas I am planning to setup my 3rd box, but I am taking same configuration 
again great job and thank you for making that firmware. Running much better than the original one.

closed 2018-09-17 by perexg