perexg/satip-axe #94: Build 15

estimadarocha, 2017-11-17:
Any plans for Build 15 with updated minisatip?

indeed the 14 its working fine.

Thanks

ecdlguy, 2017-12-04:
Yeah this would be cool since it would stop the vdr satip-plugin from spamming my logfiles with entries like
`
user.err vdr: [1220] curl_easy_perform() [rtsp.c,314] failed: RTSP CSeq mismatch or invalid CSeq (85)
user.err vdr: [1220] SATIP-ERROR: Detected invalid status code 0: rtsp://192.168.2.30/ [device 0]
`

cheers,
Thorsten

perexg, 2018-01-24:
http://s000.tinyupload.com/?file_id=08555642022216225295 , there is minisatip8 binary (with modified 0.7.15), note that some options were removed (-M) or changed (-d to -D,-9 to -S)  or added (-2). Follow /etc/config.default MINISATIP8 lines.

https://github.com/perexg/satip-axe/blob/master/dist/README
https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md

estimadarocha, 2018-01-24:
what's that -2 for? can't find it anywhere!

thanks for the great job

perexg, 2018-01-25:
minisatip8 -h

perexg, 2018-01-25:
http://s000.tinyupload.com/?file_id=53142668595401844117 , fixes -2 , added SYSLOGD_OPTS to pass extra options to syslogd from config (issue #88), modified net.ipv4.tcp_wmem (issue #102).

ecdlguy, 2018-01-27:
Hi perexg,

thanks for build 15!
Unfortunately, the new ministatip8 is not playing well together with the VDR satip-plugin. After switching channels a few times, the sound and/or video does not start when switching to a new channel.
I don't know if it is related, here's what dmesg on the digibit r1 shows:
`
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
[dms-ts] stream stop ok
Unaligned userspace access in "minisatip8" pid=823 pc=0x00420748 ins=0x2212
[dms-ts] stream stop ok
Unaligned userspace access in "minisatip8" pid=823 pc=0x00420748 ins=0x2212
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
Unaligned userspace access in "minisatip8" pid=823 pc=0x00420748 ins=0x2212
[dms-ts] stream stop ok
[pll_wait_for_lock] pll locking timeout
Unaligned userspace access in "minisatip8" pid=823 pc=0x00420748 ins=0x2212
[dms-ts] stream stop ok
Unaligned userspace access in "minisatip8" pid=823 pc=0x00420748 ins=0x2212
[dms-ts] stream stop ok
Unaligned userspace access in "minisatip8" pid=823 pc=0x00420748 ins=0x2212
`

Using minisatip7 I don't have the issues with lack of sound and/or video and dmes is quiet.

cheers,
Thorsten

perexg, 2018-01-30:
http://s000.tinyupload.com/?file_id=66895215405789335534 - fix for the unaligned access .

ecdlguy, 2018-01-31:
Hi perexg,

thanks. I've tested the new version, dmesg is quiet now.

However, it's still close to impossible to use minisatip8 with the VDR satip-plugin. After 5-10 minutes of channel switching, the video looks like slow motion and there's no audio. A few channel switches later there's no video on any channel. Even stopping vdr and give the receiver some time to power down the frontends does not help.

How can I help debugging this?

cheers,
Thorsten

lonerika, 2018-02-01:
Hello,
I tested the 20180125 ... This runs flawlessly.
Then I tested the 20180130 .... Picture freezes temporarily.
That's why I have the 20180125 .. played back.

By the way, thank you for this project.

greeting
Google translator.

perexg, 2018-02-08:
@ecdlguy : If you see VDR issues, does the http streaming work ok?
@lonerika : There are no other changes in 20180130 than fix for the signal routines. The behaviour should be identical.

I'm running minisatip8 here for 8 days and I've not noticed any issues (tvheadend backend).

ecdlguy, 2018-02-08:
@perexg Ok I'll try. However, with minisatip7 the problems described above aren't there, so something must have changed.
Is there a way to debug what's going on using logfiles? How can I enable them on satip-axe? Maybe there's something in the logs that helps.

perexg, 2018-02-08:
@ecdlguy : Create a new issue for this.  The logs can be gathered by creating /tmp/nosatip file (touch), kill the minisatip8 and run minisatip8 remotely from your ssh machine with '-lll', like:

<pre>
ssh -t root@${HOST} minisatip8 -f -lll $ARGS | tee out.txt
</pre>

The logs will be printed to terminal and saved to out.txt .

lonerika, 2018-02-09:
Hello perexg,

True, the mistake was mine. -Sorry -
Have the USB flash drive with the 20180125 .... the config set up.
On the stick with the 20180130 ... just not. Since minisat7 ran with wrong settings. - shame -
The new version was so stable in the last 5 days that I flashed it.

Excuse me again and good work - thank you -

Google translator

meijkl, 2018-02-09:
Hi perexg,
up to now I can only confirm what Ionerika wrote - 20180130 is stable and works well with the VDR satip plugin (version 224 and 230) and better on my Grundig GSS 400 than the May-2017 version (there I had freezes after 1-2 days which required a reboot).
Thanks
Klaus

P.S.: By the way - I tried to compile satip-axe myself but failed when trying on my Ubuntu Xenial 64bit installation. Do have a recommendation which environment I should use? Many thanks!

hanzoh, 2018-02-14:
Hi perexg,
20180130 is working fine for 2 weeks now.
Would it be possible to include this patch?
https://github.com/catalinii/minisatip/pull/408
This enables direct access to minisatip from Plex DVR.

perexg, 2018-02-21:
http://s000.tinyupload.com/index.php?file_id=01165966614935664637
- updated to latest minisatip code (plex should work now)
- fixed the free tuner problem for complex diseqc settings where one input is used by multiple tuners (minisatip7 and minisatip8)
- increased net.core.wmem_max=12582912 , it might help to prevent the UDP packet loss

meijkl, 2018-02-22:
Great! After using it for about 10min I can already say that it makes a great difference!

It already worked well before but now all CC errors are at zero after 10min. They never were before!

[image]

Thanks a lot!
Klaus

Trujulu, 2018-02-22:
I think port 1 (0) is little bug when using following config:

MINISATIP8_OPTS="-7 0:1,0:2,0:3 -L 0:9750-10600-11850,1:9750-10600-11850,2:9750-10600-11850,3:9750-10600-11850"

[image]

[image]

Trujulu, 2018-02-23:
Minisatip7 works good. Not version 8...

lonerika, 2018-02-28:
Will there be new build only via TinyUpload?
Since I use normal upgrade-fw? What accesses github.
greeting

perexg, 2018-03-08:
I need to check the linked input feature before I'll create the final 15 build. You can copy .fw file from the tinyupload tarball to /root directory in the satip box (scp) and run 'upgrade-fw satip-axe-201802211836-15.fw' for example.

lonerika, 2018-03-12:
Thank you,
with the v20180221 I have no more problems since I added the parameter -O. 
In the TVHeadend 4.3.xxx also converted to TCP.

Greetings and thanks for the FW

8carlos88, 2018-05-19:
hi perexg

buy digibit r1 and update satip-axe-201802211836-15.fw in usb and I did not scan for the first time, I read several post where this failure occurred, it happens to me:
https://github.com/perexg/satip-axe/issues/79
I also read this and all those related to lnb unicable does not work and some more ...
https://github.com/perexg/satip-axe/issues/95

say that the digibit r1 and with the firm of factory goes very well ....
 
could help me and sorry for my english i am spanish ....
tell you that I am very new.

use MINISATIP7 / 8_OPTS = "- u 0: 0-1210,0: 1-1420,0: 2-1680,0: 3-2040" and
MINISATIP7 / 8_OPTS = "- j 0: 0-1210,0: 1-1420,0: 2-1680,0: 3-2040"

I think that the firm MINISATIP7 / 8 does not recognize me

greetings, thanks

buckyball, 2018-09-15:
Seems like the tinyupload links are expired, is there any chance to get the last v15 release (satip-axe-201802211836-15.fw?)

lonerika, 2018-09-15:
have upload for you here https://github.com/lonerika/please-delete. I hope can help you.
Account accidentally originated. Don't know how to delete it.
Your firmware I last line.

buckyball, 2018-09-15:
@lonerika thank you!

lonerika, 2018-09-15:
Again and again gladly. Some things I archive like tvheadend, coreelec as well as satip-axe
greeting

So I deleted the repo

perexg, 2018-09-17:
I will release 15 probably in this week when I will debug the linked tuners problem.

perexg, 2018-09-17:
http://s000.tinyupload.com/?file_id=89599119253627720729 : Updated to latest minisatip (minisatip8), tested quickly the linked inputs (no issues found or I don't understand @Trujulu problems - report this in a new issue). The DNS problem for ntpd when dhcp is used for the address configuration should be resolved, too.

okurz, 2018-09-18:
Just a little hint: You can also use the "github releases" to publish binary files on your project

estimadarocha, 2018-09-24:
wtf?

perexg, 2018-09-24:
- coax1 - multiswitch 2x commited (23.5E, 0.8W)
- coax2 - like coax2
- coax3 - 10x uncommited (none, 28.2E, 19.2E, 16E, 13E, 9E, 4.9E, none, none, 15W)
- coax4 - like coax3

estimadarocha, 2018-09-24:
thanks for clarify....

shoudn't coax2 - like coax2 be coax2 - like coax1?

perexg, 2018-09-24:
yes, fixed

perexg, 2018-09-24:
rc4: https://github.com/perexg/satip-axe/releases/tag/build15-rc4 (a quick fix for the polarization issue)

perexg, 2018-09-25:
The '-S' comment was wrong. I removed it. I fixed '-9' in rc4. The '-9' allows to select another coax input to create unicable groups (for example, you have four SCR LNBs and two of them are on one coax cable and two other on another coax cable, so you connect two coax cables to the box and specify which tuner is assigned to a coax - see the decription bellow in minisatip.md).

estimadarocha, 2018-09-25:
so since i am just using one cable from one SCR LNB i must stick with -u, correct?

other thing that i always don't understand is this:

example 2: unicable: MINISATIP8_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"

compared with mine,

-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040

does this mean anything... should i really follow the example?

perexg, 2018-10-12:
DVB-S2 multistream: https://github.com/perexg/satip-axe/releases/tag/build15-rc5 , issue #126 .

perexg, 2018-10-12:
PLS ROOT scrambling scheme is supported, too: https://github.com/perexg/satip-axe/releases/tag/build15-rc6 , issue #126

perexg, 2018-10-17:
Build 15 is finally out with the next VDR fix. I am really happy about all the small changes and updated minisatip(8) which seems very stable in my complex environment. Enjoy. The build 16 can be discussed in issue #127 , but so far, I have no fixed plans for next months (until I or someone else find a bug).

closed 2018-10-17 by perexg

estimadarocha, 2018-10-17:
does the upgrade-fw function directly executed on satipaxe will list the github releases?

or we still need to upload and run manually?

perexg, 2018-10-17:
Good point. It should be fixed now.