perexg/satip-axe #95: Megasat Unicable LNB

gisbertg, 2017-11-24:
I have a new Megasat Unicable 2 LNB Diavolo. But i don't understand the logic in config. The cable (only one) from LNB goes into input 0. (the right one) 

My config minisatip 0.5 (why is there 0.5 and not 0.7) I have Version 14!
MINISATIP_OPTS="-u 0:0-1210,0:1-1420,0:2-1680,0:3-2040"
I can only use Tuner 1 in TVheadend. Why?

LNB Frequency's
Ch01 1210MHz
Ch02 1420MHz
Ch03 1680MHz
Ch04 2040MHz

perexg, 2017-11-24:
> My config minisatip 0.5 (why is there 0.5 and not 0.7) I have Version 14!

If you upgraded from previous versions of satip-axe, you must modify your config in _**/etc/sysconfig/config**_ yourself. The default config file contents is in _**/etc/config.default**_ .

> I can only use Tuner 1 in TVheadend. Why?

What's on the status page for the satip box (IP:8080) when TVH is trying to use other tuners?

gisbertg, 2017-11-24:
Now i'm on minisatip 0.7. First Tune fails, second one is ok. But when i'm would use more tuners, it fails.
[image]

perexg, 2017-11-24:
Use "-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040" config. The first argument is adapter, the second is unicable slot/channel (refer to the specification). Unicable input is expected on first coax connector by default. You can modify this settings using '-9' argument (minisatip 0.7).

closed 2017-11-24 by perexg

malvinas2, 2017-11-29:
Excuse me, but I have to re-open this issue... 

In one of the configuration examples given by you, you are using: 
`MINISATIP_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"`

Why don't you use 
`MINISATIP_OPTS="-u 0:0-1210,1:1-1420,2:2-1680,3:3-2040"` (which indeed is working too)

IMHO the assignment between adapter (number before colon) and slot/channel (number after colon) could be be chosen arbitrarily, is this right?

perexg, 2017-11-29:
Yes, the example is an example and the purpose of above example is to show that you can map the used slots/channels to tuners as you like.

In my example:

tuner #0 --> slot1/channel1 on frequency 1420Mhz
tuner #1 --> slot0/channel0 on frequency 1210Mhz

Your 'standard' mapping:

tuner #0 --> slot0/channel0 on frequency 1210Mhz
tuner #1 --> slot1/channel1 on frequency 1420Mhz