perexg/satip-axe #96: didbit-r1 - Is the internal nand flash worn out? after 3 years?

dreamcat4, 2017-11-24:
Hi there,
My digibit r1 satip box:

* 3 years old
* has stopped comming up on network DHCP (nothing else changed)
* most of its lifetime has been on satip axe version 11

I booted axe r14 off USB drive and:

* it picked up my internal flash settings
* ssh in, could touch and delete a small file
* then tried the 'update-fw' program, to reflash the internal with r14

However the internet firmware update (unfortunately) resulted in many errors:

https://gist.github.com/dreamcat4/b7965cf9e6b5dc4d14ae09e8a0c10c03

* Is it because my internal nand flash has worn out?
* I can still use axe 14 booting from USB drive yes?
* Is it worth to try reflash the factory official firmware recovery?

perexg, 2017-11-24:
The flash errors are somewhat expected (just press Enter) the second or third attempt will succeed.

Verify only /mnt/data partition, especially /mnt/data/satip-axe/sysconfig where the configuration is stored. Check 'dmesg' if you see jffs errors..

dreamcat4, 2017-11-24:
Thanks @perexg indeed you were correct - the 3rd attempt it worked. I have updated my gist file with also the output from `dmesg`. It shows a lot of nand error corrections

https://gist.github.com/dreamcat4/b7965cf9e6b5dc4d14ae09e8a0c10c03

* Next i will try rebooting (without USB key).
* Have already backed up my `/sysconfig` folder today (before trying this).

[EDIT]

* i typed 'reboot'
* the usb key was still in the device
* did not come back online
* night rider lights just kept flashing

* turned off device
* removed USB key
* turned on device
* did not boot, only power light (no ethernet light or other lights on front)
* turned off device
* replaced USB key
* night rider lights came back, however no still ping

(now I shall try rebooting again the device a few more times with the USB key)

maybe I can also try to replace the nand chips on the PCB with new ones? they are SMD yes?

dreamcat4, 2017-11-24:
Ah, it seems the device no longer boots. The night rider lights keep going (for longer than normal), then all 4 lights briefly flash together. Then nothing and no ping, etc...

dreamcat4, 2017-11-24:
OK I have found the issue now. It appears to be an unreliable power supply / ac adaptor. The thing is making a bad noise. Tried with different 12v source and it boots fine / ok no problems. And without the USB key too.

Closing this issue.

closed 2017-11-24 by dreamcat4