perexg/satip-axe #97: Not sure whats happening here, not detected on multicast, doesnt tune anymore

dreamcat4, 2017-12-07:
Hi there,

* Replaced bad PSU on my unit and that has stopped the continuus reboot loop
* Upgraded from 12 --> 14 and then back to 12 (no different)
* Can login over ssh
* Device isnt detected on multicast
* See no obvious errors in dmesg
* Device was works briefly for a 1st tune after reboot
* Then stops responding anymore, after closing 1st stream, or by itself pretty soon, always within 10 minutes
* After reboot device appears again (briefly)
* The thing is multicast on the router is OK setting (not changed, same as was before any issue)
* Have tried with multiple different sat>ip client apps.

Of course you guys were saying 14 is fine / good now. So anyway I'm not sure what to do next. This is weird behaviour.

How to debug further?

* I have opened the box up and do not see any obvious bad signs, or any dust built up
* The unit is in attic right now its winter time and pretty cool up there

Many thanks for any advice.

perexg, 2017-12-15:
Could you login to the box when the multicasts does not work? Is the whole network down?

dreamcat4, 2017-12-16:
Thanks for responding @perexg. Yes I can. ssh is always working. Even when the multicast is down. I can still login check up on any processes and run commands. Its just that I am not sure what to look for.

perexg, 2017-12-16:
Put contents from 'dmesg', 'logread', 'ps ax' to http://pastebin.com (or somewhere else - public - to check).

Also try to set the SAT>IP server number (argument '-D 88' for minisatip).

dreamcat4, 2017-12-16:
Hi @perexg,
Pastie link is in middle of my long comment here:

Since reporting this issue 9 days ago. I try to reproduce the same issue today for this but it was OK. Yet the issue happened many times before. And I have changed the power supply twice. Since getting this newest 2nd PSU from ebay a few days ago, which is higher quality than the 1st ebay junk we had a week ago.

For today we tried for a longer time but it did not dissapear from the network (satip on multicast). We could not break it to reproduce the problem. Everything seems working fine. And it stayed working for longer than ever before, since we first saw this issue. Which was somewhat frustrating now, but OTOH it might continue to be fine now! Which is good yes?

I checking the FW version, we had rolled back to r12 and its still stay on r12. Which also helps to know.

Anyhow then it occured to me that all the other times I kept doing 1 thing which was so immediately ssh back into the box, after each fresh reboot. So this working today, we were not logged in over ssh too while it was going on.

While logged in I didn't do much load, except for run these few cmds:

```sh
df
ps
dmesg
ls
ps
```

And then re-launch the elgato SAT>IP app in iOS (quit, open). Then its 'no device found'. Haha. So maybe we will try no ssh'd in for longer periods. And see if that helps it.

Then I am typing this to you and the ssh session is still remain open. The thing starts working again. This must be a few minutes more since the SAT>IP signal stopped / client lost connection.

So at this time I take the cmd output to pastie here ----> https://gist.github.com/dreamcat4/d70f736e2d218cc9e858bfbcf6c8a257

That is both broken and somehow came back / worked again.

Also thanks for the advice about setting the server number '-D 88' flag too. We have not tried that yet. We will continue to monitor the situation. Your help was very much appreciated.

Next time I reboot it, we will not log in over ssh. Another point is that I was checking the `cron` jobs to make sure they were not happening at these times. They were not but I also noticed that I have the box set to reboot automatically every 12 hours. And when we randomly come back to check if its working. Sometimes its missing. Sometimes its OK. Maybe the thing does come back by itself from time to time. Then go away again.

So I am now considering about the idea of just checking it 1x every hour during the entire day tomorrow. Or every 30 mins. And recording the results to paper. Without doing any rebooting or ssh...

If we had tvheadend working then we could just see in the tvheadend logs instead. But we have not tvheadend server right now.

If you cannot see anything more to do here. (no other suggestions / nothing in log). Then I am OK and happy to close this issue. Thank you for the help.

closed 2018-01-08 by perexg