perexg/satip-axe #98: Issue when connecting via enigma2 satip plugin

cobanos, 2017-12-09:
Hi perex,
I am using satip-axe on an Inverto 400  box in conjunction with Tvheadend (-O option via TCP enabled). In fact the original firmware running via UDP led to lot's of continuity counter errors making it impossible to watch. That's why I have switched from the origiinal firmware to satip-axe. With the TCP option the picture is perfect.

I have recently tried to connect from an enigma 2 receiver using the e2 satip plugin. I can watch a channel for about 5 -10 seconds and then get a lots of artefacts, making it impossible to see anything. Running the original firmware does not give those artefacts and the picture is great. Now I have the issue that with the original firmware I can watch TV perfectly via enigma2 and not via Tvheadend. With the satip-axe firmware I can watch perfectly via Tvheadend, but not via enigma2 ;-)

Any ideas what I do to get sati-axe running with the eniga2 satip-plugin?

perexg, 2017-12-15:
UDP is not reliable. You must have perfectly tuned all components in your ethernet network otherwise you can see the UDP packet loss.

Try to decrease buffering limit argument '-b' like '-b 7896:5775360' for enigma2, but it increases the CPU usage on the box.

cobanos, 2017-12-19:
thanks for the hint but unfortunately the issue is still present when connecting via enigma2. The picture is fine for the first 5 seconds and then full of artefacts. 

I would assert that I've a perfectly tuned network. The inverto box is connected to a gbit switch with a 10" cable and on that same switch I've a NAS attched with two 10" cables as well ( link aggregation bond) where TVH is running on. UDP is still a mess resulting in those continuity counter errors in TVH. 
Interestingly in enigna2 there are no continuity counter errors when connecting via UDP (and the enigma2 receiver is two floors away from the network switch)

catalinii, 2018-01-03:
Hi @cobanos,

try creating an issue on the satip client of enigma to support TCP protocol as well. The client is very limited (no ability to have multiple satellites from the same server as well) and this can help a lot with the experience that the users have with enigma.

Thanks

cobanos, 2018-01-16:
Thanks catalinii - will do. Do you know who is the author of that plugin or where I can refer to in order to request such enhancement? I've checked the source code but did't see any references to an author of the plugin.

perexg, 2018-09-17:
OK, because of my own requirement, I added the RTP/TCP/AVP transport to the satip plugin and it was accepted upstream: https://github.com/oe-alliance/satip-client/commit/69b4bfd827fbf0fa3c482bc82440d1d36538bdd1

Because tuners on my VU+ solo2 broke for a reason, I am using enigma2 satip client through the TCP data mode with the R1 box and it works perfectly.

If you need to compile the latest satipclient for your box for a test, tell me your box type and I can try prepare ipkg or just wait until the change is accepted to your firmware.

perexg, 2018-10-17:
Closing, the TCP data transfer mode code is in upstream and I use it really everyday without issues.

closed 2018-10-17 by perexg