perexg/satip-axe #99: Unicable & JESS DiSEqC programmer

lars18th, 2017-12-12:
Hi,

_I’m not sure if this is a right place for posting this. However, I see that here is a lot of great people with great knowledge. So, I write here if this project can support this idea, or someone likes to promote it on other place._

This is the question: Programmable Unicable products are great for reduce the investment in multisat installations. For example, with two Unicable LNBs and dishes you only need a simple combiner for use them with up to 32 receivers. The key is program one LNB as Pos-A, and the other as Pos-B. Furthermore, with static distribution you can distribute to any number of users up to 32 transponders, from any sat position (you only need one Unicable LNB for each sat position). The key in this case is program each LNB with the static frequencies. Again, it just takes only simple passive combiners for mix the signal from the different LNBs. You save the cost of any switch.

However, for programming an Unicable device you need the programmer. And here is my suggestion: add support (or create a tool) for do it! And how you can do it? Let me tell you what I found.

A time, you can found almost three different Unicable manufacturers with programmable products and programmers:

- Inverto.
- Dur-Line. 
- Axing.

I installed the software of all, and I see that the software it’s mainly the same. So, some type of compatibility can exist. Furthermore, I try to configure from start some device with one tool and load the file from another version. Sometimes, the configuration can be loaded. In addition, the configuration can be saved in three different formats:

- Binary (.bin)
- Text (.txt): like a INI file very easy to understand.
- DiSEqC (.dsq): this seems to be a sequence of DiSEqC commands.

Attached you can found a ZIP with default devices configured with each tool in the three formats.

In a nutshell, **the programmer seems to be a mere DiSEqC sender/receiver with RS232 interface** (managed over an standard FTTI USB interface). Then my idea is this: support the option for "sending" the '.dsq' file to the device for program it. Obviously you lost the option for reading the configuration from the device. But, if you create your configuration for the beginning, then you don’t need it at all.

Please, comment about this. And if you feel that this project can send this '.dsq' file to the device, please add this option to it.

Regards.

[unicable-programmer.zip](https://github.com/perexg/satip-axe/files/1551291/unicable-programmer.zip)

pbriesch, 2017-12-12:
I support this idea!

lars18th, 2018-01-11:
Hi,

No one more interested on this?

> Then my idea is this: support the option for "sending" the '.dsq' file to the device for program it.

My request is to add this function.

Huevos, 2018-02-13:
Yes, the idea is very interesting but not sure how to help.

lars18th, 2018-02-13:
> Yes, the idea is very interesting but not sure how to help.

So, as _some_ people it's interested we need to organize. We need:

- People with **hardware** unicable devices. Mainly LNBs and/or Switches with **Programming Support**.
- People with some **programming** skills, focused on DiSEqC commands.
- People with **physical** programmers SCR for testing the tools.

Please, post here if you can contribute! :smile:

charlesduco, 2018-02-13:
I found more information about this in the manual of a professional satellite meter VAROS 109, page 33: https://www.kws-electronic.com/downloads/KWS-Electronic-Manual-VAROS-109-V01-10-en_171027.pdf

> DiSEqC Script
Newer multiswitches for single-cable systems are programmable by DiSEqC-signals (like JULTEC
a2CSS series). Usually manufacturers provide special program adaptors and PC-software tools. But
this programming is also possible by this measuring receiver with the DiSEqC-Script function. **A set
of DiSEqC commands** according EN 50494/50607 can be easily edited in a common texteditor end
**saved with the ending .dsq.** Appropriate files can you get from the manufacturer (e.g. JULTEC).
These files must be stored on an USB-stick which can be connected to the measuring receiver. **The
script function sends the DiSEqC commands line by line over RF-connector to the DiSEqC-Unit.**

lars18th, 2018-02-14:
> I found more information about this in the manual of a professional satellite meter VAROS 109, page 33: 

Great info! This confirms that we can use the ".dsq" files created by the tools described in my first post. So, we only need to create a tool that can function as a "DiSEqC scripter". That is send the commands one-by-one and wait for the confirmation.

Someone has the skills for do it?
Anyone has some hardware to test it?
Can this project include this tool?

perexg, 2018-02-14:
For DiseqC, the axe driver behaves like any other DVB-S/S2 hardware in Linux, so if anyone develop this tools using Linux's dvbapi, it can be included to this axe firmware, too. And sorry - I don't plan to participate on this (serious lack of time). It might be a good exam for a student :-)

lars18th, 2018-02-14:
> For DiseqC, the axe driver behaves like any other DVB-S/S2 hardware in Linux, so if anyone develop this tool using Linux's dvbapi

Hi @perexg ! This is the reason to post here. This project has the basics for this tool. And the tool can use the current code for sending DiSEqC commands.

> And sorry - I don't plan to participate on this (serious lack of time)

Ok. Don't worry! However, can you point (isolate) where it's the code for sending the DiSEqC commands? Perhaps we can start with a simple skeleton of the tool.

perexg, 2018-02-14:
> Ok. Don't worry! However, can you point (isolate) where it's the code for sending the DiSEqC commands? Perhaps we can start with a simple skeleton of the tool.

Use any existing code for Linux. Example: https://github.com/linuxstb/dvbtools/blob/master/dvbtune/diseqc.c

I would make this tool standalone. Something like 'senddsq <file.dsq> <frontend_device>'.

lars18th, 2018-02-14:
Thank you, @perexg !

> I would make this tool standalone. Something like `dvbsenddsq <file.dsq> <frontend_device>`.

Great idea!

Huevos, 2018-02-14:
Should be able to make this as a python plugin for an enigma2 satellite receiver.

lars18th, 2018-02-14:
> Should be able to make this as a python plugin for an enigma2 satellite receiver.

I prefer to develop this tool (aka "dvbsenddsq") as a C portable program.

Huevos, 2018-02-14:
Ok, but already in enigma2 CPP code there is code to send DiSEqC control words.

lars18th, 2018-02-14:
>  in enigma2 CPP code there is code to send DiSEqC control words.

Standalone, or inside the STB code?

Huevos, 2018-02-15:
Inside the STB code.

akosinov, 2018-02-22:
Hi! I'm already tried to do this. Sorry guys, it's impossible w. our dvb cards and dvbapi due to limited length diseqc fifo buffers onboard. both my cards (w. frontends stb0899 and tas2001) has a HARDWARE diseqc sequence length limit of 8 bytes, but for unicable programming purposes we need at least 16. Plus software limitations of dvbapi (6 bytes) exists, but can be changed in minutes.
Now i looking to microcontroller (stm32) + diseqc chip interface as solution.

lars18th, 2018-02-22:
Hi @akosinov ,

> it's impossible w. our dvb cards and dvbapi due to limited length diseqc fifo buffers onboard.

I'm disappointed to hear that! 😞 

> Now i looking to microcontroller (stm32) + diseqc chip interface as solution.

And what about search for "other" devices, not regular dvb cards?
And how we can check this limit of 8 bytes in the spec of the cards? Perhaps we can see other frontends without this limit.
In any case, in the way of hardware development I recommend to use a Raspberry Pi (easy to get it) and a diseqc custom interface. What you think?

akosinov, 2018-02-22:
Just one option is exists: if unicable device not worried about length of codepart (in all dsq files it equal to 14 w. preamble 7F 10) we can try to split target binary to small portions (for example 7F 10 xx xx xx xx) and send 4 nor 14 bytes at one diseqc operation. Maybe it will be ok, i cant' test due to original programmer (reader) unavailable.

akosinov, 2018-02-22:
oh no,.... no Pi please..... ))))
will do on stm32( already has stm32eval boards) , but still can't find simple diseqc interface w. minimum passive elements (hate it). seraching.......

lars18th, 2018-02-22:
Hi @akosinov ,

Great to hear you're searching!

Regarding the Pi, it's cheap and very easy to purchase. Why not?

And regarding the splitting of the commands, I feel it will difficult that it works. In any case we need to do some tests. Any one has a "programmable" device (not the programmer" but a LNB o Switch)? We can try to send commands to it.

Regards.

lars18th, 2018-02-22:
Hi @akosinov ,

I'm searching here http://github.com/torvalds/linux/tree/master/drivers/media/dvb-frontends about what DVB-S/S2 frontends can support more than 6-8Bytes DiSeqC commands. Any idea what I need to look for?

akosinov, 2018-02-22:
I has 2 unicable II devices (switch and lnb from inverto). Can try, but i has no options for check result. i need to read config from device, but don't know how. programming protocol is not a part of standard, developed by unicable chip supplier (maxlinear) and completly closed.

akosinov, 2018-02-22:
about frontends - only "manual" checking. find 'master_cmd' and check function algo

akosinov, 2018-02-22:
if somebody "present" me original inverto programmer we can trace diseqc exchange and reverse protocol ))))

lars18th, 2018-02-22:
Hi @akosinov ,

Here an idea for "checking" if your "custom programmer" works:

- Use your unicable II LNB in a working install (check that it works with your STB unicable compatible).
- Generate a new config with the tool from the manufacturer and generate the corresponding ".dis" file.
- In this config, change the LNB to B (or C, or D).
- Reconfigure your STB for selecting the LNB B... It works? Then your "programmer" works! If not, then the programming has failed.

You like to test it? 😄

akosinov, 2018-02-22:
Hah. i thinked about this, but has no worked installation. only week ago received unicable equipment w/o programmer and think i can prog it self. aha..... now trying to solve it)

lars18th, 2018-02-22:
> about frontends - only "manual" checking. find 'master_cmd' and check function algo

I'll try! Thank you!

> if somebody "present" me original inverto programmer we can trace diseqc exchange and reverse protocol ))))

This will be the "second" task after we have a "custom programmer" that works (we need to resolve the problem of the DiSeqC command length, the source code of the tool, etc.).
For sure, if we obtain some results, then we can improve it! 😄

akosinov, 2018-02-22:
your testing method only partialy ok, because u need check ALL parameters saved correctly. cheching only 'LNB address' or some one frequency changement not guaranty 100% correct work.

lars18th, 2018-02-22:
Hi @akosinov ,

> Hah. i thinked about this, but has no worked installation. only week ago received unicable equipment w/o programmer and think i can prog it self. aha..... now trying to solve it)

If you have an Unicable II Inverto LNB it is _pre-programmed_ to a default configuration. This default config has _no LNB address_ and _dynamic mode_. So any unicable compatible STB can work without using a programmer. You can check this default configuration using the tool from Inverto and selecting the device you have. The loaded config it's the default config. Use this to see all the values of the default config.

> your testing method only partialy ok, because u need check ALL parameters saved correctly. cheching only 'LNB address' or some one frequency changement not guaranty 100% correct work.

Sure! But this will be the start. If you _can_ change this, then you can go in this direction. 😄

akosinov, 2018-02-22:
if my attempts w. microcontroller will fail, ill do this)
Another good idea from middle of russia: use line-out of soundcard. 22kHz is nothing for sound out, modulation software is also simple. But i'm not a master of analog circuits (

akosinov, 2018-02-22:
from diseqc standard: The nominal 22 kHz signalling amplitude is 650 mV (±250 mV) peak-peak.
To accommodate tolerances and voltage drop in the cable, the detector should
respond to amplitudes down to approximately 300 mV (±100 mV). The
maximum recommended amplitude to be applied to the bus is 1 volt peak to
peak

lars18th, 2018-02-22:
> not guaranty 100% correct work.

Regarding this I do this assumption: The professional satellite meter VAROS 109 (commented before) has the capability to re-program _any_ device sending only the "*.dis" file. So, if this it's true (and the manufacturer assures it), then:

- The device will load the 100% of the config if the 100% of the ".dis" file it's send.
- The 100% of the config it's loaded if each DiSeqC command (each line) it's agreeded with an OK from the device.

So we don't need to worry about _partial_ programming. If we can send any length DiSeqC commands, check the OK/FAIL response from the device, and send complete "*.dis" files... then all of this it's what we need to have.
That's all! 😄

lars18th, 2018-02-22:
Hi @akosinov ,

> if my attempts w. microcontroller will fail, ill do this)

I suggest to **first** do this try! It's easy to check it.

akosinov, 2018-02-22:
i just VERY worried about possible device bricking....

akosinov, 2018-02-22:
So i think good idea to make original programming sequences....

lars18th, 2018-02-22:
Hi @akosinov ,

Try this if you need a circuit for sending DiSeqC commands from a microcontroller:
http://github.com/acrerd/Arduino-Diseqc-solar-tracker

lars18th, 2018-02-22:
Hi,

Just for reference: This project can be used as a base of the "custom programmer":
http://github.com/superkuh/tune-s2-stepping

We can use the skeleton for the tool that reads the ".dis" file and sends the command and checks the response.

charlesduco, 2018-02-22:
Found some more info:
In the manual of these expensive JESS programmers you find the following info:
> DiSEqC commands extension according to CENELEC. EN50494 and/or EN50607

Then I googled some more and found the documents "EN 50494 2007" and "EN 50607 2015" on www.doc88.com

lars18th, 2018-02-23:
Hi @charlesduco ,

>> In the manual of these expensive JESS programmers you find the following info:

>DiSEqC commands extension according to CENELEC. EN50494 and/or EN50607

These documents are the description of Unicable I and Unicable II as CENELEC standards.

However, thank you for your contribution! :smile:

akosinov, 2018-03-01:
Hi guys! I still trying to read (first) config from my device and have some success:
1. protocol is reversed (small 'hack' dll for original programming software is done - now thie 'masterpiece' of coding is thinking i has original programmer. Reading sequences dumped)
2. my inverto devices violate diseqc standarts for answering frequences (more than 22 kHz) and signal amplitude (too small). It's avoid using ordinary DVB cards for reading answers from box. I think it's done for little protection.
3. I started using cheap STM32F072-DISCOVERY board with small additional components (one inductive, one capacitor. 2 resisters) as reader. Currently have 90% of success.
4. I think device manufacturers is too smart in programmers market protectection, so i thonk we'll not able to program devices by unstandard(short) sequences. we'll see.

lars18th, 2018-03-01:
Hi @akosinov ,

Great to hear about these good news! :+1: 

> 1. protocol is reversed (small 'hack' dll for original programming software is done - now thie 'masterpiece' of coding is thinking i has original programmer. Reading sequences dumped)

Great! You can create then a document with a short description of the _protocol_. You can?

> 2. my inverto devices violate diseqc standarts for answering frequences (more than 22 kHz) and signal amplitude (too small). It's avoid using ordinary DVB cards for reading answers from box. I think it's done for little protection.

Bad news. However, I feel this is only for **reading** the config from the device. For **writing** will be sufficient to just use standard _large_ DiSeQC commands. Right?

> 3. I started using cheap STM32F072-DISCOVERY board with small additional components (one inductive, one capacitor. 2 resisters) as reader. Currently have 90% of success. 

Yeah! You need now to share this circuit.

> 4. I think device manufacturers is too smart in programmers market protectection, so i thonk we'll not able to program devices by unstandard(short) sequences. we'll see.

Well, I don't see any reason for _protecting_ the protocol. I prefer open tools and protocols. In any case, if we can have a DIY interface for sendding the "*.dsq" files will be sufficient for 99% of the poeple.

**Perhaps is time to create a new project in GitHUB for a DYI Unicable Programmer?**

charlesduco, 2018-03-01:
> my inverto devices violate diseqc standarts for answering frequences (more than 22 kHz) and signal amplitude (too small).
Hard to believe. DiSEqC is a very tolerant standard as described in the hardware design targets of the DiSEqC bus spec.
Also, to operate correctly an Inverto Unicable II LNB needs 400mA power.

akosinov, 2018-03-09:
Im still fighting with my box. Now i has a digital osciloscope and i very surprised. I can see request (generated by STB Card, just 7A - answer free UBs) and answer. Request amplitude is near 400mV (correct), but amplitude of answer is only 50 mV. This is very close to noise level. I dont understand how i can do stable diseqc signal detection.

akosinov, 2018-03-09:
Connection is : STB Card - T-splitter(passive, only wires), 15m of nice SAT cable - Unicable Box.
osciloscope at third connector of T-splitter

charlesduco, 2018-03-09:
Did you measure the current in the circuit? Maybe the STB Card is not able to provide the required power for the LNB.
Connecting a suitable power inserter can be a solution to that problem: http://www.inverto.tv/accessories/326/unicable-ii-power-inserter-5-2400mhz

akosinov, 2018-03-09:
Im tryed with inverto power inserter too (included in my buyed bundle). Result the same.

charlesduco, 2018-03-10:
Maybe a simple t-splitter isn't the right way to view DiSEqC signals on an osciloscope.
Shouldn't you filter the 22 kHz signal with a low pass filter and a DC decoupling capacitor as described here: https://onetransistor.blogspot.be/2016/01/diseqc-monitor-soundcard.html
More info on DiSEqC electronic circuits: http://www.ringolake.com/pic_proj/satellite/sat_projects.html

akosinov, 2018-03-10:
already done w same result. also tried to amplify signal by simple one-transistor schema. same - signal level is too close to noise. My next idea - building selfmade  Fconnector-Fconnector with tap. let's solder on! :(((( uhh... hate analog circuits.....

lars18th, 2018-03-12:
Hi @akosinov ,

As I see you have problems with the signal **from** the device (response), it's this right?

akosinov, 2018-03-13:
yep

lars18th, 2018-03-13:
Hi @akosinov ,

Then the project can be divided into: a) send programming DiSEqC sequences; b) receive DiSEqC configuration from the device. Why? Because the confirmation from the device when sending DiSEqC commands can be taken for granted. So, don't worry about responses for A.

What you think?

akosinov, 2018-03-13:
Hi! As i explained above i litle scared about wrong programming & device bricking. So, i think, good idea to make reader first, programmer after. If i can't read eeprom contents i'm not 100% sure in proper device functionality.
If u has any STB card, unicable device & stresskill pills  i can provide programmer C code draft (for 4-byte portions programming) without any warranty ;-)

lars18th, 2018-03-13:
Hi @akosinov ,

> Hi! As i explained above i litle scared about wrong programming & device bricking. So, i think, good idea to make reader first, programmer after. If i can't read eeprom contents i'm not 100% sure in proper device functionality.

OK. Sorry! I accept it. :+1:

akosinov, 2018-03-15:
OOhhhh my god. I did it! Now i can read all settings from device. Main problem - low signal amplitude - solved. Too low signal amplitude is a result of high input impedance of my reader and much lower input impedance of STB card. So i had a "resistor divider" and stb card simle "eat" all signal. Solved by diode.
The second problem - a 'little compression' in programming diseqc sequences also solved.
Now i started "STB card as unicable II programmer" project.

And i has a little question - maybe some people will start to reverse engineering binary data itselt. I has no time for it. Just try to change something in original prog, save BINARY (not DSQ) file and find a differences.... and more... and more.....

akosinov, 2018-03-15:
If some people has ANY documetation for unicable SoCs from Maxlinear please contact me. It is realy hard to do something with closed eyes. NDA is not a problem.

lars18th, 2018-03-15:
> Now i started "STB card as unicable II programmer" project.

Great! :smile: 

Please, publish something and we can start to check it...

charlesduco, 2018-03-16:
> I did it!
Excellent work, so all signals coming from the Inverto devices are meeting the DiSEqC bus spec, as I expected.

> maybe some people will start to reverse engineering binary data itselt.
I'm already started. I know how to decode the first 28 bytes of a .bin file to a .txt file, so I can generate ODU_VER, ODU_NAME and ODU_LO parts from a .bin file and I have notes about the rest of the file, I know the location of ODU_BAND and ODU_UB_CSS and ODU_BOOT parts.
I will write a Python script and compare the output from it to the output of the Unicable II Programmer software.

lars18th, 2018-03-16:
Hi,

And what about this?

> small 'hack' dll for original programming software is done - now thie 'masterpiece' of coding is thinking i has original programmer.

Why not share this hack and start to develop a DIY interface using the original software?

akosinov, 2018-03-17:
It's not so easy. I can see 'application to programmer' bytes, but not known proper answers ;-) just 0x0101 is ok for INIT command and i can see diseqc reading sequences. It's all. I think this job cannot done without dump of real exchange with original programmer.
my simple HookDLL code on https://github.com/akosinov/unicable/mpusbapi.cpp
Change path of logfile, use you preferred compiler for 32bit dll and replace dll in original program folder.

lars18th, 2018-03-19:
Hi @akosinov ,

> my simple HookDLL code on https://github.com/akosinov/unicable/blob/master/mpusbapi.cpp

Thank you!

> I think this job cannot done without dump of real exchange with original programmer.

Ok. So we need to found one.
However, one suggestion/idea for your HookDLL: if you add support to pipe it to the real DLL, then we can share the code with someone with the programmer and log all commands. The idea is quite simple: your DLL replaces the original, and your DLL then loads the original DLL (renamed to `mpusbapi-org.dll`) and it call to the exported functions.
What you think?

[5 more replies]

closed 2018-04-03 by lars18th