perexg/satip-axe #100: No network after boot

Oretuni, 2017-12-21:
Hello,

my digibit runs fine the last 2 years with version 10 on USB. Today I read about safe electricity with 
the parameter -P 0. I've tried it. After than i did not have any network. I could not login by ftp and the digibit was lost in my network. 
Than I've tried to to build a new USB stick with the lates version, I did the reset an so on ...

Nothing helps.
Any idea what can i do to run the digibit with satip-axe?

thanks 
Daniel

perexg, 2018-01-09:
Try the configuration reset - as described in README (recovery from the wrong configuration).

Oretuni, 2018-01-10:
In the meantime I have tried a lot. Of course I read the readme. Maybe I have now done some things in the wrong order. Anyway, I think I did not immediately reset the configuration. Instead, I flashed the new version 14 to USB - no change. 
Then I flashed the 14 on the internal memory. after that it looked like it would work, but but I suddenly had many artifacts when I used more then 1 tuner. Than I tried to reset to the original firmware to find out if the device may be damaged. Or whether it is on the satip-axa. With the Orignal software there was never such a behavior. But the reset did not work. It always starts the satip-axe from the flash.
Then I modified the config to the oldest minisatip version. Now the artifacts start only from the simultaneous use of 3 tuners. 
So I have 2 problems:
- No original firmware anymore
- Maybe network problems

perexg, 2018-01-25:
To return to the original firmware use idl4k.recovery script (rename it to idl4k.scr on your flash drive). If it does not work - try reflash the original firmware from the flash drive.

If you use only UDP clients, the probability of the missed packets is a bit higher with minisatip than the original firmware which does really weird packet timing things to avoid such packet loss. I'm using TCP data transfer mode only.

godoll, 2018-02-10:
"reflash the original firmware from the flash drive"

How do you do this? (If I'm not mistaken I think you once had the procedure in your README, but I can't find it any longer.)
If I recall correctly you'd somehow have to use the files from the firmware dir?

closed 2018-09-17 by perexg