perexg/satip-axe #101: [INFO] --dmx_source parameter no longer used for setting the source on Unicable inputs

catalinii, 2018-01-03:
The line: https://github.com/catalinii/minisatip/blob/49785a82cc2d76f06d59a465a37320a0268ca6ea/src/axe.c#L417 no longer uses dmx_source and as a result the source can be set using -S or link-adapters

Considering that set_link_adapters and set_slave_adapter has the same functionality one of them, we can deprecate set_link_adapters and the code from minisatip.c.

Additionally a slave adapter should not have a master adapter that has different polarity band or diseqc.

Thanks

perexg, 2018-01-25:
> Additionally a slave adapter should not have a master adapter that has different polarity band or diseqc.

The AXE hardware is a bit different, this input number means really the physical input only. There's buildin matrix, so a tuner can be connected to any input. So the config like: unicable at input 0 which connects tuners 2 and 3 and unicable at input 3 which connects tuners 0 and 1 is possible.

catalinii, 2018-01-25:
Yeah that is exactly how FBC tuners are. 2 hw inputs and all other tuners are connected to one of them (based on procfs configuration). The diseqc/unicable commands are sent thru the HW tuner only (0, 1) and I have the feeling there are a lot of similarities with axe.

9000h, 2018-01-25:
the  AXE may have MaxLinear the newer  broadcom or silabs
https://dreambox.de/board/index.php?attachment/12826-silabs-jpg/

perexg, 2018-01-25:
AXE: The used chips are STV6120 and STV0900 and LNBH24. Basically, STV6120 has connected all four inputs to the coaxial inputs. So the 'running' tuner number might totaly differ from the 'input' number (so any tuner can be master and any tuner can use any input). The diseqc is tied with the 'input' of course but that's all.

kianusch, 2018-01-30:
could you give me an example what parameters to use for minisatip8 with following HW setup:
One cable coming in from a UNICABLE Switch (Channel 1,1076, Channel 2,1178) - 
Do I need to use the (slave) -S at all - or is this handled by the AXE Hardware?

perexg, 2018-01-30:
Input 0 is used when no master source is defined for unicable, so you don't need to use -S.

closed 2018-09-17 by perexg