perexg/satip-axe #102: Seeing bad artifacts with http streams compared to original software

jollyjinx, 2018-01-06:
One of my clients only has a 40Mbit/s connection and that one is using http stream urls like:

http://satip/?src=1&freq=11494&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=on&sr=22000&fec=23&pids=0,17,18,5100,5101,5102,5104

using the original software everything looks fine, but using the satip-axe software I can only watch SD channels without problem, HD channels then break up in pixelgarbage.
Using the satip receiver with the satip-axe software locally on a gigabit network works fine.

It seems to me something related to the network stack/buffers - like it's not correctly configured. Any ideas on how to work around that ? 
I really would like to to use the satip-axe distribution as it does work better for my recording software (eyetv) which has problems with the original software seeing the receiver in use by others with the original software - even that's not the case.

perexg, 2018-01-08:
Try '-b 4042752' option or higher.

jollyjinx, 2018-01-08:
great, will try when I'm back at the site and will report back.

jollyjinx, 2018-01-09:
No it does not work, I fiddled around with -B and -b options with small and large buffers and nothing works.
I found out that I can use the Network link conditioner provided with macOS with a Bandwidth setting of 20mbps/8mbps (delay 15ms) to replicate the issue locally and the original software that still works with HD streams ( which use up 16mbps ).

Any ideas on settings on how to get around it or what might be the problem ? Maybe the kernel has to be compiled with different network buffer settings ?

perexg, 2018-01-09:
http://www.linux-admins.net/2010/09/linux-tcp-tuning.html - net.ipv4.tcp_wmem  ?

jollyjinx, 2018-01-09:
Great, this works ! using no additional options for minisatip itself and just using sysctl

net.ipv4.tcp_wmem = 65536 65536 33554432

and it works - even with a 20MBit line setting - so enough headroom for my 40Mbit line. Now I just have to figure out how to persist this value as writing that in sysctl.conf wasn't saved. 

Many thanx for your help

perexg, 2018-01-09:
Please, keep this open. I'll probably make this as default in the next release.

estimadarocha, 2018-01-10:
how do i make those "net.ipv4.tcp_wmem" values permanent @perexg? everytime i reboot thei're gone!

i remember i used to to that on crontabs but really can't remember how to do it now...

thanks

jollyjinx, 2018-01-11:
Log in via ssh  root/satip

and uncomment the last line in /etc/sysconfig/config to look like:

CUSTOMD="/etc/sysconfig/myboot"

add a file in /etc/sysconfig/myboot containing:

----- /etc/sysconfig/myboot ----
#!/bin/sh
/sbin/sysctl -w net.ipv4.tcp_wmem='65536 65536 33554432'
-----

chmod 755 to /etc/sysconfig/myboot

reboot

and you are done

perexg, 2018-01-11:
Or create /etc/sysconfig/localboot. This is a part of /etc/init.d/rcS:

<pre>
# local boot
if test -x /etc/sysconfig/localboot ; then
  /etc/sysconfig/localboot
fi
</pre>

jollyjinx, 2018-01-11:
I wasn't aware of that - that's even better - only one file to add :-)

```
echo "#\!/bin/sh\nsysctl -w net.ipv4.tcp_wmem='65536 65536 33554432'\n" >/etc/sysconfig/localboot
chmod 755 /etc/sysconfig/localboot
```

estimadarocha, 2018-01-11:
thanks for the help...

estimadarocha, 2018-01-16:
@perexg sorry again to be persisting on this... and if you think this can be a new issue just say and i will open a new one.

I have been using satipaxe for a long time now.. but from time to time it's almost unusefull.

Getting the exemple of today... from one hour to the other it started increasing the cc errors on webinterface and them tvheadend start with:

2018-01-16 15:19:33.178 TS: DVB-S 30/12322H/RTP 1: MPEG2VIDEO @ #5136: Invalid start code 9d:63:50
2018-01-16 15:19:33.178 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5137: Invalid start code 88:06:e9
2018-01-16 15:19:33.181 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5139: Invalid start code b5:0b:9f
2018-01-16 15:19:42.315 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 23169600, last = 21308400
2018-01-16 15:19:42.315 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 24014889, last = 22153689
2018-01-16 15:19:42.445 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23133600, last = 21278160
2018-01-16 15:19:42.445 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23978889, last = 22123449
2018-01-16 15:19:42.447 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23145120, last = 21289680
2018-01-16 15:19:42.447 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23990409, last = 22134969
2018-01-16 15:19:52.277 TS: DVB-S 30/12322H/RTP 1: MPEG2VIDEO @ #5136 Continuity counter error (total 17)
2018-01-16 15:19:52.279 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5137 Continuity counter error (total 16)
2018-01-16 15:19:52.279 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5139 Continuity counter error (total 15)
2018-01-16 15:19:52.279 TS: DVB-S 30/12322H/RTP 1: TELETEXT @ #5138 Continuity counter error (total 14)
2018-01-16 15:19:52.281 tsfix: transport stream TELETEXT, DTS discontinuity. DTS = 24022207, last = 23895238
2018-01-16 15:19:52.281 tsfix: transport stream TELETEXT, DTS discontinuity. DTS = 24867496, last = 24740527
2018-01-16 15:19:52.396 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 24073200, last = 23940000
2018-01-16 15:19:52.396 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 24918489, last = 24785289
2018-01-16 15:19:52.399 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24040800, last = 23896080
2018-01-16 15:19:52.399 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24886089, last = 24741369
2018-01-16 15:19:52.513 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24052320, last = 23907600
2018-01-16 15:19:52.513 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24897609, last = 24752889

i don't think it's a problem with satip itself cause i have 3. Of those 3 2 are working and i got same behaviour.... i did run "top" on it and cpu wasn't not even close to 100% not even 50% it was just streaming a simple channel.

the configuration is tweaked with things i been catching here:

MINISATIP7_OPTS="-q *:100-15-0-5-5-0 -u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -H 5:25 -b 4042752 -B 10000 -W 1"

by the way the sat signal are very good in all of them... so no more ideas

anyone have new ideas?

perexg, 2018-01-16:
Do you see CC errors also on the satip boxes ?

estimadarocha, 2018-01-16:
yes
[image]

estimadarocha, 2018-01-16:
issue #103

perexg, 2018-01-25:
Fixed in new build 15 (tcp_wmem).

closed 2018-01-25 by perexg