perexg/satip-axe #103: CC Error's increase from time to time

estimadarocha, 2018-01-16:
@perexg sorry again to be persisting on this... and if you think this can be a new issue just say and i will open a new one.

I have been using satipaxe for a long time now.. but from time to time it's almost unusefull.

Getting the exemple of today... from one hour to the other it started increasing the cc errors on webinterface and them tvheadend start with:

2018-01-16 15:19:33.178 TS: DVB-S 30/12322H/RTP 1: MPEG2VIDEO @ #5136: Invalid start code 9d:63:50
2018-01-16 15:19:33.178 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5137: Invalid start code 88:06:e9
2018-01-16 15:19:33.181 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5139: Invalid start code b5:0b:9f
2018-01-16 15:19:42.315 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 23169600, last = 21308400
2018-01-16 15:19:42.315 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 24014889, last = 22153689
2018-01-16 15:19:42.445 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23133600, last = 21278160
2018-01-16 15:19:42.445 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23978889, last = 22123449
2018-01-16 15:19:42.447 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23145120, last = 21289680
2018-01-16 15:19:42.447 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 23990409, last = 22134969
2018-01-16 15:19:52.277 TS: DVB-S 30/12322H/RTP 1: MPEG2VIDEO @ #5136 Continuity counter error (total 17)
2018-01-16 15:19:52.279 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5137 Continuity counter error (total 16)
2018-01-16 15:19:52.279 TS: DVB-S 30/12322H/RTP 1: MPEG2AUDIO @ #5139 Continuity counter error (total 15)
2018-01-16 15:19:52.279 TS: DVB-S 30/12322H/RTP 1: TELETEXT @ #5138 Continuity counter error (total 14)
2018-01-16 15:19:52.281 tsfix: transport stream TELETEXT, DTS discontinuity. DTS = 24022207, last = 23895238
2018-01-16 15:19:52.281 tsfix: transport stream TELETEXT, DTS discontinuity. DTS = 24867496, last = 24740527
2018-01-16 15:19:52.396 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 24073200, last = 23940000
2018-01-16 15:19:52.396 tsfix: transport stream MPEG2VIDEO, DTS discontinuity. DTS = 24918489, last = 24785289
2018-01-16 15:19:52.399 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24040800, last = 23896080
2018-01-16 15:19:52.399 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24886089, last = 24741369
2018-01-16 15:19:52.513 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24052320, last = 23907600
2018-01-16 15:19:52.513 tsfix: transport stream MPEG2AUDIO, DTS discontinuity. DTS = 24897609, last = 24752889

i don't think it's a problem with satip itself cause i have 3. Of those 3 2 are working and i got same behaviour.... i did run "top" on it and cpu wasn't not even close to 100% not even 50% it was just streaming a simple channel.

the configuration is tweaked with things i been catching here:

MINISATIP7_OPTS="-q *:100-15-0-5-5-0 -u 0:0-1210,1:1-1420,2:2-1680,3:3-2040 -H 5:25 -b 4042752 -B 10000 -W 1"

and network is also tweaked with net.ipv4.tcp_wmem='65536 65536 33554432

by the way the sat signal are very good in all of them... so no more ideas

anyone have new ideas?

estimadarocha, 2018-01-16:
[image]

estimadarocha, 2018-01-16:
few minutes after last picture

[image]

perexg, 2018-01-16:
Those are reception errors. The software (minisatip) gets the incomplete data. The hardware decoder fails to reconstruct the original stream.

estimadarocha, 2018-01-16:
[image]
 
second adapter!

@perexg 

i understand that... but its a bit strange that somehow they work well for sometime and sudendly start to make this... and this is even more awkward because it happens on 2 different satipaxe's that are on completely different places (houses)... could it be a satelite problem?

estimadarocha, 2018-01-16:
from 145 to 220 in few minutes

[image]

perexg, 2018-01-16:
Note that those "level meters" do not show all reception errors, you should probably analyze the quality of the signal using a measurement tool or try to replace components step-by-step. Also, in some cases, the attenuation might help when the signal is too strong (outside the tuner specifications). Look for STV6210_BAMP in /etc/sysconfig/config or try to add a hardware attenuator. But it's just a guess, everything (wind, other environment conditions) might cause this.

And yes, it might be also a broadcaster problem (the uplink to the orbital device).

estimadarocha, 2018-01-16:
yes i will check components but its a bit strange the errors start at exact same time.

I am putting my thir unit to run (also on a third place) to see if the behaviour is the same!

estimadarocha, 2018-01-16:
@perexg it seen's like STV6210_BAMP can improve results... however after that i no longer can get "ok" in scan status on tvheadend... all the frequencies say "OK (partial)"

2018-01-16 18:28:42.073 mpegts: 10770H in DVB-S 30W - scan complete (partial - 21/1 tables)
2018-01-16 18:28:42.073 subscription: 0129: "scan" unsubscribing
2018-01-16 18:29:03.468 mpegts: 10890V in DVB-S 30W - scan complete (partial - 19/1 tables)
2018-01-16 18:29:03.468 subscription: 012E: "scan" unsubscribing

is this normal?
related to this changes i am testing or is this any commit of tvheadend?

closed 2018-01-24 by perexg