perexg/satip-axe #106: minisatip without libdvbcsa?

kianusch, 2018-01-30:
What's the reason, that the minisatip versions in satip-axe have no libdvbcsa support?

Especially because oscam is included - but oscam does not make sense without libdvbcsa support in minisatip.

Just wondering.

CvH, 2018-01-30:
soc is heavily limited by cpu so libdvbcsa is already too much here

kianusch, 2018-01-30:
Hmmm ... but this works (at least) in my setup:

Digibit - MinitsatIP on one HD encrypted channel  - with oscam running on an remote raspberry.
A second client can watch SD.

I was also wonderung if it would make sense - or if it would work - or if it would help - to replace libdvbcsa with ffdesca?

perexg, 2018-01-30:
> Digibit - MinitsatIP on one HD encrypted channel - with oscam running on an remote raspberry.

Decoding of only one channel of this device does not sound like a good usage. There's hardware descrambler which should be used, but I don't have time to play with it. Issue #17 .

> libdvbcsa with ffdesca

FFdesca has serious trouble on other architectures than x86. And this code is no longer maintained.

closed 2018-01-30 by perexg