perexg/satip-axe #109: ntpd not working

ecdlguy, 2018-02-14:
Hi,

ntpd is running, however the arguments ('-n', '-I') seem to be outdated for this version of ntpd so ntpd does nothing. I've touched /tmp/nontpd in order to prevent /etc/init.d/ntpd to respawn it.
After that, starting ntpd with the '-p' option only like

`ntpd -p 0.pool.ntp.org`

immediately sets the correct time.

cheers,
Thorsten

perexg, 2018-02-14:
<pre>
# cat /etc/motd 

Welcome to SATIP-AXE server 201801301900-15

# ntpd --help
BusyBox v1.26.2 (2018-01-25 08:55:26 CET) multi-call binary.

Usage: ntpd [-dnqNwl -I IFACE] [-S PROG] [-p PEER]...

NTP client/server

	-d	Verbose
	-n	Do not daemonize
	-q	Quit after clock is set
	-N	Run at high priority
	-w	Do not set time (only query peers), implies -n
	-S PROG	Run PROG after stepping time, stratum change, and every 11 mins
	-p PEER	Obtain time from PEER (may be repeated)
		If -p is not given, 'server HOST' lines
		from /etc/ntp.conf are used
	-l	Also run as server on port 123
	-I IFACE Bind server to IFACE, implies -l
# ps ax | grep ntpd
  722 root       0:00 {ntpd} /bin/sh /etc/init.d/ntpd
  727 root       0:00 /usr/sbin/ntpd -n -I eth0 -p 192.168.100.1
  947 root       0:00 grep ntpd
</pre>

No issue here..

ecdlguy, 2018-02-14:
Hi perexg,

```
# cat /etc/motd 
Welcome to SATIP-AXE server 201801301900-15
```

After reboot, ntpd is running 

```
# ps ax | grep ntpd
  721 root       0:00 {ntpd} /bin/sh /etc/init.d/ntpd
  730 root       0:00 /usr/sbin/ntpd -n -I eth0 -p 0.pool.ntp.org
  787 root       0:00 grep ntpd
```

However, date and time are wrong:

```
 # date
Thu Jan  1 13:00:18 CET 2015
```

So I do
```
touch /tmp/nontpd
kill 730
```
Check if there's no ntpd running:
```
# ps ax | grep ntpd
  794 root       0:00 grep ntpd

```
And manually start ntpd:
```
ntpd -p 0.pool.ntp.org
```
Finally, the date and time are correct.
```
 # date
Wed Feb 14 18:04:26 CET 2018
```
Maybe it's a problem with '-n' vs. '-N' or '-I' (implies '-l' ?). I don't know, it only works without these options.

cheers,
Thorsten

perexg, 2018-02-14:
Is sufficient to kill ntpd after start? It might be a network boot timing issue.

ecdlguy, 2018-02-14:
> Is sufficient to kill ntpd after start? It might be a network boot timing issue.

Now I'm flabbergasted. I thought I'd tried just killing it. That's why I started to play with the options. 
Be that as it may, it's sufficient to kill ntpd after start.

Notwithstanding the above, I suggest to change

`/usr/sbin/ntpd -n -I eth0 $NTPD_OPTS`

in /etc/init.d/ntpd to 

`/usr/sbin/ntpd $NTPD_OPTS`

and add the options to the config. 

cheers,
Thorsten

perexg, 2018-02-15:
The '-n' is mandatory - otherwise the /etc/init.d/ntpd script will be in an endless loop.

The '-I eth0' is harmless.

Could you run 'strace -p <ntpd_pid>' when things does not work?

ecdlguy, 2018-02-15:
```
~ # strace -p 728
Process 728 attached - interrupt to quit
restart_syscall(<... resuming interrupted call ...>) = 0
socket(PF_NETLINK, SOCK_RAW, 0)         = 4
bind(4, {sa_family=AF_NETLINK, pid=0, groups=00000000}, 12) = 0
getsockname(4, {sa_family=AF_NETLINK, pid=728, groups=00000000}, [12]) = 0
time(NULL)                              = 1420113767
sendto(4, "\24\0\0\0\26\0\1\3g7\245T\0\0\0\0\0\0\0\0", 20, 0, {sa_family=AF_NETLINK, pid=0, groups=00000000}, 12) = 20
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"0\0\0\0\24\0\2\0g7\245T\330\2\0\0\2\30\200\376\1\0\0\0\10\0\1\0\177\0\0\1"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 108
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"@\0\0\0\24\0\2\0g7\245T\330\2\0\0\n\200\200\376\1\0\0\0\24\0\1\0\0\0\0\0"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 192
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"\24\0\0\0\3\0\2\0g7\245T\330\2\0\0\0\0\0\0\1\0\0\0\24\0\1\0\0\0\0\0"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 20
close(4)                                = 0
open("/etc/hosts", O_RDONLY|O_CLOEXEC)  = 4
fstat64(4, {st_mode=S_IFREG|0755, st_size=82, ...}) = 0
old_mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x2975e000
read(4, "127.0.0.1\tlocalhost.localdomain\t"..., 4096) = 82
read(4, "", 4096)                       = 0
close(4)                                = 0
munmap(0x2975e000, 4096)                = 0
gettimeofday({1420113767, 385819}, NULL) = 0
gettimeofday({1420113767, 386353}, NULL) = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420113767, 387541}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "\230\335\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\311_\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420113767, 392468}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "\230\335\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\311_\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
gettimeofday({1420113767, 394886}, NULL) = 0
gettimeofday({1420113767, 395434}, NULL) = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420113767, 396468}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "O\301\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\3710\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420113767, 399865}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "O\301\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\3710\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
write(2, "ntpd: bad address '0.pool.ntp.or"..., 35) = 35
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigaction(SIGCHLD, NULL, {SIG_DFL, [], 0}, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
nanosleep({5, 0}, ^C <unfinished ...>

```

perexg, 2018-02-16:
It seems that your /etc/sysconfig/resolv.conf (or /etc/resolv.conf) is wrong. It points to 127.0.0.1 (localhost) at the boot time. Do you use DHCP? But there's `killall -9 ntpd` command in /etc/init.d/dhcpc-event , so it should be handled correctly.

ecdlguy, 2018-02-16:
I'm using a static IP setup:
```
#
# IPV4 config
#
#IPV4="dhcp"
# or
IPV4="192.168.2.30"
IPV4_NETMASK="255.255.255.0"
IPV4_BROADCAST="192.168.2.255"
IPV4_GATEWAY="192.168.2.1"
```
Here's my resolv.conf:
`nameserver 192.168.2.1`

perexg, 2018-02-16:
Ok, so the question is where ntpd gets the 127.0.0.1 address for the DNS resolution.

ecdlguy, 2018-02-16:
/etc/resolv.conf is a link to /mnt/data/satip-axe/sysconfig/resolv.conf. 
Maybe 'data' is not yet mounted when ntpd is started?

perexg, 2018-02-16:
Could you try to add 'killall -9 ntpd' to /etc/sysconfig/localboot ? (Make this script as executable).

ecdlguy, 2018-02-16:
Ok I did
```
touch /etc/sysconfig/localboot
chmod 755 /etc/sysconfig/localboot
nano /etc/sysconfig/localboot
```
and added the line 'killall -9 ntpd'
after a reboot, ntpd is running but the date / time is not correct.

PS: If I call the script manually, ntpd as well as the "watchdog script" gets killed, so ntpd is not automatically restarted after 'localboot' has been executed.

PPS: Do I have to enable this:
```
# custom daemon/service/setup executed from sysinit
#CUSTOMD="/etc/sysconfig/myboot"
```
 and rename the script accordingly?

perexg, 2018-02-16:
The first line for the executable script should be '#!/bin/sh' . But I think that ntpd is not started at this point yet. The /etc/init.d/rcS script which mounts the config partition is executed before spawned tasks. CUSTOMD is another way to run own code.

ecdlguy, 2018-02-16:
I've added '#!/bin/sh' to the script just to be sure but after a reboot the problem remains.
:-(

Just in case, here's my dmesg after boot:
```
Linux version 2.6.32.42_stm24_0208-idl4k_7108 (perex@stlinux.perex-int.cz) (gcc version 4.5.3 20111123 (STMicroelectronics/Linux Base 4.5.3-99) (GCC) ) #116 PREEMPT Tue Jan 30 19:00:18 CET 2018
Boot params:
... MOUNT_ROOT_RDONLY - 00000001
... RAMDISK_FLAGS     - 00000000
... ORIG_ROOT_DEV     - 00000200
... LOADER_TYPE       - 00000000
... INITRD_START      - 00000000
... INITRD_SIZE       - 00000000
Booting machvec: idl4k
Node 0: start_pfn = 0x70000, low = 0x8f000
Zone PFN ranges:
  Normal   0x00070000 -> 0x0008f000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00070000 -> 0x0008f000
On node 0 totalpages: 126976
free_area_init_node: node 0, pgdat 8145feb8, node_mem_map 82b45000
  Normal zone: 992 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 125984 pages, LIFO batch:31
Inverto idl4k board initialisation
STx7108 version 2.x, HOST core
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line:  console= hw=16001000000000000000000000000000 sn=1S81T51303001145 uuid=1F9C0DF4B11E4B10816327096D7DB9FD revBoot=stm24_0056-B0_80-idl4k-rev47 console=ttyAS0,115200 bigphysarea=20000
bpa2: partition 'bigphysarea' created at 0x72f29000, size 80000 kB (0x04e20000 B)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
PVR=04909300 CVR=60880000 PRR=0000a320
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Memory: 379008k/507904k available (3385k kernel code, 1095k data, 22312k init)
Hierarchical RCU implementation.
NR_IRQS:500
Failed to enable clk CLKA0_REF, ignoring
Failed to enable clk CLKA0_PLL0LS, ignoring
Failed to enable clk CLKA1_REF, ignoring
Failed to enable clk CLKA1_PLL0LS, ignoring
Console: colour dummy device 80x25
console [ttyAS0] enabled
sh_tmu: TMU0 used for clock events
sh_tmu: TMU0 used for periodic clock events
sh_tmu: TMU1 used as clock source
Calibrating delay loop... 595.96 BogoMIPS (lpj=297984)
Mount-cache hash table entries: 512
CPU: STx7108
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource TMU1
NET: Registered protocol family 2
IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
TCP established hash table entries: 16384 (order: 5, 131072 bytes)
TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
TCP reno registered
NET: Registered protocol family 1
RPC: Registered udp transport module.
RPC: Registered tcp transport module.
RPC: Registered tcp NFSv4.1 backchannel transport module.
[... 186 lines cut ...]
usbcore: registered new interface driver cp210x
cp210x: v0.09:Silicon Labs CP210x RS232 serial adaptor driver
USB Serial support registered for pl2303
usbcore: registered new interface driver pl2303
pl2303: Prolific PL2303 USB to serial adaptor driver
USB Serial support registered for FTDI USB Serial Device
usbcore: registered new interface driver ftdi_sio
ftdi_sio: v1.5.0:USB FTDI Serial Converters Driver
eth0: device MAC address 00:24:a6:00:0e:59
STMMAC MII Bus: Can't get IRQ 25 (PHY)
 No MAC Management Counters available
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
PHY: 1:01 - Link is Up - 1000/Full
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
```

makabar, 2018-03-09:
same problem here

bylderup, 2018-03-13:
I have the same problem. After restart of device the time/date is somewhere in 2015.

I always have to use the command "date MMDDHHMMYY"  in putty to get the correct date and time.

As far as I have seen  ntpd does not work on startup because the difference to hardware clock and/or system clock is to big.

hwclock --systohc  did not work.

Can you please explain 
how to set the timezone to Europe/Berlin
how to set hwclock  after correcting date/time with date command in putty

I think problems with ntpd will have gone after that, but I don´t know how..

perexg, 2018-03-13:
There is no RTC hardware, so hwclock won't work. Also, there are no time zones in the firmware to keep things simple. The initial date is set in /etc/init.d/rcS script:

<pre>
date 010112002015
</pre>

If 'ping 0.pool.ntp.org' works (the /etc/sysconfig/resolv.conf exists and is linked to /etc/resolv.conf), then ntpd should set the correct date. (You may change the NTP server in the config file.)

<pre>
# ping 0.pool.ntp.org
PING 0.pool.ntp.org (78.46.37.9): 56 data bytes
64 bytes from 78.46.37.9: seq=0 ttl=42 time=51.204 ms
</pre>

bylderup, 2018-03-13:
Ping is ok
but ntpd does not set correct time/date after device start.

will wait - maybe there is a solution in next release

your firmware is great btw.  Thanks a lot for it !

bylderup, 2018-03-14:
just some more feedback
changed Installation to lastest fw 15 beta
device is still coming up with 01.01.2015 12.00

hopsi, 2018-03-17:
I'm using 201802211835-15, dhcp, usb-boot (not flashed yet).
ntpd is not starting here as well. Have to killall ntpd to get the time.

estimadarocha, 2018-03-28:
i was experiencing time to time unresponsive hw from one of my digibit's... so i think a good solution was to run a reboot on specific hour each days... when i tried to do this i also noticed that ntpd was also not working for me... tried with and without dhcp and it seens it also using 127.0.0.1 for dns in my case!

Thanks

gitoriu, 2018-04-26:
fw: 201705251043-14

i have a timeserver for my lan.
not working: NTPD_OPTS="-p FQDNSNAME"  
working: NTPD_OPTS="-p IPOFTIMESER"

bylderup, 2018-04-26:
after changing NTPD_OPTS to the ip Adress of my router (giving time to lan) 
I do get UTC time now - but not the local time
I took a file in /etc/sysconfig named localtime with Europe/Berlin in it.
seems to be wrong  Europe/Amsterdam  didn´t work either

so time differs for 2 h now

What do I have to do to get correct local time ?

perexg, 2018-04-26:
Copy the file from your linux distribution. It's zoneinfo file not a text file. Example:

<pre>
$ scp /usr/share/zoneinfo/Europe/Prague root@satip1:/etc/sysconf/localtime
$ ssh root@satip1
# reboot
</pre>

phaze75, 2018-04-28:
@gitoriu  @perexg 
Changing the time server's FQDN to the IP address worked also for me. So ntpd has obviously a resolve issue. So how can I/we solve this? I don't like to have IP addresses in config files. /etc/resolv.conf is correctly pointing to my local name server. Using ping in the command line also correctly resolves the FQDN of the name server. So what could be the issue?

perexg, 2018-04-28:
It seems that /etc/init.d/dhcpc-event needs some more love to restart ntpd properly when the DNS parameters changes.

perexg, 2018-09-17:
Please, test the new build 15 - issue #94 .

estimadarocha, 2018-09-17:
> Copy the file from your linux distribution. It's zoneinfo file not a text file. Example:
> 
> $ scp /usr/share/zoneinfo/Europe/Prague root@satip1:/etc/sysconfig/localtime
> $ ssh root@satip1
> # cd /etc
> # ln -s sysconfig/localtime

is this still needed?

because i tested with that and even with that the time isn't correct!

perexg, 2018-09-17:
Yes, the zone file should be copied from a linux system to /etc/sysconfig/localtime (and reboot or symlink it to /etc).

estimadarocha, 2018-09-17:
cat /etc/motd

Welcome to SATIP-AXE server 201809171812-15

~ # ntpd --help
BusyBox v1.26.2 (2018-01-25 08:55:26 CET) multi-call binary.

Usage: ntpd [-dnqNwl -I IFACE] [-S PROG] [-p PEER]...

NTP client/server

        -d      Verbose
        -n      Do not daemonize
        -q      Quit after clock is set
        -N      Run at high priority
        -w      Do not set time (only query peers), implies -n
        -S PROG Run PROG after stepping time, stratum change, and every 11 mins
        -p PEER Obtain time from PEER (may be repeated)
                If -p is not given, 'server HOST' lines
                from /etc/ntp.conf are used
        -l      Also run as server on port 123
        -I IFACE Bind server to IFACE, implies -l
~ # ps ax | grep ntpd
  814 root       0:00 grep ntpd

no ntpd daemon running after reboot!

perexg, 2018-09-17:
You're right. The ntpd was killed including the shield script. Here is the fixed version of build 15: http://s000.tinyupload.com/?file_id=58656269804852319071\

estimadarocha, 2018-09-17:
is the link correct? it just return "download.php"

rkorn86, 2018-09-17:
Just remove trailing \

estimadarocha, 2018-09-17:
still no luck @perexg ... see the outdated below comment... even with ntpd running i don0t get correct time...

however if i kill it and run it again it set time correctly!

perexg, 2018-09-17:
Cannot confirm. Ntpd is killed correctly now with 201809171929-15 . Ntpd needs just some seconds to sync. I have correct time with the dhcp settings on my bod. Anyway, it's generally bad idea to use DHCP for server IPs.

ecdlguy, 2018-09-19:
Hi, it's still not working for me using 201809171929-15.
I did never use dhcp with the box, I also don't know why the title of the issue was changed.
ntpd is running after reboot, however date is not correct. After killing ntpd, date is correct.

perexg, 2018-09-21:
@ecdlguy please test rc2: https://github.com/perexg/satip-axe/releases/tag/build15-rc2

estimadarocha, 2018-09-21:
for me the problem still exist... but like @perexg i am using dhcp!

perexg, 2018-09-21:
Anyway, ntpd should exists after boot.. It seems that you killed it from your own script?

estimadarocha, 2018-09-21:
yes @perexg it exists after boot but time was not set!!!

however if i kill it and start again it works!

estimadarocha, 2018-09-21:
strace -p 969
Process 969 attached - interrupt to quit
restart_syscall(<... resuming interrupted call ...>) = 0
socket(PF_NETLINK, SOCK_RAW, 0)         = 4
bind(4, {sa_family=AF_NETLINK, pid=0, groups=00000000}, 12) = 0
getsockname(4, {sa_family=AF_NETLINK, pid=969, groups=00000000}, [12]) = 0
time(NULL)                              = 1420114273
sendto(4, "\24\0\0\0\26\0\1\3a9\245T\0\0\0\0\0\0\0\0", 20, 0, {sa_family=AF_NETLINK, pid=0, groups=00000000}, 12) = 20
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"0\0\0\0\24\0\2\0a9\245T\311\3\0\0\2\30\200\376\1\0\0\0\10\0\1\0\177\0\0\1"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 108
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"@\0\0\0\24\0\2\0a9\245T\311\3\0\0\n\200\200\376\1\0\0\0\24\0\1\0\0\0\0\0"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 192
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"\24\0\0\0\3\0\2\0a9\245T\311\3\0\0\0\0\0\0\1\0\0\0\24\0\1\0\0\0\0\0"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 20
close(4)                                = 0
open("/etc/hosts", O_RDONLY|O_CLOEXEC)  = 4
fstat64(4, {st_mode=S_IFREG|0755, st_size=82, ...}) = 0
old_mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x2975e000
read(4, "127.0.0.1\tlocalhost.localdomain\t"..., 4096) = 82
read(4, "", 4096)                       = 0
close(4)                                = 0
munmap(0x2975e000, 4096)                = 0
gettimeofday({1420114273, 661891}, NULL) = 0
gettimeofday({1420114273, 662051}, NULL) = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420114273, 663128}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "\333\346\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\376a\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420114273, 666914}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "\333\346\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\376a\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
gettimeofday({1420114273, 669280}, NULL) = 0
gettimeofday({1420114273, 669733}, NULL) = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420114273, 670370}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "^\271\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\203\22\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
socket(PF_INET, SOCK_DGRAM|SOCK_NONBLOCK, IPPROTO_IP) = 4
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
gettimeofday({1420114273, 673843}, NULL) = 0
poll([{fd=4, events=POLLOUT}], 1, 0)    = 1 ([{fd=4, revents=POLLOUT}])
send(4, "^\271\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\203\22\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
write(2, "ntpd: bad address '0.pool.ntp.or"..., 35) = 35
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigaction(SIGCHLD, NULL, {SIG_DFL, [], 0}, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
nanosleep({5, 0}, 0x7bcf0a00)           = 0
socket(PF_NETLINK, SOCK_RAW, 0)         = 4
bind(4, {sa_family=AF_NETLINK, pid=0, groups=00000000}, 12) = 0
getsockname(4, {sa_family=AF_NETLINK, pid=969, groups=00000000}, [12]) = 0
time(NULL)                              = 1420114278
sendto(4, "\24\0\0\0\26\0\1\3f9\245T\0\0\0\0\0\0\0\0", 20, 0, {sa_family=AF_NETLINK, pid=0, groups=00000000}, 12) = 20
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"0\0\0\0\24\0\2\0f9\245T\311\3\0\0\2\30\200\376\1\0\0\0\10\0\1\0\177\0\0\1"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 108
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"@\0\0\0\24\0\2\0f9\245T\311\3\0\0\n\200\200\376\1\0\0\0\24\0\1\0\0\0\0\0"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 192
recvmsg(4, {msg_name(12)={sa_family=AF_NETLINK, pid=0, groups=00000000}, msg_iov(1)=[{"\24\0\0\0\3\0\2\0f9\245T\311\3\0\0\0\0\0\0\1\0\0\0\24\0\1\0\0\0\0\0"..., 4096}], msg_controllen=0, msg_flags=0}, 0) = 20
close(4)                                = 0
[... cut ...]
send(4, "\226\311\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\1\0\1", 32, MSG_NOSIGNAL) = 32
poll([{fd=4, events=POLLIN|POLLOUT}], 1, 5000) = 1 ([{fd=4, revents=POLLOUT|POLLERR}])
send(4, "\372\312\1\0\0\1\0\0\0\0\0\0\0010\4pool\3ntp\3org\0\0\34\0\1", 32, MSG_NOSIGNAL) = -1 ECONNREFUSED (Connection refused)
close(4)                                = 0
write(2, "ntpd: bad address '0.pool.ntp.or"..., 35) = 35
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigaction(SIGCHLD, NULL, {SIG_DFL, [], 0}, 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
nanosleep({5, 0},

estimadarocha, 2018-09-21:
This is the config i am using now! i changed from dhcp to static!

#
# IPV4 config
#
#IPV4="dhcp"
# or
IPV4="192.168.1.114"
IPV4_NETMASK="255.255.255.0"
IPV4_BROADCAST="192.168.1.255"
IPV4_GATEWAY="192.168.1.254"

#
# ntpd
#
NTPD_OPTS="-p 0.pool.ntp.org"

is it normal that with static there's no resolve.conf?

perexg, 2018-09-21:
You must write your own resolve.conf and place it to /etc/sysconfig.

I don't know at this point, what task is accepting connections on TCP port 53:

<pre>
connect(4, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = 0
</pre>

on my box:

<pre>
# netstat -an | grep 53
#
</pre>

<pre>
# strace telnet localhost 53
connect(3, {sa_family=AF_INET, sin_port=htons(53), sin_addr=inet_addr("127.0.0.1")}, 16) = -1 ECONNREFUSED (Connection refused)
</pre>

weird... what's your 'netstat -an' ?

estimadarocha, 2018-09-21:
/ # netstat -an | grep 53
/ #

meanwhile i will create resolve.conf and put it there!

ecdlguy, 2018-09-21:
Is it really "resolve.conf" with an 'e' ?

I have resolv.conf in /etc/sysconfig with the following content:
`nameserver 192.168.2.1`

My IP ist static:
```
IPV4="192.168.2.30"
IPV4_NETMASK="255.255.255.0"
IPV4_BROADCAST="192.168.2.255"
IPV4_GATEWAY="192.168.2.1"

```

However, it doesn't work for me at the moment.
@estimadarocha What's the content of your resolv.conf?

cheers,
Thorsten

PS @perexg I've seen you answer in the other thread #108 and the new release, I'm going to test this evening, I'm very excited!

estimadarocha, 2018-09-21:
put resolv.conf file to /etc/sysconfig/resolv.conf

my bad sorry!

ecdlguy, 2018-09-21:
Ok and what's the content of this file?

estimadarocha, 2018-09-21:
nameserver 192.168.1.254

ecdlguy, 2018-09-21:
Ok thank you, just wanted to check if "search" and "domain" entries are mandatory. 
I'll test again this evening.

ecdlguy, 2018-09-21:
Yes! Using the latest release 15 RC 2 it works!
Thank you very much perexg!

closed 2018-09-21 by ecdlguy

phaze75, 2018-09-29:
@perexg: I just tried release 15 RC4 **with dhcp activated** and it still does not work for me. Only after killing ntpd and restarting it manually the time is set correctly. I am a bit confused by the earlier comments. Is it supposed to work now also with dhcp activated?

estimadarocha, 2018-09-29:
From my side I changed to fixed ip

phaze75, 2018-09-29:
I see. Still imho it can't be a final solution to disable dhcp.

perexg, 2018-10-01:
Guys, you may check the /etc/init.d/dhcpc-event script: https://github.com/perexg/satip-axe/blob/master/fs-add/etc/init.d/dhcpc-event#L49-L51 , the ntpd should be killed when /etc/resolv.conf and ip configuration changes. I did some tests and it works in my network.

phaze75, 2018-10-06:
If I was able to script I would be happy to contribute. Fact is I am not, so besides testing I will not be of much help.