perexg/satip-axe #111: Network to slow

lonerika, 2018-03-08:
Hello,
Of course, I would like to announce my results (iperf) as well.
First the Messund between 2 notebooks in the G-Lan:

[3] local 192.168.1.15 port 55310 connected with 192.168.1.100 port 5001
[ID] Interval Transfer Bandwidth
[3] 0.0-10.0 sec 1.09 GBytes 939 Mbits / sec

Here's the Digibit. Set from v14 to v15 (Feb)

[3] local 192.168.1.15 port 44456 connected with 192.168.1.99 port 5001
[ID] Interval Transfer Bandwidth
[3] 0.0-10.0 sec 56.1 MBytes 47.0 Mbits / sec

It is not very good.
Unfortunately I could not test the original FW.

Either the hardware does not manage it or it is a driver problem of the network card. At least that is my opinion.
My TV Headend is not running well (jerky) with more than 1 stream

greeting

Google translator

perexg, 2018-03-08:
test

<pre>
[  3]  0.0-10.0 sec   559 MBytes   469 Mbits/sec
</pre>

satip box

<pre>
# ethtool eth0
Settings for eth0:
	Supported ports: [ TP MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Half 1000baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 1
	Transceiver: external
	Auto-negotiation: on
	Supports Wake-on: ug
	Wake-on: d
	Current message level: 0x0000003f (63)
			       drv probe link timer ifdown ifup
	Link detected: yes
</pre>

switch

<pre>
                      name: ether18-slave-local
                    status: link-ok
          auto-negotiation: done
                      rate: 1Gbps
               full-duplex: yes
           tx-flow-control: no
           rx-flow-control: no
               advertising: 10M-half,10M-full,100M-half,100M-full,1000M-half,1000M-full
  link-partner-advertising: 10M-half,10M-full,100M-half,100M-full,1000M-half,1000M-full
</pre>

<pre>
                            name:  ether18-slave-local
                  driver-rx-byte:                    0
                driver-rx-packet:                    0
                  driver-tx-byte:          372 411 973
                driver-tx-packet:            7 026 641
                        rx-bytes:      103 154 472 076
                    rx-too-short:                    0
                           rx-64:              339 281
                       rx-65-127:              367 235
                      rx-128-255:            3 755 055
                      rx-256-511:            3 946 126
                     rx-512-1023:            1 713 851
                    rx-1024-1518:           67 186 368
                     rx-1519-max:                    0
                     rx-too-long:                    0
                    rx-broadcast:               11 318
                        rx-pause:                    0
                    rx-multicast:              511 243
                    rx-fcs-error:                    0
                  rx-align-error:                    0
                     rx-fragment:                    0
                     rx-overflow:                    0
                        tx-bytes:        6 277 954 034
                           tx-64:            9 651 204
                       tx-65-127:           25 976 618
                      tx-128-255:            2 820 669
                      tx-256-511:            3 502 860
                     tx-512-1023:               81 071
                    tx-1024-1518:            1 295 473
                     tx-1519-max:                    0
                     tx-too-long:                    0
                    tx-broadcast:            6 221 268
                        tx-pause:                    0
                    tx-multicast:           16 706 056
                     tx-underrun:                    0
                    tx-collision:                    0
          tx-excessive-collision:                    0
           tx-multiple-collision:                    0
             tx-single-collision:                    0
           tx-excessive-deferred:                    0
                     tx-deferred:                    0
               tx-late-collision:                    0
                tx-queue0-packet:           10 187 138
                  tx-queue0-byte:          974 939 907
                tx-queue1-packet:           31 797 191
                  tx-queue1-byte:        5 206 078 206
                tx-queue2-packet:            1 339 647
                  tx-queue2-byte:           96 567 535
                tx-queue3-packet:                3 919
                  tx-queue3-byte:              368 386
                tx-queue4-packet:                    0
                  tx-queue4-byte:                    0
                tx-queue5-packet:                    0
                  tx-queue5-byte:                    0
                tx-queue6-packet:                    0
                  tx-queue6-byte:                    0
                tx-queue7-packet:                    0
                  tx-queue7-byte:                    0
        tx-all-queue-drop-packet:                    0
          tx-all-queue-drop-byte:                    0
    tx-queue-custom0-drop-packet:                    0
      tx-queue-custom0-drop-byte:                    0
    tx-queue-custom1-drop-packet:                    0
      tx-queue-custom1-drop-byte:                    0
              policy-drop-packet:                    0
              custom-drop-packet:                    0
                 current-learned:                    1
                     not-learned:                    4
</pre>

perexg, 2018-03-08:
So it's definitely not problem with the axe hardware in general. You may try to play with the auto-negotiation, pause frame settings, speed, duplex. It seems that some packets are lost or so in your environment. Or it might be, that your satip box is faulty.

rkorn86, 2018-03-08:
```
# ethtool eth0
Settings for eth0:
        Supported ports: [ TP MII ]
        Supported link modes:   10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Half 1000baseT/Full 
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes:  10baseT/Half 10baseT/Full 
                                100baseT/Half 100baseT/Full 
                                1000baseT/Half 1000baseT/Full 
        Advertised pause frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        Supports Wake-on: ug
        Wake-on: d
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: yes
```

`[  4]  0.0-30.0 sec  1.05 GBytes   300 Mbits/sec` in my environment

lonerika, 2018-03-08:
Also do not believe in a bug in the satip-ax software. Since I have tested all and the same result everywhere.
Does the eth0 driver also come from satip-ax?
Here are my data:
# ethtool eth0
Settings for eth0:
        Supported ports: [TP ​​MII]
        Supported link modes: 10baseT / Half 10baseT / Full
                                100baseT / Half 100baseT / Full
                                1000baseT / Half 1000baseT / Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Advertised link modes: 10baseT / Half 10baseT / Full
                                100baseT / Half 100baseT / Full
                                1000baseT / Half 1000baseT / Full
        Advertised break frame use: No
        Advertised auto-negotiation: Yes
        Speed: 1000Mb / s
        Duplex: Full
        Port: MII
        PHYAD: 1
        Transceiver: external
        Auto-negotiation: on
        Supports Wake-on: ug
        Wake-on: d
        Current message level: 0x0000003f (63)
                               drv sample link timer ifdown ifup
        Link detected: yes

Are identical to yours.
By the way, the original FW is even worse.
But hardware ??

lonerika, 2018-03-08:
How did you start iperf (parameters)?

perexg, 2018-03-08:
My firmware uses identical kernel with the original one. I didn't use any extra argument for iperf. If you tried multiple cables / switches then I guess that you have faulty hardware and I would ask for a replacement (if your device is under the warranty). Don't forget to flash and activate the original firmware before your return.

9000h, 2018-03-08:
@lonerika 
if you have AVM Fritzbox make sure the ports are not in "green mode"

lonerika, 2018-03-08:
perexg wrote:
"My firmware uses identical kernel with the original one. I didn't use any extra argument for iperf. If you tried multiple cables / switches then I guess that you have faulty hardware and I would ask for a replacement (if your device is under the warranty). Don't forget to flash and activate the original firmware before your return

Oh, I had to enter at least -s for server operation and for client side iperf -c IP-No
Start via USB! Is this the problem?

with -u the thing is fast. (UDP)
Please refer:
[3] local 192.168.1.15 port 49212 connected with 192.168.1.99 port 5001
[ID] Interval Transfer Bandwidth
[3] 0.0-10.0 sec **1.11 GB 957 Mbits / sec**
[3] Sent 813854 datagrams

9000h wrote:
That was the first thing I did.

greeting

perexg, 2018-03-08:
> Oh, I had to enter at least -s for server operation and for client side iperf -c IP-No

Yes, -s and -c are mandatory, so I did not mentioned them.

lonerika, 2018-03-09:
I have order a new Hardware!

closed 2018-03-09 by lonerika