perexg/satip-axe #114: satip over tcp not working with Dvbviewer/Media Server

bylderup, 2018-04-16:
This won´t work as discussed here :

http://www.dvbviewer.tv/forum/topic/61221-satip-server-%C3%BCber-tcp-verbinden/

Is it possible to to change satip in a way that Media Server shows the option to enable tcp ?

The developers of dvbviewer won´t change because they say thats not part of SAT>IP Protocol Specification Version 1.2.2

9000h, 2018-04-16:
as far as I know rtsp over tcp is broken atm, at least with VDR

perexg, 2018-04-16:
I've testing TCP data transfer only with tvheadend and it's rock stable. The VDR implementation have had always issues.

bylderup, 2018-04-17:
yes, with tvheadend it´s top. But dvbviewer does not recognize that tcp is possible. Only with internal satip.

perexg, 2018-04-17:
The client should ask for the TCP mode (it's in RFC2326 standard - https://www.ietf.org/rfc/rfc2326.txt - section 10.12). The server should deny or allow this transfer method after the appropriate SETUP command (Transport: RTP/AVP/TCP;interleaved=0-1). You may use a network analyzer, if dvbviewer is trying to use that command. If not - we cannot do much.

bylderup, 2018-04-17:
I will change from dvbviewer/Media Server i5 Notebook to an Android TV box with CoreELEC, 905D chipset, Gigabit Lan and TvHeadend.  Was astonished that such a device in combination with Digibit R1 can handle 4 HD Streams without problems.  Will save energy costs and no more MS Win10 updates.
Thanks for your efforts with satip-axe.

closed 2018-04-17 by bylderup