perexg/satip-axe #116: Jess and Minisatip5 not working

Misaal, 2018-04-28:
Hello
I recently switched to Jess (Jultec JPS0901-12A).
It runs minisatip5 version satip-axe-201605311610-12

There are still 4 frequencies open in the Jess multiswitch (1790MHz, 1892MHz, 1994MHz, 2096MHz).

Unfortunately, this setting does not work:
MINISATIP_OPTS = "- X 0"

Any tips?

Best regards

perexg, 2018-04-30:
https://github.com/perexg/satip-axe/blob/master/dist/minisatip.md#unicableunicable-ii

Misaal, 2018-05-02:
Hi perexg
Thanks for the link. I think I understood it.
Could this setting work or are there too many parameters?
MINISATIP8_OPTS="-j 0:9-1790,1:10-1892,2:11-1994,3:12-2096 -A 0:0:0,1:0:1,0:1:0,1:1:1,0:2:0,1:2:1,0:3:0,1:3:1 -7 0:1,0:2,0:3 -P 1"

perexg, 2018-09-17:
Add log from '-l general,http,stream,adapter,axe' . Use ssh command like from a linux machine:

`ssh -t root@${SATIP_HOST} minisatip8 $MINISATIP8_OPTS | tee out.txt`

Kill the standard daemon before:

`touch /tmp/nosatip; killall -9 minisatip8`

perexg, 2018-10-17:
Closing, no further discussion.

closed 2018-10-17 by perexg