perexg/satip-axe #119: multiple SAT>IP servers on net

nickiman, 2018-09-06:
Hi,
Regarding the Warning of multiple satip servers on net... described here:
https://github.com/perexg/satip-axe/tree/master/dist

Wich should be the correct configuration in config file **/etc/sysconfig/config**... something like this?
# minisatip 0.7
MINISATIP7="yes"
MINISATIP7_OPTS="-u 0:1-1420,1:0-1210,2:2-1680,3:3-2040 **-d 1**"

and for another satip... can we follow it with option **-d 2** for example?

I´m asking that because I have not clear the unique numbers meaning...

Thanks

9000h, 2018-09-06:
it's
-D --device-id DVC_ID: specify the device id (in case there are multiple SAT>IP servers in the network)
but this is optional as the discovery protocol will discover other ones on the net and will avoid conflicts automatically if all went right

perexg, 2018-09-06:
I've not checked or reviewed the current code, but there was issue with the endless loop when there were two minisatip servers and both tried to allocate own ID. Perhaps, it's fixed now. Anyway, it's much safer to allocate static number per server (like IP address).

nickiman, 2018-09-10:
Hi,
Thanks @9000h and @perexg for your feedback.
Actually the boxes has the IP address "assigned" via DHCP reservation in the router. 

The issue was opened to know if we could avoid any problem in the network for having two boxes running in the same LAN since the WARNING is still present in the Minisatip8 (0.7.15) config.

> it's
-D --device-id DVC_ID: specify the device id (in case there are multiple SAT>IP servers in the network)
but this is optional as the discovery protocol will discover other ones on the net and will avoid conflicts automatically if all went right

@9000h The boxes are still running minisatip 5 (that´s why I´m referring to -d instead **of -D**), maybe we will update them to the latest minisatip supported in the future but, we are considering that...

But before, I would like clarify with both of you which should be the parameter to be written in the minisatip config line because is still not clear for me...
...is the mac address of the box, the ip address assigned or just a simple number??

And which should be the configuration in the minisatip_OPTS line, something like this for example?:

MINISATIP5_OPTS="**-d 1** -u 0:1-1420,1:0-1210,2:2-1680,3:3-2040"

Thanks for your help and best regards.

perexg, 2018-09-17:
There is a lot confusion about this. The really old minisatip binary uses '-d' to specify the device number but all other minisatip binaries (5,7,8) use '-D'. Fixed README.

nickiman, 2018-09-25:
Thanks @perexg for clarify and update the documentation.