perexg/satip-axe #121: No boot after update

Themesaul, 2018-09-23:
Hello, I had a version of 2017 working correctly. It was in Flash memory not in USB. After updating and writing "reboot" it does not start. Only the power red light turns on and nothing else. I have tried everything in your reset guide. I've also tried flashing the official firmware, but it does not work. Only the power light is turned on and nothing else.

I have used several different sUSB memories (including the one I used to flash the 2017 memory) and I used both ports (especially the upper port)

What happened? Can it be fixed?

Thank you very much

perexg, 2018-09-23:
It looks like that uboot does not work or is misconfigured. I assume that you tried to copy idl4k.recovery file as idl4k.scr file to the root directory on the USB stick (and optionally reflash the original firmware).

The serial connection to the uboot console might help to detect what's wrong.

Themesaul, 2018-09-24:
Exactly, I tried to do that. Nothing has worked, it is as if that ignored "idl4k.src".I'll look into connecting Uboot by serial connection, although the truth is that I do not have anything clear right now, I'll have to investigate.Thank you very much, I will keep this updated according to my steps in case I can help someone else

Themesaul, 2018-09-25:
I was doing tests all day. I have not managed to connect to Uboot. Could you give me some idea of ​​how to do it? Many thanks

Themesaul, 2018-09-25:
I have verified that my USB TTL does not work properly. I have bought another one, when it arrive I will try it and I will comment on any progress. Thank you very much for the help and your quick response. 😁

Themesaul, 2018-09-28:
Hi, I've been trying to recover the system but I could not. Could you give me some advice?

```console
idl4k>printenv
bootdelay=1
baudrate=115200
board=idl4k
monitor_file_name=u-boot.bin
load_addr=0x84000000
autostart=yes
silent=1
resetenv=eeprom write $load_addr 0x00080000 1000
updatefw=usb start;set filesize;fatload usb 0 $load_addr idl4k.scr;if test "${filesize}" != ""; then autoscr $load_addr; fi
bootargs=
mtdids=nand0=gen_nand.1
mtdparts=mtdparts=gen_nand.1:128k(env-nand0),4M(system-nand0),32M(fw1-nand0),32M(fw2-nand0),-(data-nand0)
filesize=C54D4D
partition=nand0,0
mtddevnum=0
mtddevname=env-nand0
bootcmd=run updatefw;nboot.jffs2 $load_addr fw2-nand0
satip-axe=1
ethaddr=00:24:A6:XX:XX:XX
stdin=serial
stdout=serial
stderr=serial

Loading from NAND 128MiB 3,3V 8-bit, offset 0x2420000
   Image Name:   Linux-2.6.32.42_stm24_0208-idl4k
   Image Type:   SuperH Linux Kernel Image (gzip compressed)
   Data Size:    13346334 Bytes =  12.7 MiB
   Load Address: 81000000
   Entry Point:  81001000
Automatic boot of image at addr 0x84000000 ...
   Image Name:   Linux-2.6.32.42_stm24_0208-idl4k
   Image Type:   SuperH Linux Kernel Image (gzip compressed)
   Data Size:    13346334 Bytes =  12.7 MiB
   Load Address: 81000000
   Entry Point:  81001000
   Verifying Checksum ... Bad Data CRC
```

It also does not detect Sat-Axe on the USB to reinstall it:

```console
idl4k> usbboot

** No boot device **
```

And I do not get IP address to start by network

```console
idl4k> dhcp
Using MAC Address 00:24:A6:XX:XX:XX
STM-GMAC: Unable to find a known PHY!
STM-GMAC: info: PHY at address=0x00 has ID=0x001cc915
STM-GMAC: info: PHY at address=0x01 has ID=0x001cc915
STM-GMAC: ERROR: no PHY detected
```

Any help would be great. I've been trying hard but I have not.

Thanx you 😄

perexg, 2018-09-28:
What does show 'run updatefw' when you insert the USB stick with the satip-axe firmware (and correct idl4k.scr file)? Run it from the cold boot (power on), otherwise USB does not work in uboot.

Themesaul, 2018-09-29:
When I run "run updatefw" I see that it detects USB but can not read the .src file. The USB I use is the same one I used to flash the 2017 version and it went well.

```console
idl4k> run updatefw
(Re)start USB...
USB:   scanning bus for devices... 2 USB Device(s) found
       scanning bus for storage devices... 1 Storage Device(s) found
reading idl4k.scr

** Unable to read "idl4k.scr" from usb 0:1 **
```
Here the content of the USB:

```console
root@debian:/mnt/usb# tree
.
├── idl4k.scr
└── satip-axe-201809242038-15.fw

0 directories, 2 files
```

What do you recomend?

Thanx you so much!

perexg, 2018-10-01:
The good thing is that the USB hardware seems to be working properly. There are several uboot filesystem commands which might you use for the debugging, run 'usb start' before any of them to start the usb subsystem in uboot:

https://mediawiki.compulab.com/w/index.php?title=U-Boot:_Quick_reference#File_system_support_commands

I would try to reformat the flash to FAT (mkfs.vfat).

Themesaul, 2018-10-02:
RECOVERED!!

I formatted the usb in vfat WITHOUT any partition and fatls and fatload have worked.

The device has a system. Its alive!

Thank you very much for your help. Thank you very much for this wonderful software and for keeping it.

😍😍😍🍿

perexg, 2018-10-08:
I am glad to hear this :-) Closing (and fixing the first answer 'try reformat flash' in my head).

closed 2018-10-08 by perexg