perexg/satip-axe #123: DLNA server available?

okurz, 2018-09-30:
I have tried out satip-axe and it works fine so far with tvheadend. However, with the original firmware a "miniDLNA Server" was included which could be activated/deactivated over the webUI. I assume this is the component that publishes the TV channel I can see over UPNP in VLC. Is there such a component/possibility in satip-axe?

perexg, 2018-10-01:
No, there is no dlna server in the satip-axe firmware. In my opinion, it's better to serve .m3u playlist through the available HTTP server to clients.

okurz, 2018-10-01:
Thanks for the fast and helpful answer. Do you have a reference ready how I can come up with a .m3u playlist in the most efficient way?

9000h, 2018-10-01:
http://www.satip.info/resources?tid=23

perexg, 2018-10-01:
The HTTP server on port 80 (standard) in the satip-axe firmware serves content from volatile /media directory, so you may copy contents on boot there or make a link to a non-volatile directory (/mnt/data or /etc/sysconfig for example). The script /etc/sysconfig/localboot is called on each boot.

Another way is to insert a flash disk - it will be mounted to /media tree automatically.

okurz, 2018-10-01:
thx, great. I understood you want to keep the issue open as a valid enhancement proposal so I will not close it :)

9000h, 2018-10-11:
http://xupnpd.org could be a candidate