perexg/satip-axe #126: Multistream support

perexg, 2018-10-11:
Appearently STV0900 demodulators with chip id 0x30+ supports multistream.

The axe hardware has chip id 0x30:

<pre>
~ # axehelper i2c_reg_read 0xd0 0xf100
0x30
</pre>

**Linux kernel code:**

https://github.com/torvalds/linux/blob/64c5e530ac2cab7e710c3bd08ef8594a016e4f9a/drivers/media/dvb-frontends/stv090x.c#L3433-L3474

https://linux-media.vger.kernel.narkive.com/RyQdvAhf/patch-stv0900-multistream-support

**Goals:**

- inject the current driver to support it
  - old DVBAPI (use another way to control PLS/MIS filter - sysfs or proc)
- extend minisatip to support PLS/MIS settings

perexg, 2018-10-12:
Ok, build 15-rc5 is out with the multistream support, tvh supports the newly added SAT>IP extension, too.

perexg, 2018-10-12:
Build 15-rc6 supports both GOLD and ROOT PLS schemes (with latest tvh).

perexg, 2018-10-13:
It seems that there's just one PLS scrambling algorithm and ROOT/GOLD just specifies a different way to get the starting value for the descrambler:

https://github.com/DigitalDevices/dddvb/blob/master/apps/pls.c

The COMBO seems like a bug..

perexg, 2018-10-13:
BTW: https://github.com/oe-alliance/satip-client/pull/10 , even enigma2 on my openpli can do multistream through SAT>IP protocol now !!!!

gpoerzgen, 2018-10-16:
Is it possible, that you provide a satipclient package for the vu solo 2?

perexg, 2018-10-16:
@gpoerzgen : http://s000.tinyupload.com/?file_id=42666432231835511189

perexg, 2018-10-16:
Note that if you like to activate the TCP data transport add tcpdata=1 option like:

<pre>
0=desc:minisatip,ipaddr:192.168.100.237,tuner_type:DVB-S,uuid:11223344-9999-0000-b7ae-0022b7f0056e,vtuner_type:satip_client,tcpdata:1
</pre>

gpoerzgen, 2018-10-16:
Thanks, I will test it tonight

perexg, 2018-10-17:
Build15 final is out with the multistream support.

closed 2018-10-17 by perexg