perexg/satip-axe #127: Build 16

perexg, 2018-10-17:


perexg, 2018-10-17:
Fix the upgrade-fw (remote files).

doraid131, 2018-10-18:
tvheadend dont recognize satip anymore with the update firmaware .....

perexg, 2018-10-18:
Do cold boot (turn off, wait 30 seconds, turn on).

estimadarocha, 2018-10-18:
@perexg i am confirming the behavior but i am almost 80% sure that it really happens.

I have one network "dvb-s" that uses 4 tuner's of satipaxe and gives me around 100 channels.

I have other networks "dvb-c" and "dvb-t" that have exactly same channels so channels with same name are merged together.

I put highter priority on satipaxe tuner's so that i am sure that subscribers will pick first all the channels from satipaxe until there's no more available tuners.

i restart tvheadend to make sure the clients subscribing will follow priority rules...and it happens... i see in few seconds a few channels being picked from satipaxe.... however and the bad behavior appears now slowly the subscriptions start to move from the satipaxe to other networks. I didn't yet realize if this is caused by own minisatip from satipaxe, own tvheadend, or something wrong on descrambling.

So i am adding a "iptv network" where the mux's will be http stream's from satipaxe and try to figure if same happens!

perexg, 2018-10-18:
@estimadarocha : report this for tvheadend. It has nothing to do with minisatip/satip-axe.

estimadarocha, 2018-10-18:
@perexg thanks for the quick answer!

estimadarocha, 2018-10-18:
@perexg is there anyway to see error's already happening or do i have to kill minisatip and start it again with log enabled? 

Got it stucked on 2 of 4 tuner's, tuner 0 and 2 stop responding to http streaming requests!

[image]

phaze75, 2018-10-21:
oscamd is missing.

perexg, 2018-10-21:
Any multiple access to the transponder stream consts too much time, because the stream must be parser per one MPEG-TS packet and split to different clients:

https://github.com/perexg/minisatip/blob/axe8/src/stream.c#L990-L992

So, for the optimal performance, use only one client per tuner and do the stream splitting somewhere else (install minisatip proxy on a more powerful machine for example). The SH4 CPU on this hardware is really slow.

estimadarocha, 2018-10-22:
@perexg so if i grab a complete mux pids=all its a single access, correct?

meijkl, 2018-10-22:
Is it an option to include Midnight Commander and htop in build 16?

perexg, 2019-10-08:
https://github.com/perexg/satip-axe/issues/155

closed 2019-10-08 by perexg