perexg/satip-axe #130: CC Errors

estimadarocha, 2018-10-26:
Hi @perexg:

do you also have these kind of issues?

I know i am pulling a lot (tvheadend) but i already played with STV6210_BAMP values (8 now) and even so CC Errors are always incresing... cpu is around 60%. 

Signal is perfect if i connect cables direct on enigma box!

[image]

estimadarocha, 2018-10-26:
with STV6210_BAMP on 16

[image]

estimadarocha, 2018-10-29:
[image]

i am using satipaxe to tvheadend on same network and forcing play of 15 channels, one of them is UHD!

there's somethings working bad...  

1. The cc errors always increasing
2. Sometimes when i get cc errors and the client thats is pushing from tvheadend stops and restart's i cannot recover to a working state. (picture) Only solution is restart satipaxe or own tvheadend.
3. To make this work i had to disable fe support (image 2)

[image]

estimadarocha, 2018-10-29:
[minisatip.log](https://github.com/perexg/satip-axe/files/2525213/minisatip.log)

estimadarocha, 2018-10-29:
start point:
[image]
[image]
after 30mins:
(to be updated)

estimadarocha, 2018-10-29:
stucked again:

[image]

perexg, 2018-10-29:
Try to find a minimal configuration where things hangs. Do not unset 'FE supported' - otherwise the problem is just migrated to another tuner. Provide minisatip logs for 'general,http,stream,adapter,dvb,axe' subsystems. I have very little time to look into, so it depends on your description quality, if I can fix something.

Anyway high tt values means usually that minisatip is under high load.

estimadarocha, 2018-10-29:
so.... back to work...

i will set again FE supported (but if i remember well with that option unset i was not able to run it) i will look further... give me some mins!

catalinii, 2018-10-29:
Hi,

Just commenting on a trend here:

1) at 15M tt is over 1000ms (which would mean cpu is at 100%)
2) the CC errors is what the hardware reports (not minisatip)
3) the number of cc errors (in percentage) are very low.

Thanks

estimadarocha, 2018-10-29:
65% last time I checked... Or am I checking wrong?

catalinii, 2018-10-29:
My point was that cpu spikes to 100%, the question is can you reproduce the issue lets say with few channels under 7MB/s (15M/2)? 

Basically we would like to know if the issue happens not because potentially the CPU is overloaded....

PS. I never had an axe myself so I cannot comment on the CPu capabilities. I had an amiko with similar CPU the fact that it can get to 15M shows how much work @perexg has put into it.

estimadarocha, 2018-10-29:
I will see with half of bw. Regarding the cc errors everytime one new stream is added do you thinks that's normal? I don't think I see same behavior on minisatip... Maybe a specific axe problem!

perexg, 2018-10-29:
Try '-W 1'.

estimadarocha, 2018-10-29:
Tried already

estimadarocha, 2018-10-31:
so 1st log....

3 streams running and when increasing one by one.... satipaxe stops suddendly streaming

[minisatip.log](https://github.com/perexg/satip-axe/files/2535051/minisatip.log)

catalinii, 2018-10-31:
The stream stopped because of this:
[31/10 15:17:19.945 main]: read RTSP (from handle 16 sock 10, len: 0, sid 0):
[31/10 15:17:19.945 main]:
[31/10 15:17:19.946 main]: Most likely not an RTSP packet sock_id: 10 sid: 0 rlen: 0, dropping ....
[31/10 15:17:19.946 main]: select_and_execute[10]: Close on socket 16 (sid:0) from 192.168.1.200:56420 - type rtsp errno 0
[31/10 15:17:19.946 main]: Requested stream close 0 timeout 30000 type 3
[31/10 15:17:19.946 main]: closing stream 0

Which I assume is because for some reason the tvh closed the connection. I would check tvh logs and also make sure the client (kodi ?) did not close the connection to tvh.

perexg, 2018-10-31:
@catalinii : tvh always sends teardown when the client closes the streaming. It looks like a serious issue somewhere.

perexg, 2018-10-31:
@estimaradocha: This minisatip8 binary should fix the rlen = 0 log issue : http://s000.tinyupload.com/?file_id=85686441378171183817

Anyway, you should check, who/what breaks the TCP RTSP connection.

catalinii, 2018-10-31:
Yes but the socket is closed (read returns 0) not some connection issue
where read would return -1

perexg, 2018-10-31:
@catalinii : I do not see confirmed read() == 0 in the log. The "read RTSP (from handle 16 sock 10, len: 0, sid 0)" prints always zero because of this bug (the last change): https://github.com/catalinii/minisatip/pull/473 .

estimadarocha, 2018-11-01:
@perexg i saw you made a new commit https://github.com/catalinii/minisatip/pull/474/commits/215747c826b428878ac7a89b8c1016403558d997 the last binary link already have all the fixes?

Thanks for everything

perexg, 2018-11-01:
logs from minisatip ?

estimadarocha, 2018-11-01:
I will grab them tonight when my son goes to bed... And paste them here

estimadarocha, 2018-11-02:
[minisatip1.log](https://github.com/perexg/satip-axe/files/2541765/minisatip1.log)

i got a lot of testing / bad on tvh on the startup let's see if log have something

estimadarocha, 2018-11-02:
[image]

most of the time (90%) i have less than 600ms on tt, but time to time it spikes to 1000-1100 with exact same channels playing and without any operation (add/subtract channels).

[image]

i don't see any variation on cpu % when tt spikes

[image]

is it normal so have such variation on constant play?

perexg, 2018-11-02:
Ok, so we have confirmed the socket close (read() == 0):

<pre>
[02/11 08:55:03.010 main]: read RTSP (from handle 17 sock 12, len: 0, sid 1):
[02/11 08:55:03.010 main]: 
[02/11 08:55:03.011 main]: Most likely not an RTSP packet sock_id: 12 sid: 1 rlen: 0, dropping ....
[02/11 08:55:03.012 main]: select_and_execute[12]: Close on socket 17 (sid:1) from 192.168.1.200:40532 - type rtsp errno 0
[02/11 08:55:03.012 main]: Requested stream close 1 timeout 30000 type 3
[02/11 08:55:03.012 main]: closing stream 1
</pre>

I would start with this. Who or what closes the socket? Could you show tvh log and minisatip log around time what you see 'read RTSP .. len: 0 ... closing stream' lines? Also, tcpdump/wireshark RTSP TCP port 554 socket analysis might help.

estimadarocha, 2018-12-06:
have some time now to grab logs... what traces do you need from the tvh side?

perexg, 2018-12-06:
tvh: --trace satip,mpegts

estimadarocha, 2018-12-06:
tcpdump something like this:

tcpdump -i ens3 -nn -s0 -v port 554??

perexg, 2018-12-06:
write the output to a file for the further analysis: -w /filename/

perexg, 2018-12-06:
@9000h : the kernel drivers are really different than used on the enigma2 machines and there's 4MB ring buffer per MPEG-TS input. I don't know about this check in axe drivers (the hardware CC errors /reception,signal decoding/ are displayed in minisatip webif for axe).

perexg, 2019-01-09:
@estimadarocha : The fix is only for logging, it does not affect the operation. You stopped providing of logs and try to determine who closes the TCP connection, so I don't care.

estimadarocha, 2019-01-09:
> @estimadarocha : The fix is only for logging, it does not affect the operation. You stopped providing of logs and try to determine who closes the TCP connection, so I don't care.

i tought it was something that affects the operation... about logging i tought i would have time on past december but i was not sucessfull.... much because i need to add some external storage maybe... i run out of space in few time.

small apart is there any limitation of pids for total amount of adapters? i mean... for example can i get pids=all at same time on all adapters (internally of course not wan)?

9000h, 2019-01-09:
there is also another open issue in minisatip, where http/rstp reply's get messed up
```
[09/01 12:36:07.053 main]: RTSP/1.0 404 Not Found
Date: Wed, Jan 9 12:36:07 2019 GMT
Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
Server: minisatip/0.7.16-axe215
```
https://github.com/catalinii/minisatip/issues/517

estimadarocha, 2019-01-09:
that ip ... 128.1.84.138 is from outside maybe really a wrong call, possible?

estimadarocha, 2019-01-09:
@perexg i am using the old tweaked settings from you "-b 4042752 -B 10000" it seens like it exausts memory to limit and them minisatip8 crash, is it possible?

catalinii, 2019-01-09:
@estimadrocha,

That means that every adapter will allocate up to 4042752 *10000 bytes if there is not enough network bandwidth. If minisatip would crash the memory would be freed and you would be able to use the system again, but from my tests minisatip does not crash when out of memory.

Use lower memory settings based on available memory and number od adapters (4 for axe).

Eg if you have free 100M, you need 25M for each adapter which means -b 188000 -B 140.

-b should be multiple of 188.

estimadarocha, 2019-01-09:
@catalinii thanks... First i need only 2 adapters I can't get more than 2 adapters with pids=all because the CPU Max's out. So I need to split the free space for only 2 sockets. What's better use -b and -B or for example use only a bigger value on -b?

Thanks again

catalinii, 2019-01-09:
Bigger -b might be better.

estimadarocha, 2019-01-10:
i set -b to 1880000 and no -B

if i look at memory the consume its always slowly increasing... doesn't buffers get cleared sometime?

~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        189        202          0          0         23
-/+ buffers/cache:        166        225
Swap:            0          0          0
~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        189        202          0          0         23
-/+ buffers/cache:        166        225
Swap:            0          0          0
~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        189        202          0          0         23
-/+ buffers/cache:        166        225
Swap:            0          0          0
~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        191        201          0          0         23
-/+ buffers/cache:        167        224
Swap:            0          0          0
~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        191        201          0          0         23
-/+ buffers/cache:        167        224
Swap:            0          0          0
~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        191        200          0          0         23
-/+ buffers/cache:        167        224
Swap:            0          0          0
~ # free -m
             total       used       free     shared    buffers     cached
Mem:           392        191        200          0          0         23
-/+ buffers/cache:        167        224
Swap:            0          0          0

9000h, 2019-01-10:
you set ? 
> i set -b to 1880000 and no -B
@catalinii did suggest

> Eg if you have free 100M, you need 25M for each adapter which means -b 188000 -B 140.

estimadarocha, 2019-01-10:
@9000h and them he said 

> Bigger -b might be better.

so i tried last option

estimadarocha, 2019-01-10:
[image]

after 6 hours playing 2 full mux's not even a single CC (that's impossible when adding/removing pids).

memory 

[image]

9000h, 2019-01-10:
I'm unsure if the CC counter is right under all conditions.

perexg, 2019-01-10:
Unless the client does not see any issues (video/audio CC errors / corruptions), it's fine. We don't know the exact metering of those CC errors. Perhaps, the CC error increases when the PID filter is changed, which might not be a problem. It should not increase massively.