perexg/satip-axe #131: Could be LNB parameters (-L --lnb (low,high and switch frequency)) be added for wide band LNBs

beallwayscreative, 2018-11-01:
Hi,
thank you for the great alternative firmware!

I have a wide band uni-cable LNB: Kathrein UAS481 and I can not get it working.

Kathrein UAS481 LNB specs:

Input frequency: 10700 – 12750 MHz

Supply voltage LNB (input): 11.5-19 (power supply: 11.5-14: control: 16-19 & DiSEqC acc. To EN 50494:2006)

ID1 (SCR0): 1400 MHz
ID2 (SCR1): 1516 MHz
ID3 (SCR2): 1632 MHz
ID4 (SCR3): 1748 MHz

LOF: 10200 MHz (for low- and high-band)

I would be great if you could add this Minisatip feature:

    • -L --lnb specifies the adapter and LNB parameters (low, high and switch frequency)
        ◦ eg: -L *:9750-10600-11700 - sets all the adapters to use Universal LNB parameters (default) 
        ◦ eg: -L *:10750-10750-10750 - sets the parameters for Sky NZ LNB using 10750 Mhz 
        ◦ eg: -L 0:10750-10750-10750,1:9750-10600-11700 - adapter 0 has a SKY NZ LNB, adapter 1 has an Universal LNB

to your firmware or did I miss something.

Thank you so much and have a nice day.

perexg, 2018-11-01:
It's standard unicable LNB, isn't? Use '-u'.

<pre>
* -u --unicable unicable_string: defines the unicable adapters (A) and their slot (S), frequency (F) and optionally the PIN for the switch:
	* The format is: A1:S1-F1[-PIN][,A2:S2-F2[-PIN][,...]]
	eg: 2:0-1284[-1111]
	* When * character is used before frequency, force 13V only for setup
</pre>

It's already covered in https://github.com/perexg/satip-axe/blob/master/dist/README#L109 .

beallwayscreative, 2018-11-01:
Hi Jaroslav,

thank you for answering so quickly. Unfortunately my LNB is a wide band one like the Sky one and not a standard one. It requires individual LNB inputs for low and high (LOF is 10200).

My last satip-axe configure attempt was:

# minisatip 0.8

MINISATIP8_OPTS=”-L *:10200-10200-10200 -u 0:1-1400,1:0-1516,2:2-1632,3:3-1748”

However, it did not work and I realized that there is no -L option like in minisatip. That is way, I hope that you could include a functional patch in your wonderful firmware.

Thank you for helping us and have a nice evening

perexg, 2018-11-01:
The -L option should work. You can check this in the log. There's an example in the Kathrein UAS481 datasheet:

ASTRA 19,2° 11,836 GHz, H-Pol. - DiseqC E0 10 5A 0D 99

The minisatip8 calculator:

<pre>
freq=11836-10200
ufreq=1400
((freq + ufreq + 2) / 4 - 350) = 0x199

+ hiband |= 0x400
+ pol |= 0x800

result = 0xd99 (last two bytes, perfectly ok)
</pre>

Try minisatip8 with ”-L *:10200-10200-10200 -u 0:0-1400,1:1-1516,2:2-1632,3:3-1748”, you're trying to shift slots (copied probably from the example without checking).

beallwayscreative, 2018-11-02:
Hi, and thank you so much for the big hint and for checking the plausibility of the input by looking up the specs. Indeed, I copied the -u 0:0-1400,1:1-1516,2:2-1632,3:3-1748 “string” format without much thought. However, there is still no signal nor SDR. I checked the LNB with the current Digibit R1 firmware and no signal can be detected here as well. Maybe the LNB is broken since all wiring is OK.
Is there a way to check the uni-cabel LNB out without getting a sat-receiver?

Have a great day and thank you!

closed 2018-11-02 by beallwayscreative