perexg/satip-axe #134: Multistream on Digibit R1?

tipexfm, 2018-12-02:
Hello!

I've read on the README file that DVB-S2 multistream support is added on the latest update. As I don't see any multicast reference on the official Digibit R1 specifications, I would like to ask; Is multicast supported on the R1?

Thank you!

9000h, 2018-12-02:
https://github.com/perexg/satip-axe/blob/master/tools/multicast-rtp

perexg, 2018-12-09:
This tool is available as package (fetch-package). Use multicast-rtp-1.tar.gz - the v2 is broken.

closed 2018-12-09 by perexg

tipexfm, 2018-12-09:
Hello all and thank you for the answers,

I'm sorry, I misspelled the word. I wanted to know if **multistream** (DVB_S2) is receivable with satip-axe on a Digibit R1.

Btw it's great that it can use multicast! It can be very useful for large environment distribution.

Thanks again!

tipexfm, 2018-12-09:
aaaah find myself the answer on a thread below. 

A big THANK YOU for this software!