perexg/satip-axe #135: Minisat server not working after reboot

scho0ck, 2018-12-30:
I'm new to minisat and this firmware but I'm having the issue that once I reboot my Digibit R1 I'm unable to tune to any MUX and the webinterface with the satip.xml and the stats is offline as well I then have to kill the minisat process and restart it in order to make it work however the web crashes after a while again but then at least I'm able to watch TV.

Using the latest fix1 firmware  with the parameters "-e 0-1 -O" logread and dmesg show no error at all?

scho0ck, 2018-12-30:
This is my config 

`# minisatip 0.8
#
MINISATIP8="yes"
MINISATIP8_OPTS="-e 0-1 -O"`

Dmesg
`Linux version 2.6.32.42_stm24_0208-idl4k_7108 (perex@stlinux.perex-int.cz) (gcc version 4.5.3 20111123 (STMicroelectronics/Linux Base 4.5.3-99) (GCC) ) #134 PREEMPT Sun Oct 21 15:49:45 CEST 2018
Boot params:
... MOUNT_ROOT_RDONLY - 00000001
... RAMDISK_FLAGS     - 00000000
... ORIG_ROOT_DEV     - 00000200
... LOADER_TYPE       - 00000000
... INITRD_START      - 00000000
... INITRD_SIZE       - 00000000
Booting machvec: idl4k
Node 0: start_pfn = 0x70000, low = 0x8f000
Zone PFN ranges:
  Normal   0x00070000 -> 0x0008f000
Movable zone start PFN for each node
early_node_map[1] active PFN ranges
    0: 0x00070000 -> 0x0008f000
On node 0 totalpages: 126976
free_area_init_node: node 0, pgdat 8145feb8, node_mem_map 82b4d000
  Normal zone: 992 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 125984 pages, LIFO batch:31
Inverto idl4k board initialisation
STx7108 version 2.x, HOST core
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125984
Kernel command line:  console= hw=16001000000000000000000000000000 sn=1S81T51717000269 uuid=5ADA6D378BCB48A09635CC561E2EB80F revBoot=stm24_0056-idl4k console=ttyAS0,115200 bigphysarea=20000
bpa2: partition 'bigphysarea' created at 0x72f31000, size 80000 kB (0x04e20000 B)
PID hash table entries: 2048 (order: 1, 8192 bytes)
Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
PVR=04909300 CVR=60880000 PRR=0000a320
I-cache : n_ways=2 n_sets=512 way_incr=16384
I-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
D-cache : n_ways=2 n_sets=512 way_incr=16384
D-cache : entry_mask=0x00003fe0 alias_mask=0x00003000 n_aliases=4
Memory: 379008k/507904k available (3385k kernel code, 1095k data, 22344k init)
Hierarchical RCU implementation.
NR_IRQS:500
Failed to enable clk CLKA0_REF, ignoring
Failed to enable clk CLKA0_PLL0LS, ignoring
Failed to enable clk CLKA1_REF, ignoring
Failed to enable clk CLKA1_PLL0LS, ignoring
Console: colour dummy device 80x25
console [ttyAS0] enabled
sh_tmu: TMU0 used for clock events
sh_tmu: TMU0 used for periodic clock events
sh_tmu: TMU1 used as clock source
Calibrating delay loop... 595.96 BogoMIPS (lpj=297984)
Mount-cache hash table entries: 512
CPU: STx7108
NET: Registered protocol family 16
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
Switching to clocksource TMU1
NET: Registered protocol family 2
IP route cache hash table entries: 4096 (order: 2, 16384 bytes)
TCP established hash table entries: 16384 (order: 5, 131072 bytes)
TCP bind hash table entries: 16384 (order: 4, 65536 bytes)
TCP: Hash tables configured (established 16384 bind 16384)
TCP reno registered
NET: Registered protocol family 1
[... 198 lines cut ...]
[dms-ts] stream stop ok
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 1
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 1
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
`

perexg, 2018-12-30:
Have you tried to boot without your minisatip8 options? MINISATIP8_OPTS=""

scho0ck, 2018-12-30:
> Have you tried to boot without your minisatip8 options? MINISATIP8_OPTS=""

Yes that gives me the same error that's why I actually added the OPTS in the hope it would fix my issue.

perexg, 2018-12-30:
Create /tmp/nosatip file (touch /tmp/nosatip), kill the minisatip task and run minisatip8 from the ssh console:

  minisatip8 -f -D 88 -l general,http,stream,adapter,dvb,axe

scho0ck, 2018-12-30:
Okay this is what I'm getting

```
minisatip8 -f -D 88 -l general,http,stream,adapter,dvb,axe
[30/12 22:01:16.910 main]: minisatip version 0.7.16-axe215, compiled with s2api version: 0501
[30/12 22:01:16.910 main]: Built without dvbcsa
[30/12 22:01:16.911 main]: Built without CI
[30/12 22:01:16.911 main]: Built without dvbapi
[30/12 22:01:16.911 main]: Built without AES (OpenSSL)
[30/12 22:01:16.911 main]: Built without tables processing
[30/12 22:01:16.912 main]: Built without pmt processing
[30/12 22:01:16.912 main]: Built with satip client
[30/12 22:01:16.912 main]: Built with linux dvb client
[30/12 22:01:16.912 main]: Built with backtrace
[30/12 22:01:16.912 main]: Built without netceiver
[30/12 22:01:16.912 main]: Built for satip-axe
[30/12 22:01:16.913 main]: New UDP socket 3 bound to 0.0.0.0:1900 
[30/12 22:01:16.913 main]: setting multicast for 239.255.255.250
[30/12 22:01:16.914 main]: New UDP socket 4 bound to 239.255.255.250:1900 (mcast:239.255.255.250)
[30/12 22:01:16.915 main]: Mutex init 0x4425a0
[30/12 22:01:16.915 main]: Mutex init 0x45a45c
[30/12 22:01:16.915 main]: sockets_add: handle 3 (type 0) returning socket sock 0 [0.0.0.0:0] read: 0x407b80
[30/12 22:01:16.915 main]: Mutex init 0x45a8d4
[30/12 22:01:16.915 main]: sockets_add: handle 4 (type 0) returning socket sock 1 [0.0.0.0:0] read: 0x407b80
[30/12 22:01:16.916 main]: Mutex init 0x45a9d4
[30/12 22:01:16.916 main]: sockets_add: handle 5 (type 2) returning socket sock 2 [0.0.0.0:0] read: 0x409d14
[30/12 22:01:16.916 main]: Mutex init 0x45aad4
[30/12 22:01:16.916 main]: sockets_add: handle 6 (type 2) returning socket sock 3 [0.0.0.0:0] read: 0x409d14
[30/12 22:01:16.916 main]: Mutex init 0x45abd4
[30/12 22:01:16.916 main]: sockets_add: handle -2 (type 0) returning socket sock 4 [0.0.0.0:0] read: 0x407b80
[30/12 22:01:16.917 main]: set_socket_thread: thread 29ec9328 for sockets 4
[30/12 22:01:16.917 main]: Mutex init 0x45ad6c
[30/12 22:01:16.917 main]: sockets_add: handle -2 (type 0) returning socket sock 5 [0.0.0.0:0] read: 0x407b80
[30/12 22:01:16.918 main]: set_socket_thread: thread 29ec9328 for sockets 5
[30/12 22:01:16.918 main]: starting init_all_hw 0
[30/12 22:01:16.918 main]: Mutex init 0x442dcc
[30/12 22:01:16.919 signal]: Starting select_and_execute on thread ID 29ec9328, thread_name signal
[30/12 22:01:16.960 main]: testing device /dev/axe/frontend-0 -> fd: 8
[30/12 22:01:16.962 main]: testing device /dev/axe/frontend-1 -> fd: 8
[30/12 22:01:16.964 main]: testing device /dev/axe/frontend-2 -> fd: 8
[30/12 22:01:16.966 main]: testing device /dev/axe/frontend-3 -> fd: 8
[30/12 22:01:16.966 main]: Mutex init 0x45ae6c
[30/12 22:01:16.966 main]: trying to open [0] adapter 0 and frontend 0
[30/12 22:01:16.971 main]: opened DVB adapter 0 fe:8 dvr:9
[30/12 22:01:16.972 main]: deleting pids on adapter 0, sid -1, pids=NULL
[30/12 22:01:16.972 main]: Delivery System DVB-S/DVB-S2 (AXE)
[30/12 22:01:16.972 main]: Mutex init 0x473abc
[30/12 22:01:16.972 main]: sockets_add: handle 9 (type 5) returning socket sock 6 [0.0.0.0:0] read: 0x407b04
[30/12 22:01:16.973 main]: set_socket_thread: thread 2a6c9328 for sockets 6
[30/12 22:01:16.973 main]: done opening adapter 0 delivery systems: dvbs dvbs2 undefined undefined
[30/12 22:01:16.973 main]: FE 1 mapped to Adapter 0, sys dvbs2
[30/12 22:01:16.974 main]: Mutex init 0x45cafc
[30/12 22:01:16.974 main]: trying to open [1] adapter 1 and frontend 0
[30/12 22:01:16.975 AD0]: Starting select_and_execute on thread ID 2a6c9328, thread_name AD0
[30/12 22:01:16.979 main]: opened DVB adapter 1 fe:10 dvr:11
[30/12 22:01:16.979 main]: deleting pids on adapter 1, sid -1, pids=NULL
[30/12 22:01:16.979 main]: Delivery System DVB-S/DVB-S2 (AXE)
[30/12 22:01:16.980 main]: Mutex init 0x485664
[30/12 22:01:16.980 main]: sockets_add: handle 11 (type 5) returning socket sock 7 [0.0.0.0:0] read: 0x407b04
[30/12 22:01:16.980 main]: set_socket_thread: thread 2aec9328 for sockets 7
[30/12 22:01:16.980 main]: done opening adapter 1 delivery systems: dvbs dvbs2 undefined undefined
[30/12 22:01:16.981 main]: FE 1 mapped to Adapter 0, sys dvbs2
[30/12 22:01:16.981 main]: FE 2 mapped to Adapter 1, sys dvbs2
[30/12 22:01:16.981 main]: Mutex init 0x45e78c
[30/12 22:01:16.981 main]: trying to open [2] adapter 2 and frontend 0
[30/12 22:01:16.982 AD1]: Starting select_and_execute on thread ID 2aec9328, thread_name AD1
[30/12 22:01:16.986 main]: opened DVB adapter 2 fe:12 dvr:13
[30/12 22:01:16.986 main]: deleting pids on adapter 2, sid -1, pids=NULL
[30/12 22:01:16.987 main]: Delivery System DVB-S/DVB-S2 (AXE)
[30/12 22:01:16.987 main]: Mutex init 0x49720c
[... 49 lines cut ...]
[30/12 22:01:46.989 AD2]: Destroying mutex 0x45e78c
[30/12 22:01:46.989 AD2]: done closing adapter 2
[30/12 22:01:46.990 AD2]: sockets_del: sock 8 -> handle 13, sid -1, overflow bytes 0, allocated 0, used 0, unsend packets 0
[30/12 22:01:46.991 AD2]: sockets_del: 8 Last open socket is at index 9 current_handle 13
[30/12 22:01:46.991 AD2]: Destroying mutex 0x49720c
[30/12 22:01:46.995 AD3]: closing DVR socket 15 pos 9 aid 3
[30/12 22:01:46.995 AD3]: closing adapter 3  -> fe:14 dvr:15, sock:9, fe_sock:-1
[30/12 22:01:46.996 AD3]: AXE standby: adapter 3
[30/12 22:01:46.997 AD3]: adapter.c:1067: get_adapter returns NULL for adapter_id 3
[30/12 22:01:46.997 AD3]: adapter.c:837: get_adapter returns NULL for adapter_id 3
[30/12 22:01:46.997 AD3]: Destroying mutex 0x46041c
[30/12 22:01:46.997 AD3]: done closing adapter 3
[... cut ...]
[30/12 22:01:47.078 AD0]: add_join_thread: pthread 2a6c9328
[30/12 22:01:47.084 AD1]: No enabled sockets for Thread ID 2aec9328 name AD1 ... exiting 
[30/12 22:01:47.084 AD1]: add_join_thread: pthread 2aec9328
[30/12 22:01:47.091 AD2]: No enabled sockets for Thread ID 2b6c9328 name AD2 ... exiting 
[30/12 22:01:47.091 AD2]: add_join_thread: pthread 2b6c9328
[30/12 22:01:47.099 AD3]: No enabled sockets for Thread ID 2bec9328 name AD3 ... exiting 
[30/12 22:01:47.099 AD3]: add_join_thread: pthread 2bec9328
```

perexg, 2019-01-01:
I would try to determine the minisatip8 issue, because minisatip7 is really unmaintained right now. Could you try to activate the log (like before) and connect the client to the minisatip8? The log is without any activity. Also, put the complete log to http://pastebin.com or to another public place and write only URL here.

perexg, 2019-10-08:
No response, no support. You may try build 16 anyway: https://github.com/perexg/satip-axe/issues/155 .

closed 2019-10-08 by perexg