perexg/satip-axe #136: Mistake flashing

jlazkano, 2019-01-09:
Hello,

I make some mistakes flashing the firmware for the Digibit R1, here are the steps:

1. I change some configuration of MINISATIP_OPT and after the reboot, I can not access to the SSH.
2. Try to flash the last firmware, but I can not do it from USB. It doesn't responso to ping.
3. Install original firmware from USB. Works great.
4. Upgrade the "satip-axe-201810211549-15.fw" file from the original firmware web interface.

Now I can not flash nothing, no axe firmware no original firmware.

Is there any way to recover the box? I will really appreciate your help.

Kind regards.

perexg, 2019-01-16:
You have to re-flash the original firmware from USB. The uboot (the boot loader) should not be affected. Ensure that your USB stick has only one FAT partition (or also try to place the FAT filesystem to the stick without partitions).

jlazkano, 2019-01-16:
Thanks perexg,

I return the device to Amazon and get new one.

Regards.

closed 2019-01-19 by perexg