perexg/satip-axe #137: Megasat SAT>IP Server 2

danielkucera, 2019-01-16:
Hi @perexg ,
do you think, this device is also supported? I tried to boot from flash but without success.
Can you make an educated guess which port might be the serial console from this picture? Does it look similar to other devices you have worked with?
[image]

http://cdn-reichelt.de/documents/datenblatt/F100/MAN_SAT-IP_SERVER2_EN.pdf

perexg, 2019-01-16:
I don't think that it's the compatible device with the axe firmware. I think that the serial port is on one of the white 4-pin connector (I would try the one nearby the DC input at first - usually, those connectors have GND/3V on the border pins).

Have you tried to obtain the GPL software from the vendor? They should provide all GPL software they used in the firmware according the licence to the end user (kernel/uboot etc.).

danielkucera, 2019-01-17:
Hi @perexg , is the storage layout (1 MB NOR / 128 MB NAND) similar to Telestar? I suppose that the SPI NOR chip contains bootloader. Could you provide a dump from your device?

danielkucera, 2019-01-18:
Ok, I dumped that nor (attached). It contains:
```
$ binwalk megasat.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1048576       0x100000        Squashfs filesystem, little endian, version 4.0, compression:gzip, size: 8355 bytes, 4 inodes, blocksize: 131072 bytes, created: 2014-03-21 11:38:37
2031760       0x1F0090        PEM certificate
```
squashfs contains:
```
$ ls -lah /mnt/tst
total 22K
drwxrwxrwx 2  501  501   58 Mar 21  2014 .
drwxr-xr-x 7 root root 4.0K Jul 23 09:58 ..
-rw-r--r-- 1  501  501  253 Mar 21  2014 crc.sh
-rw-r--r-- 1  501  501  143 Mar 21  2014 ldr.sh
-rwxr-xr-x 1  501  501  17K Mar 21  2014 WDReset
```
some interesting strings from WDReset:
```
/opt/STM/STLinux-2.4/devkit/sh4/lib/gcc/sh4-linux/4.5.3/include
GCC: (GNU) 4.5.3 20110525 (STMicroelectronics/Linux Base 4.5.3-93)
/home/macaroni/users/products/stm2.4/build/packages/stm-target-glibc-sh4/BUILD/glibc-2.10.2/csu
GNU C 4.5.3 20110525 (STMicroelectronics/Linux Base 4.5.3-93)
/dev/mem
```

danielkucera, 2019-01-18:
[megasat.zip](https://github.com/perexg/satip-axe/files/2774462/megasat.zip)

danielkucera, 2019-01-18:
[bootlog.txt](https://github.com/perexg/satip-axe/files/2774534/bootlog.txt)

danielkucera, 2019-01-25:
Ok, I managed to compile and boot u-boot, still need to correctly configure NAND flash
```
Board: STx7108-HDK  [32-bit mode]     
U-Boot 1.3.1-0-g3cbc4aae (Jan 25 2019 - 15:22:52) - stm24_0056-hdk7108

DRAM:  128 MiB
ERROR: flash_write_cmd() ignoring write request (info->portwidth=16)
## Unknown FLASH on Bank 1 - Size = 0x00000000 = 0 MiB
NOR:     0 KiB
NAND:  nand_bbt: ECC error while reading bad block table
128 MiB
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
HDK7108> 
HDK7108> 
HDK7108> 
HDK7108>
```

danielkucera, 2019-01-29:
Ok, I'm much further today. I have booted kernel with loaded drivers, working ethernet and running minisatip8:
```
/ # minisatip8 -f &
/ # [01/01 12:01:16.027 main]: getlocalip: Error 'Network is unreachable'' during connect
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
@@@@@@@@@@@@ fe_open 0
@@@@@@@@@@@@ fe_open 1
@@@@@@@@@@@@ fe_open 2
@@@@@@@@@@@@ fe_open 3
[01/01 12:01:16.107 main]: Initializing with 4 devices

/ # [dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
[dms-ts] stream stop ok
```
Now I need to find a sat dish with LNB to test if it really works...
@perexg , are you interrested in pulling those changes with instructions into your repo or should I maintain my own fork?

danielkucera, 2019-03-02:
@perexg , booting stucks on following line for me:
`nc -l 127.0.0.1:1001 -e /bin/true`
https://github.com/perexg/satip-axe/blob/master/fs-add/etc/init.d/rcS#L124

How should this work?

perexg, 2019-03-02:
https://github.com/perexg/satip-axe/blob/master/fs-add/root/main_init.sh

danielkucera, 2019-03-02:
Thanks @perexg ,
this fixes my issue:
```
@@ -121,7 +130,7 @@ for i in $(cat /proc/devices); do
   maj=
 done
 /etc/init.d/axe &
-nc -l 127.0.0.1:1001 -e /bin/true
+nc -l -p 1001 -e /bin/true
 rm -f /root/main_init.sh /tmp/axe-done
 logger -p local0.notice "AXE drivers load finish"
```
**What version of nc are you using that it works?** Mine is:
```
/ # nc
BusyBox v1.29.3 (2019-01-28 00:01:15 UTC) multi-call binary.
```

danielkucera, 2019-03-02:
And the same here:
```
diff --git a/fs-add/etc/init.d/satip b/fs-add/etc/init.d/satip
index 9513572..a8ebc49 100755
--- a/fs-add/etc/init.d/satip
+++ b/fs-add/etc/init.d/satip
@@ -24,7 +24,7 @@ fi
 cd /tmp
 while test ! -f /tmp/nosatip; do
   while ! test -r /tmp/satip-network; do
-    nc -l 127.0.0.1:999 -e /etc/init.d/satip-network
+    nc -l -p 999 -e /etc/init.d/satip-network
   done
   MINISATIP_PROGRAM=minisatip
   MINISATIP5_PROGRAM=minisatip5
```

danielkucera, 2019-03-02:
my nc accepts following parameters:
```
/ # nc
BusyBox v1.29.3 (2019-01-28 00:01:15 UTC) multi-call binary.

Usage: nc [OPTIONS] HOST PORT  - connect
nc [OPTIONS] -l -p PORT [HOST] [PORT]  - listen

        -e PROG Run PROG after connect (must be last)
        -l      Listen mode, for inbound connects
        -lk     With -e, provides persistent server
        -p PORT Local port
        -s ADDR Local address
        -w SEC  Timeout for connects and final net reads
        -i SEC  Delay interval for lines sent
        -n      Don't do DNS resolution
        -u      UDP mode
        -v      Verbose
        -o FILE Hex dump traffic
        -z      Zero-I/O mode (scanning)
```

perexg, 2019-03-03:
<pre>
~ # nc
BusyBox v1.26.2 (2018-01-25 08:55:26 CET) multi-call binary.

Usage: nc [-iN] [-wN] [-l] [-p PORT] [-f FILE|IPADDR PORT] [-e PROG]

Open a pipe to IP:PORT or FILE

	-l	Listen mode, for inbound connects
		(use -ll with -e for persistent server)
	-p PORT	Local port
	-w SEC	Connect timeout
	-i SEC	Delay interval for lines sent
	-f FILE	Use file (ala /dev/ttyS0) instead of network
	-e PROG	Run PROG after connect
</pre>

paluseb, 2020-01-16:
@danielkucera I bought this device but it won't start. The device after switching on is looking for new software on the internet because it is set up but it does not find it and later nothing happens. What can I do? Which connector is rs232 and what should I upload to start the server and get to the settings?

danielkucera, 2020-01-16:
Not a good device to buy.
[image]

danielkucera, 2020-01-16:
Connect some usb to serial (TTL) adapter and upload the log here.

paluseb, 2020-01-16:
I turned on the device, then connected the network cable and then connected the pendrive with the new software. Sorry for my English
[bootlog.txt](https://github.com/perexg/satip-axe/files/4073308/bootlog.txt)

paluseb, 2020-01-16:
Why "Not a good device to buy."
I uploaded a log on github.

danielkucera, 2020-01-16:
Because it's not supported by the manufacturer nor by satip-axe.

these lines right at the begining are suspicious:
```
LOADER.C 4273> Verifying Upgrader Group [1]
Erasing FlashRom
Programming Image
LOADER.C 1494>Preparing to start Application
```

do you have some tool for SPI flash programming? I can send you alternative u-boot I compiled.

paluseb, 2020-01-16:
I don't have a programmer. But I may be looking for someone who has the right equipment and skills. If possible, send these files and commands that need to be programmed.

danielkucera, 2020-01-16:
What's your goal? To run the original firmware or my not yet working version of satip-axe?

paluseb, 2020-01-16:
At the beginning I want to run the original firmware. I don't know that later. Can your software run with a usb stick?

danielkucera, 2020-01-16:
It can run from USB but not without modifying and disabling the original software.
Try USB upgrade from here 
https://www.megasat.tv/en/support/downloads-de/download-info/satip-server-2-firmware/

paluseb, 2020-01-16:
I tried it but it doesn't work. However, I noticed an error here:
`webremote/advertise.c 914>Advertise Created Successfully{Tsk[0x1c1c3cc] Ret[0] Sudp[38]}ETH LinkUpStatus...: Linkup
IP Settings Mode...: Manual
DHCP Status........: Success
IP Addr............: 192.168.5.10
NetMask............: 255.255.255.0
Gateway Addr.......: 192.168.5.254
PriDns.............: 192.168.5.254
Sec Dns.............: 0.0.0.0
MAC Addr...........: B8:87:1E:**0**:AF:55.`
[log1.txt](https://github.com/perexg/satip-axe/files/4074298/log1.txt)

After changing the network settings (pc-server lan connection) I uploaded the loader again and now I have such a log.
[log3.txt](https://github.com/perexg/satip-axe/files/4074306/log3.txt)

Unfortunately, megasat is not connecting to the network now. As if it wasn't downloading settings from a router.

danielkucera, 2020-01-17:
So does everything work ok now?

paluseb, 2020-01-18:
Yes, probably yes. I haven't connected the antenna yet but I can make all settings on the device