perexg/satip-axe #139: Unable to build busybox

danielkucera, 2019-01-27:
```
make -C apps/busybox-1.26.2 CROSS_COMPILE=/opt/STM/STLinux-2.4/devkit/sh4/bin/sh4-linux-
make[1]: Entering directory '/root/satip-axe/apps/busybox-1.26.2'
  SPLIT   include/autoconf.h -> include/config/*
  GEN     include/bbconfigopts.h
  GEN     include/common_bufsiz.h
  HOSTCC  applets/usage
  GEN     include/usage_compressed.h
  HOSTCC  applets/applet_tables
  GEN     include/applet_tables.h include/NUM_APPLETS.h
  GEN     include/applet_tables.h include/NUM_APPLETS.h
  CC      applets/applets.o
In file included from /opt/STM/STLinux-2.4/devkit/sh4/lib/gcc/sh4-linux/4.5.3/include/syslimits.h:7:0,
                 from /opt/STM/STLinux-2.4/devkit/sh4/lib/gcc/sh4-linux/4.5.3/include/limits.h:34,
                 from include/platform.h:141,
                 from include/libbb.h:13,
                 from include/busybox.h:8,
                 from applets/applets.c:9:
/opt/STM/STLinux-2.4/devkit/sh4/lib/gcc/sh4-linux/4.5.3/include/limits.h:169:61: error: no include path in which to search for limits.h
In file included from include/libbb.h:13:0,
                 from include/busybox.h:8,
                 from applets/applets.c:9:
include/platform.h:152:23: fatal error: byteswap.h: No such file or directory
compilation terminated.
scripts/Makefile.build:197: recipe for target 'applets/applets.o' failed
make[2]: *** [applets/applets.o] Error 1
Makefile:372: recipe for target 'applets_dir' failed
make[1]: *** [applets_dir] Error 2
make[1]: Leaving directory '/root/satip-axe/apps/busybox-1.26.2'
Makefile:452: recipe for target 'apps/busybox-1.26.2/busybox' failed
make: *** [apps/busybox-1.26.2/busybox] Error 2
[root@fedora satip-axe]# 
```

danielkucera, 2019-01-27:
Ok, this firmware is really not easy to build, but finally I have at least booted kernel and busybox running:
```
/ # busybox cat /proc/cpuinfo 
machine         : hdk7108
processor       : 0
cpu family      : sh4
cpu type        : STx7108
cut             : 2.x
cpu flags       : fpu icbi synco fpchg
cache type      : split (harvard)
icache size     : 32KiB (2-way)
dcache size     : 32KiB (2-way)
address sizes   : 32 bits physical
bogomips        : 495.61
```
@perexg , can you please explain and/or document how to properly create the directory `STLINUX=/opt/STM/STLinux-2.4` which is used for building? The Makefile doesn't say a word about it...

perexg, 2019-01-29:
I don't remember. I think that I just installed their iso on centos 7. I have two byteswap.h files:

<pre>
/opt/STM/STLinux-2.4/devkit/sh4/target/usr/include/bits/byteswap.h
/opt/STM/STLinux-2.4/devkit/sh4/target/usr/include/byteswap.h
</pre>

danielkucera, 2019-01-29:
Cool, can you please find that iso or link where to get it?

danielkucera, 2019-03-03:
Ok, I managed to build image from scratch with installed STLinux-2.4-sh4-20141119.iso and I'd like to add a pull request with howto. But I still hit two problems:
```
miscutils/nandwrite.c:151:35: error: ‘MTD_FILE_MODE_RAW’ undeclared (first use in this function)
```
and
```
util-linux/blkdiscard.c:71:26: error: ‘BLKSECDISCARD’ undeclared (first use in this function)
```
I see two possible solutions to this: 
- add patches to /patches to fix these and apply them before building
```
vi apps/busybox-1.26.2/miscutils/nandwrite.c
#include <linux/version.h>
#if LINUX_VERSION_CODE < KERNEL_VERSION(3, 2, 0)
# define MTD_FILE_MODE_NORMAL MTD_MODE_NORMAL
# define MTD_FILE_MODE_RAW MTD_MODE_RAW
#endif
```
and
```
vi apps/busybox-1.26.2/util-linux/blkdiscard.c 
                        (opts & OPT_SECURE) ? BLKDISCARD : BLKDISCARD,
```
- disable these features like this:
```
@@ -450,7 +466,9 @@ apps/$(BUSYBOX)/Makefile:
 
 apps/$(BUSYBOX)/busybox: apps/$(BUSYBOX)/Makefile
        make -C apps/$(BUSYBOX) CROSS_COMPILE=$(TOOLCHAIN)/bin/sh4-linux- defconfig
-       make -C apps/$(BUSYBOX) CROSS_COMPILE=$(TOOLCHAIN)/bin/sh4-linux-
+       sed -i 's/CONFIG_NSENTER=y/CONFIG_NSENTER=n/' apps/$(BUSYBOX)/.config
+       sed -i 's/CONFIG_SYNC=y/CONFIG_SYNC=n/' apps/$(BUSYBOX)/.config
+       make -C apps/$(BUSYBOX) -j $(CPUS) CROSS_COMPILE=$(TOOLCHAIN)/bin/sh4-linux-
 
 .PHONY: busybox
```

@perexg , **do you remember these problems? Which way you preffer?**

danielkucera, 2019-03-03:
proposed change #143