perexg/satip-axe #145: Oscam 11516

bylderup, 2019-04-21:
a version with build 11516 (or newer9 would be nice

antaril, 2019-04-24:
yes that would be cool

Knacki2212, 2019-05-01:
yes, version 11516 (or newer) would be fantastic

bylderup, 2019-05-05:
my interim solution:
Downlad an Oscam 11517 sh4 binary file 
SSH (e.g. use Winscp) to Digibit R1
upload binary file to /etc/sysconfig/
rename file to oscamsh4 and chmod 
new file in /etc/sysconfig/ named myboot
edit myboot :
#!/bin/sh
exec /etc/sysconfig/oscamsh4 -c /etc/sysconfig/oscam -u
save file and chmod
edit /etc/sysconfig/config file:
disable buildin OSCAMD with #
enable CUSTOMD, remove # in line CUSTOMD="/etc/sysconfig/myboot"
save config file

reboot and have fun

antaril, 2019-05-06:
thx works like a charme ;)

jandy123, 2019-06-14:
Sorry, but what is the added value of oscam 11516 ???

perexg, 2019-10-08:
https://github.com/perexg/satip-axe/issues/155

closed 2019-10-08 by perexg