perexg/satip-axe #146: minisatip8 with SW descrambling

jandy123, 2019-05-06:
In case people are not aware, I've managed to get minisatip descramble encrypted streams on the box. Performance is decent, i.e., I can stream 4x decrypted channels to LAN clients. 

Regarding clients, I use the TV Mosaic Live app; works very good with minisatip :).

Did anyone manage to get the 2nd CPU working ? I'm thinking to recompile the kernel with SMP and #CPU = 2. Any insights ?

@perexg: Great firmware, btw !

pbriesch, 2019-05-06:
Could you please elaborate a bit more how you accomplished this?

perexg, 2019-05-06:
I think that minisatip8 was compiled with the sw descrambling which is slow. It's a dup of issue #17 (including the second CPU question). I don't think that the second CPU can be used for the linux kernel, but I may be wrong.

jandy123, 2019-05-06:
Yes, indeed, I've compiled minisatip8 with SW descrambling (libdvbcsa) and made a package replacing original minisatip8.

Slow, or not, I managed 4x simultaneous SD clients, which here is good enough :).

====

@perexg: Well, have you tried enabling SMP in your kernel ? I have tried but box does not boot. The wired part is that also compiling the kernel  with the original config file I get the same result... Any ideas ? Could you please make a *bootable* test image with SMP enabled and #CPUs = 2 ? Thank in advance !

perexg, 2019-05-06:
We are using also the DVB driver binaries from Inverto which must be recompiled (assuming that 2 ST40 CPUs are supported in the used kernel), so the massive kernel config changes are not possible.

jandy123, 2019-05-06:
For testing, do the follwing:
1. modify /etc/sysconfig/config and append to the MINISATIP8_OPTS -o 127.0.0.1:9000
2. configure oscam
3. upload the attached package in /mnt/data/satip-axe/packages/minisatip/
4. connect with SAT>IP client and select scrambled channel.

I attached the package.

jandy123, 2019-05-06:
@perexg: Yes, I thought about this... If the problem is that precompiled modules do not load, we could probably solve this (change version info, and/or enable options in kernel config allowing such packages). The problem is: does the kernel boot ? I do not have a serial-to-usb adapter to test this; this is my problem. Also, it's unclear to me why I am unable to boot the kernel compiled with the original config file (and also with the original cpio romfs).... Any idea ?

jandy123, 2019-05-06:
Sorry, here is the package:

[minisatip.tar.gz](https://github.com/perexg/satip-axe/files/3149080/minisatip.tar.gz)

jandy123, 2019-05-06:
@perexg: Btw., does the original kernel support dual core ? If yes, that would be an easy hack..., but I doubt ;).

jandy123, 2019-05-10:
Further updates regarding descrambling on the box:
- enabling 2nd CPU with SMP config in the kernel is not going to work, as there is no SMPimplementation in ST-provided 2.6.x kernels for our CPU STX7108.
- for HW descrmabling we need (at least ?) the stpti5_ioctl.ko module which we do not have. If anyone can get it from a similar STM, please let me know.
- the only remaining option seems to be to use to coprocessors, which I'll look into. If anyone has any info about this, let me know.

palik42, 2019-05-10:
Thank you. But I have issue :D descrablim working for me only when I using parameter -lll :D ... (watching with VLC)

CvH, 2019-05-10:
@jandy123 did you build https://github.com/glenvt18/libdvbcsa ? thats an arm optimized version that should help, a bit at least

jandy123, 2019-05-11:
@palik42: Sorry, but I do not understand what you say...

@CvH: Yes, I've compiled that version with native support for uint32 + all possible compiler optimizations. Unfortunately, sh4 does not have vectorised (SIMD) instructions, as far as I can tell. I found some mentioning of some multimedia instructions, but they are nowhere to be found... 

Question:
- oscam has support for sh4 HW decoding on some boxes. Can anyone help and tell me what options enable those ?

perexg, 2019-05-11:
>     * oscam has support for sh4 HW decoding on some boxes. Can anyone help and tell me what options enable those ?

Unfortunately, inverto didn't include the secure co-processor STAPI to their drivers :-( It seems like that the core of this code is available only in a binary form (under NDA or so? - just guess).

palik42, 2019-05-11:
@jandy123 
I install your package by your steps.. I have Digibit R1.
When satip-axe start minisatip8 with oscam .. (minisatip8 -f -g -o 'IP' 9000 .. descrambling is not working for me.... But when I start debuging with parameter -lll ... minisatip8 -f -o 'IP' 9000 -lll  descrambling start working....

jandy123, 2019-05-11:
@perexg: Right, but it seems that we do actually have the modules, right ? We have this "/dev/stapi/stpti5_ioctl". We "only" do not know the ioctls, or am I wrong ?

Regarding the use of ST200. This will require a major write-up of libdvbcsa and/or minisatip, Do you have any tools for ST200 ?

jandy123, 2019-05-11:
@palik42. What does option "-III" mean ?

perexg, 2019-05-11:
> Right, but it seems that we do actually have the modules, right ? We have this "/dev/stapi/stpti5_ioctl". We "only" do not know the ioctls, or am I wrong ?

The PTI API is not enough for the descrambling.

jandy123, 2019-05-11:
@perexg: But which other module is needed ? I've compiled oscam + stapi5 (with some liboscam_stapi5.a for a different device), but I only see "/dev/stapi/stpti5_ioctl" being used/required. Is there another module needed ?

palik42, 2019-05-11:
@jandy123 -l specifies the modules comma separated that will have increased verbosity, logging to stdout in foreground mode or in /tmp/minisatip.log when a daemon Possible modules: general,http,socketworks,stream,adapter,satipc,pmt,tables,dvbapi,lock,netceiver,ca,axe,socket,utils,dmx,ssdp,dvb

jandy123, 2019-05-11:
@palik42: Ok, I understand. Wired that I do not need that version, although I used the original oscamd...

palik42, 2019-05-11:
@jandy123 I now tried in dreambox 920 with satipclient and works.. So then that problem is only for VLC client on PC.

jandy123, 2019-05-11:
@palik42: Ok, good to know. As I said, I'm using it with TV Mosaic Live app on android boxes/TV and it works quite good.

jandy123, 2019-05-11:
@perexg: I think this https://yadi.sk/d/9DiDOKyk3NeTTi may provide all we need to enable the HW descrmabler :). See stapisdk/apilib/src/stpti5/ folder from SDK/STAPI_SDK_REL_42.1.tar.gz to get an idea :).

palik42, 2019-05-11:
@jandy123 Panasonic TV working too :)

perexg, 2019-05-11:
@jandy123 : From stpi5_dvb.rel.pdf:

> STPTI5 is a STAPI-compliant driver for performing transport level DEMUX. In addition to this it
can also perform Section Filtering, PES & ES extraction, Indexing and Packet Insertion &
Replacement operations.

The answer is sttkdma_plain:

> STTKDMA is a STAPI-compliant library for performing key decryption for transport control
words and hardware feature management. It also enables DMA operations with on-the-fly
encryption or decryption.

jandy123, 2019-05-12:
@perexg: Indeed, we do not have the required sttkdma library for 7108.

But, apparently, descrambling is available in stpti4. Maybe we can use that somehow ???

jandy123, 2019-05-12:
@perexg: I cannot find anywhere the st200 micro toolset. I've seen you posted a link which is now dead. Can you share it ?

perexg, 2019-05-12:
I've found only this now: 

- https://www.google.com/search?q=stlinux23-cross-st231
- https://www.google.com/search?q=booting+the+st231+cpus

jandy123, 2019-05-13:
@perexg: Ok, so I've managed to recompile the kernel and enable the 1st coproc: basically append "coprocessor_mem=4m@0x10000000" to the CONFIG_CMDLINE in .config, recompile and run. I can see that the 1st coproc is properly initialized when booting.

Then, I managed to cross-compile a simple test executable for the ST231 using stlinux23-cross-st231* packages, which is loaded fine using the "stslave" application in the sh4 (from package stlinux24-sh4-stslave-0.7-18.sh4.rpm from http://archive.stlinux.com/stlinux/2.4/STLinux/sh4/index.html). 

Problem is that the ST231 does not run/execute. I tested this using peek/poke functionality of the stslave host. Most likely the problem is my ld script. I[ve tried about everything, but still I cannot get this simple test to work. At this stage, I believe it's impossible to move on without proper documentation and/or example(s). 

Well, this is where I'm right now. If you have ideas, let me know.

perexg, 2019-05-13:
I found also this - https://www.google.com/search?q=en.CD18032082.pdf , but it looks like outdated. Perhaps, there should be .ld scripts for reference boards in the cross compiler packages. Yes, the documentation is missing or we don't know where to look. Also, the cache coherency might be in the data path.

jandy123, 2019-05-14:
@perexg: I found that document, but is not very useful. 

Anyhow, I've further modified my simple test program (an infinite loop incrementing some global variable in the data segment so that I can read the var using "stslave" host executable). No libc, no static/constructor initialization, no main function (just _start), etc. Today I'll give it one more try. If this fails, I'm out of options...

You've mentioned caching, what exactly you mean by that ? I suppose I can also write a host executable that mmaps the st231 data-segment location to read the data, but I'm quite sure that's not really different than what stslave does.

jandy123, 2019-05-18:
I've tried about everything I could think of, but didn't get this working. To try more, I'd need the "ST200 micro toolset", which is nowhere to find. If anyone knows where to get t, let me know. Until then, I have to abandon this...

@perexg: I've been experimenting with standby and wake-on-lan. Unfortunately this doesn't work either.  All logs seem normal but tuners have zero signal. Any idea what's going on ? Is there anything to look into//try ?

jandy123, 2019-05-21:
Well, in case anyone is interested, I've managed to get standby & wake-on-lan working. For now, it's some ugly hack of minisatip, but well, it works here. Idea is that whenever no tuner is active (no matter what the CPU activity is, so please pay attention !!!), after 20 minutes of inactivity, the box goes to standby. It can be woken-up (only) by sending a wake-on-lan magic package.  

To test, do the following:
1. Upload minisatip_wol.tar.gz (see attachment) under /mnt/data/satip-axe/packages/ 
2. Create and edit a /mnt/data/satip-axe/sysconfig/localboot file with the following content:

#!/bin/sh
/sbin/ethtool -s eth0 wol g

3. Reboot the machine and test. Please report your findings here. 

Since I do not have access to a power-wall watt meter, I'd appreciate if anyone could report here the power savings. Thanks !

(Btw., this version has also SW descrambling enabled.)

CvH, 2019-05-21:
> For now, it's some ugly hack of minisatip

could you share the changes you made ? :)

jandy123, 2019-05-22:
@CvH:

I attached the diff. As I said, this is just some hack. Probably this can be implemented as a pair of crond scripts, provided that we can easily get the tuner state from a script (maybe using procfs). Any how, it seems that there is very little interest, so, well, for me this is good enough...  

[minisatip8_wol.diff.gz](https://github.com/perexg/satip-axe/files/3207152/minisatip8_wol.diff.gz)

Geiselmen, 2019-06-29:
Thanks for minisatip.tar.gz.
I can stream 2x decrypted  HD channels to LAN clients.
At the third it starts to jerk but is not bad.
I have 2 Digibit R1 for my Clients.
Without WOL:
Idle: ca. 9W
1 Tuner aktiv: ca. 14W
2 Tuner aktiv: ca. 15,5W
3 Tuner aktiv: ca. 17W
4 Tuner aktive: ca. 18W

jandy123, 2019-06-30:
@Geiselmen: Thanks for the numbers regarding power consumption. Still on the high side, I'm actually wondering if there is any difference between sleep and no sleep. Also, there are more options to be tested regarding sleep state (standby is used right now, but mem is also supported). Could you please test that as well ? I can provide a new minisatip8 executable if you want.

Geiselmen, 2019-06-30:
Test the WOL for you?
Or more?
I can Test it at the Weekend.

jandy123, 2019-07-01:
@Geiselmen: Yes, could you please test power usage in standby ? Please use the wol version from my post on 21st May above.

Very curious about this...

Geiselmen, 2019-07-01:
Power usage in WOL standby: ~3,6W
What I'm still interested in is whether a Raspberry with tvheadend can more stream then 2x decrypted so I can deactivate a Satip server.

jandy123, 2019-07-03:
@Geiselmen: Thanks for the info on power usage. So, it is worth using standby with WOL...

Regarding  Raspberry with tvheadend, given the fact that we can decrypt 2-3 channels on this box, decrypting 4-6 channels should work fine on the rpi. Also, libdvbcsa has explicit optimizations for ARM arch. Make just sure that you use a version which has optimizations for ARM enabled. 

You won't be able to deactivate minisatip server, i.e., you still have to use it for the tvheadend to be able to talk to the box via the sat2ip protocol. Just make sure you disable decryption on the box, i.e., remove option "-o 127.0.0.1:9000" in MINISATIP8_OPTS. Alternatively, you can just use the original minisatip8 version, but then you won;t have standby & WOL.

Geiselmen, 2019-07-03:
The problem that i have with WOL is: sending a wake-on-lan magic package.
I only can send it with my Phone not Automatik with my TV.

jandy123, 2019-07-03:
Geiselmen: Yep, WOL is not standard on TVs, but probably is on your router...

cesarsferrreira, 2020-02-02:
> For testing, do the follwing:
> 
> 1. modify /etc/sysconfig/config and append to the MINISATIP8_OPTS -o 127.0.0.1:9000
> 2. configure oscam
> 3. upload the attached package in /mnt/data/satip-axe/packages/
> 4. reboot box.
> 5. connect with SAT>IP client and select scrambled channel.
> 
> I attached the package.
> 
> [minisatip.tar.gz](https://github.com/perexg/satip-axe/files/3149084/minisatip.tar

i did reached to make it work, with your minisatip, but not with your oscam4sh. i used SATIP-AXE server 201810211549-15 version, because newer release don´t work.
but i´ve still have problem to descramble more than one stream.

any news how to make it work??

zildan, 2020-03-05:
[tdt/cvs/driver/pti](https://gitorious.org/open-duckbox-project-sh4/pingulux-git/commit/11173a802bf9974ad9383aa6e6b96156b2e7862e?p=open-duckbox-project-sh4:pingulux-git.git;a=snapshot;h=11173a802bf9974ad9383aa6e6b96156b2e7862e;sf=tgz)

nrpetonr, 2020-04-09:
> Well, in case anyone is interested, I've managed to get standby & wake-on-lan working. For now, it's some ugly hack of minisatip, but well, it works here. Idea is that whenever no tuner is active (no matter what the CPU activity is, so please pay attention !!!), after 20 minutes of inactivity, the box goes to standby. It can be woken-up (only) by sending a wake-on-lan magic package.
> 
> To test, do the following:
> 
> 1. Upload minisatip_wol.tar.gz (see attachment) under /mnt/data/satip-axe/packages/
> 2. Create and edit a /mnt/data/satip-axe/sysconfig/localboot file with the following content:
> 
> ```
> #!/bin/sh
> /sbin/ethtool -s eth0 wol g
> ```
> 
> 1. Reboot the machine and test. Please report your findings here.
> 
> Since I do not have access to a power-wall watt meter, I'd appreciate if anyone could report here the power savings. Thanks !
> 
> (Btw., this version has also SW descrambling enabled.)
> 
> [minisatip_wol.tar.gz](https://github.com/perexg/satip-axe/files/3203857/minisatip_wol.tar.gz)

@jandy, what has been your experience meanwhile? do you by chance use kodi for sending wol? if so, does it work?

cesarsferrreira, 2020-04-09:
Does anyone has this problem i have:

i have telestar digibit R1, with firmware:

version: SATIP-AXE server 201810211549-15

and local OSCam r11434

problem: when i open an inicial tv channel it can decrypt, and if i open another diferent channel it open fine and decrypt, 
but if i close all tv channels, it don´t decrypt until i reboot oscam (i´ve tried also with tvheadend with oscam, and happens the same, with tvheadend, i only have to disable and disable oscam on tvheadend to decrypt. again.

i´m since the begining this problem. and i can´t fix this problem

regards

nrpetonr, 2020-04-10:
@cesarsferrreira, no such problem for me. have a look into oscam log, difficult to say what the problem is...

cesarsferrreira, 2020-04-10:
Hello nrpetonr,

thanks for your attention. 

attached logs from my oscam and also oscam prints, if they may help you

i hope you can help me.

regards
[oscam-log.txt](https://github.com/perexg/satip-axe/files/4462013/oscam-log.txt)
[image]
[image]
[image]

nrpetonr, 2020-04-10:
well, from what i see, oscam seems to be descrambling, no sign of problem there. sorry not to be of better help.

cesarsferrreira, 2020-04-10:
Hello nrpetonr,

please as they say, picture (video) is more explaning than words.

i attached a video, explaining, that i always need to activate and deactivate oscam in tvheadend, for decrypting channels, after closing previous channel.

any ideas?? 

thanks 

regards

link for video:
https://we.tl/t-f5OkGrTe7i

nrpetonr, 2020-04-11:
hope however you use it is legal, otherwise we are breaking rules...

in general, oscam withing this minisatiip is functional, so it does not really belong here. problem is within your setting of oscam most probably.
few hints, might help you to solve it:
1st, try to change the protocol u use within tvh, see attached picture. Don't forget to change respective [dvbapi] setting under oscam.conf
2nd, your ecm times are quite high, can cause problems when descrambling or even timeout (that could explain your problem). might be fixed by 1st idea.

good luck...
[image]

cesarsferrreira, 2020-04-11:
Hello nrpetonr,

it worked thanks..

and this system as i told is for testing purpose...

regards