perexg/satip-axe #148: Digibit r1 satip-axe with dreambox 920 satipclient plugin

palik42, 2019-05-08:
Hello,
I have problem. Have dreambox 920 with satipclient plugin and Digibit R1 with satip-axe as server.
I have default setting in satip-axe and I am using TWIN (1 and 2 port).. I can connect to minisatip via VLC and can see TV... But when I setup everything in dreambox minisat ip showing this but I don't see channel at dreambox. Can you help me please?
[image]

knowthelist, 2019-05-09:
I have the same problem on my VU Solo2 with satipclient and Digibit R1. But only after 2-3 weeks successfully viewing. Then I have to restart the satip-axe server and it works again for a while. But a restart with satip-axe is not very nice due to the bad start behaviour. You have to switch off power at least 4 times until a successful boot is possible.

knowthelist, 2019-05-09:
I figured out, that it helps for me to kill minisatip8, which restarts automatically.

palik42, 2019-05-11:
I found problem... I have two tuners in DM920 - BCM45208 and BCM45308 ... When I disconnected BCM45308 .. satip start working... But now I don't know if it is problem in satipclient or in dreamos(because I tried lot of images)...

palik42, 2019-05-11:
"

No Chance. When you use a dm920 with two fbc tuners it is not possible to use vtuners... but vtuners are needed for satip client.

This is a limitation of the hardware, the drivers, and last but not least the linux tv api which supports a max of 16 tuners.

Each FBC Tuner create 8 frontend devices... so with two fbc tuners you reached the limit.
" text is by Ghost

closed 2019-05-21 by palik42