perexg/satip-axe #150: additional packages

jandy123, 2019-06-11:
I've made a few additional packages for satip-axe. Copy them under  /mnt/data/satip-axe/packages/ and reboot the box.

Packages:
1. htop
[htop.tar.gz](https://github.com/perexg/satip-axe/files/3276156/htop.tar.gz)

2. midnight commander
[mc.tar.gz](https://github.com/perexg/satip-axe/files/3276157/mc.tar.gz)

3. pshare for exporting playlists over dlna; see http://xupnpd.org/pshare/ for usage
[pshare.tar.gz](https://github.com/perexg/satip-axe/files/3276161/pshare.tar.gz)

Remark:
Although pshare works, I have no luck playing some services on my LG tv over dlna. I have no clue why, since another satip client (tvmosaic under android) works very well. So, I suppose there is some issue with minisatip8 streaming over http, but I'm not sure. I'd appreciate if anyone can share some light. 

Due to this, I'll compile tvheadend to run on the box. From my initial experiments minisatip8 + tvheadend seems to have no issue streaming to clients over http. I'll test some more and see if this combination minisatip8 + tvheadend on the box is viable...

4. for minisatip + WOL support + SW descrambling see https://github.com/perexg/satip-axe/issues/146

jandy123, 2019-06-11:
Ok, tvheadend is no go. It's too heavy on the box.

So, at this stage I'm stuck using misatip8 to stream over http. This just doesn't work reliably... Any idea ?

9000h, 2019-06-11:
Hi.

> I'm also experimenting with a satip sattelite scanner.
why not uploading the code on github
CU
9000h

jandy123, 2019-06-12:
The sources of the scanner are not yet in a stage in which they could be shared ;). And, the scanner is not of much use, provided that minisatip http streaming is not working correctly... 

@perexg: Any clue what can go wrong with http streaming ?

perexg, 2019-06-12:
@jandy123 : Do you use my version of minisatip8 or your own? Does VLC work with the HTTP URLs?

jandy123, 2019-06-12:
@perexg: I use minisatip8 with SW descrambling (the one from #146). The situation is this.

I have problems streaming over http using m3u playlists from minisatip, directly to IPTV and DLNA clients. This only happens with some services, while others do work. Btw., decryption is done on the box using minisatip compiled with oscam support. I know that the pids are correct since I can play the problematic services using other satip clients (tvheadend, tvmosaic live android app, etc.). Again, this only happens on certain channels (even on the same transponder on Thor 0.8W). 

I should also add that this happens in VLC player and using any IPTV app on my LG TV. Playing problematic services does work using ffplay and mplayer, after some fairly large delay. Also strange enough, if I start playing a service using ffplay and then try VLC, I can play the problematic channels on my TV and using VLC.

Any idea what can be the problem ?  I would really like to be able to directly play from digibit box to "smart TVs" and clients based on VLC. 

I can provide logs, exact commands, example services etc. 

Thanks in advance !

jandy123, 2019-06-12:
I attached a capture of a problematic channel using:

wget "http://192.168.2.134:8080/?freq=11996&src=1&pol=v&msys=dvbs&sr=28000&fec=78&pids=0,151,163,150"

This plays fine using ffplay, but not with vlc: lots of errors with vlc, ffplay works fine.

@perexg: What can this be ???

[sample.mp2ts.gz](https://github.com/perexg/satip-axe/files/3282631/sample.mp2ts.gz)

jandy123, 2019-06-13:
Turns out my issue is (related ???) to this: https://github.com/catalinii/minisatip/issues/437.

After setting opts.pmt_scan = 0, problematic channels play via http, at least with vlc. Will try on IPTV clients and report here.

Pity that people are unwilling to help. For sure some are aware about this issue...

Anyhow, I'll continue updating this  post, maybe others find it useful.

I wonder what the implications of setting opts.pmt_scan = 0. I do not even dare to ask, since most likely I won't get any reply... pity.

jandy123, 2019-06-13:
Good news ! IPTV clients also work (including my LG TV). And so far I haven't seen any downside of this modification...

Will work now on the satip satellite scanner. Will post here when I have something reasonable.

perexg, 2019-06-14:
> I wonder what the implications of setting opts.pmt_scan = 0. (I do not even dare to ask, since most likely I won't get any reply... pity. I'm just trying to help here, although I'm a complete newbie in all this stuff.)

The problem is that minisatip allocates the PMT PIDs for the scan purposes when this option is turned on. Minisatip does not handle properly the limit of PIDs available in the hardware filter when the scan is active, so the new PIDs for clients are just not subscribed, so the result MPEG-TS stream is filtered (and you can get no-video/no-audio or both).

perexg, 2019-06-14:
And I already asked @catalinii to not enable this by default or the code should be fixed (client PID allocations should have the preference).

jandy123, 2019-06-14:
@perexg: Ok, I understand. But is it safe to use pmt_scan = 0 ? As far as I can tell, all other clients still work, but maybe the zapping time is somewhat larger. Btw., thanks for the reply.

jandy123, 2019-06-14:
@9000h: I edited the first post and included the satip scanner. Please modify/upload it to github, etc. as you wish...