perexg/satip-axe #153: Adapter allocation regression: No longer able to use multiple tuners with single input

Emil-66, 2019-09-05:
There is a regression between the included minisatip7 and minisatip8 binaries: With minisatip7 it is possible to use a single input with "-7 0:1,0:2,0:3" and tune different transponders on the same polarization/band/diseqc combination. With minisatip8, the slave adapters are (erroneously) rejected under the same circumstances. Minisatip upstream has since been updated with a fix affecting this aspect (compare_slave_parameters in adapters.c). Please rebuild the firmware with a current version of minisatip. That should fix the regression.

perexg, 2019-10-08:
Could you test this updated fw: http://s000.tinyupload.com/?file_id=55622426912282482797 ?
Note that minisatip 1.0.2 is controlled using MINISATIP_ lines in the config. 

<pre>
MINISATIP="yes"
MINISATIP_OPTS=""
</pre>

Disable MINISATIP8 or any other.

perexg, 2019-10-08:
https://github.com/perexg/satip-axe/issues/155

Emil-66, 2019-10-11:
Preliminary testing of just this feature shows that the regression is fixed. I confirmed that minisatip version 1.0.2-axe1 is running and was able to tune multiple transponders simultaneously with just one input. Thank you!

closed 2019-10-11 by Emil-66