perexg/satip-axe #154: Problem in vertical frequencies...

rubenqs12, 2019-09-23:
I´m using the latest perexg/satip-axe firmware (satip-axe-201810211549-15 - Sun Oct 21 2018) for the Digibit R1.
The problem is that the channels in vertical frequencies (example Movistar +) do not receive a signal, but if I first reproduce a channel of a horizontal frequency and change to a vertical frequency, if it is played.
If I only play the vertical channel, I only receive 0% SNR on tvheadend.

An example:
Play "Canal sur" in tvheadend = Not work.
Play in this order... "Travel 4K" and then "Canal sur" = It works perfectly and stable and even the other tunners work.

I have a monoblock LNB for Astra and Hotbird respectively, and only under one cable to input 0 ...

In minisat I have configured the next boot.
-7 0:1,0:2,0:3 '/ minisatip8

But I have the problem commented ... could someone help me to make it work properly, please? Thank you

rubenqs12, 2019-10-10:
Amazing... 
I had been with this problem for weeks ... now it works from the beginning.

Thank you!

This work with monoblock
/sbin/minisatip8 -f --diseqc-timing *:155-54-15-15-15-0 -7 0:1,0:2,0:3

m4tt075, 2019-10-24:
@rubenqs12: Bloody h***! I've been chasing the same issue for weeks, too. Didn't expect to find the solution here. Thank You!

@perexg: Just for information, in case you'll see similar reports elsewhere: The firmware upgrade alone was not sufficient to fix it. Changing the diseqc timing as above did. I'm running Digibit R1s.